roleoroleo/yi-hack-Allwinner-v2 #192: Support for Y211GU

AaronAutomation, 2021-06-16:
Replaced one of my dead cameras and discovered SN: IY211GUxxxxxxxxxx
Any potential support for this version?

roleoroleo, 2021-06-17:
Which fw version is?

AaronAutomation, 2021-06-17:
9.0.36.00_202011101611

roleoroleo, 2021-06-17:
Already supported.
Check the readme.

AaronAutomation, 2021-06-17:
There are no matching file prefixes for y211gu in the readme that I could find. I already tried y211ga but the camera does not attempt to flash revisions, it just boots up as normal. Previously, I used the v4 version on my older cameras. Am I just placing the extracted .tar file on the root of the SD card and then inserting it into the camera and then turning it on? The directions seem pretty straight forward so because it isn't flashing I assumed it was an unsupported file version.

roleoroleo, 2021-06-17:
The serial is not strictly related to the suffix.
To know exactly the suffix you need to dump the fw and check the content.
Check the wiki to know how to dump the fw.
But I think your cam is a y211ga.

AaronAutomation, 2021-06-17:
Same result attempting to do the FW dump. Putting the decompressed backup file (backup.tar) on the root of a freshly formatted FAT32 sdcard and putting it in the camera then powering up results in it just turning on. The camera creates a 'log' folder with the files log_first_login.tar.gz + log_wifi_connected.tar.gz and then a 'record' folder with recordings regardless of what's on the root of the card.

roleoroleo, 2021-06-17:
Share the dump with me.

AaronAutomation, 2021-06-17:
https://github.com/AaronAutomation/YiDump/blob/main/YiDump.7z

roleoroleo, 2021-06-17:
You need to copy to the sd card the content of the archive.
Your card should appear with this structure:
```
/
|-- Factory/
|   |-- config.sh
|   |-- factory_test.sh
```

AaronAutomation, 2021-06-17:
Thank-you. I thought I had read somewhere to decompress the files rather than extract. I had tried every combination as far as folder depth earlier, but I suspect now it may have been a format issue with the card originally. It did take the flash this time. Thanks again.

closed 2021-06-17 by AaronAutomation