roleoroleo/yi-hack-Allwinner-v2 #194: Unable to use firware with camera YI Home Camera 1080P AI+

KrisRux, 2021-06-18:
Hi,

the camera is an IFUS with firmware 8.3.0.0A but the firmware not working.

can you help me ?

roleoroleo, 2021-06-18:
Probably it's not an Allwinner-v2.
Try the Allwinner.

KrisRux, 2021-06-18:
NOT working with Allwinner
Yi 1080p Home 9FUS | 8.3.0* | y25ga

Thanks

roleoroleo, 2021-06-18:
Try to create a dump of the flash and I will check the model.
Read the wiki.

KrisRux, 2021-06-19:
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6680115/backup.zip)

SevBi, 2021-06-20:
Hi both,

I'm facing the same with a Yi camera AI+. My cam is BFUS with 9.0.19.02 firmware. I tried several times to install the

Antuzzo, 2021-06-20:
Hi,
I have the same issue.
My cam is home 1080p IFUS with 9.0.36 firmware.
The camera not flash firmware.

Can you help me?

KrisRux, 2021-06-22:
Ho @roleoroleo 

Any news about my backup?

Regards

SevBi, 2021-06-22:
> Hi @roleoroleo
> 
> Any news about my backup?
> 
> Regards

Hi KrisRux,

sorry to pollute your initial request. But I guess a common solution would be helpful.

Best
S

roleoroleo, 2021-06-22:
> 
> 
> Hi @roleoroleo
> 
> Any news about my backup?
> 
> Regards

After unpacking your file, I confirm that your cam is an Allwinner y25 model and it's already hacked.

roleoroleo, 2021-06-22:
90% the problem is the sd card (when your cam is listed in the readme and both model and fw version match):

- try another sd card
- try to reformat it
- check if you extracted correctly the archive
- try to extract it again (if you are a windows user, use 7zip)

The remainig 10%, the cam is new and not supported.

SevBi, 2021-06-22:
> 90% the problem is the sd card (when your cam is listed in the readme and both model and fw version match):
>
> * try another sd card
> * try to reformat it
> * check if you extracted correctly the archive [...]

Hi @roleoroleo 

If I share a backup file with you: are you able to check my status and let me know if my cam is supported or not?
I have the impression that yes but I prefer to have an external view.

thanks
S

roleoroleo, 2021-06-22:
> Hi @roleoroleo
>
> If I share a backup file with you: are you able to check my status and let me know if my cam is supported or not?
> I have the impression that yes but I prefer to have an external [...]

Yes.

KrisRux, 2021-06-22:
> 90% the problem is the sd card (when your cam is listed in the readme and both model and fw version match):
>
>     * try another sd card
>
>     * try to reformat it
>
>     * check if you extracted [...]

Hi @roleoroleo 

I have tried with 3 different MicroSD (Samsung 64Gbs, Samsung 32Gbs and Sandisk 8Gbs) and with the same results.
I use 7zip....

I think this is the 10%<remainig .... :(

roleoroleo, 2021-06-23:
> I think this is the 10%<remainig .... :(

This is not possible.
I checked your backup.

KrisRux, 2021-06-23:
> 
> 
> > I think this is the 10%<remainig .... :(
> 
> This is not possible.
> I checked your backup.

when i connect the power with sd the camera do some reboot (click - click) and remaing with yellow led blincking...

what can i do for check ?

roleoroleo, 2021-06-23:
Try again from scratch.
1. Use the unbrick feature and restore the cam to the original state.
2. Reset the cam with the button.
3. Pair the cam with the app as a new cam.
4. Apply again the hack.

SevBi, 2021-06-24:
Hi @roleoroleo 

Finally I succeeded and now my yi 1080 AI+ is hacked. Thanks for your information.

alvise, 2021-06-24:
I have IFUS with firmware 8.3.0.0A as well, which file you flashed?

KrisRux, 2021-06-24:
> 
> 
> I have IFUS with firmware 8.3.0.0A as well, which file you flashed?

This: https://github.com/roleoroleo/yi-hack-Allwinner

Yi 1080p Home 9FUS | 8.3.0* | y25ga

Ppatou, 2021-06-28:
Hello 
I have a Yi  Home 1080P
9FUS Y21KH 1QP ...
Is this compatible with : Yi 1080p Home 9FUS  8.2.0*  y20ga
or with :  Yi 1080p Home BFUS | 9.0.19* | y21ga
or with not compatible at all ?
Thanks for your help
Patou

roleoroleo, 2021-07-03:
I need to know the firmware version.

Ppatou, 2021-07-03:
Hello
Thanks for your reaction
I used the files for the 9FUS and it works perfect
The firmware in the cam is 4.5.0.0C..... I only discover it later !!
Is there a way to turn on/off the camera with a switch defined in Home Assistant ? Some MQTT switch for example ?
I found this HTTP request : 
[CAMERAHTTPDPort/cgi-bin/camera_settings.sh?save_video_on_motion=yes&sensitivity=low&led=no&ir=yes&rotate=no&switch_on=no](url)
This HTTP request is working fine in a browser but I cannot find how to implement it in HA
Good night
Patou

Ppatou, 2021-07-04:
Hello
I investigate more and found the switch platform command line. But I do something wrong because it does not work.
If I put the command line in a browser it works but not with the switch in HA
Here is my yaml code :
```
switch:
  - platform: command_line
    switches:
      yi_cam_on:
       command_on: "http://192.168.0.14:8080/cgi-bin/camera_settings.sh?switch_on=yes"
       command_off: "http://192.168.0.14:8080/cgi-bin/camera_settings.sh?switch_on=no" 
       command_state: "http://192.168.0.14:8080/cgi-bin/camera_settings.sh?switch_on"
```
Can you help ?

roleoroleo, 2021-07-04:
You miss the command to send:
```
command_on: "curl http://192.168.0.14:8080/cgi-bin/camera_settings.sh?switch_on=yes"
...
```

But I'm working to a switch in the next release of this custom components.
https://github.com/roleoroleo/yi-hack_ha_integration

Ppatou, 2021-07-04:
Thanks I wait the next release

closed 2023-01-08 by stale[bot]