roleoroleo/yi-hack-Allwinner-v2 #196: PTZ inverted?

roleoroleo, 2021-06-18:
I have a question for those who have a ptz camera.
One of the following models: r30gb, r40ga ,h51ga, h52ga, h60ga or q321br_lsx.
Is ptz inverted?
If yes: left/right, up/down or both?

Thanks

Mi-Sanchez, 2021-06-18:
App or webinterface? On webinterface you can modify the buttons (html).

WalkingDot, 2021-06-19:
r30gb is not inverted.

seventhplug, 2021-06-22:
Yi Dome U Pro (Black)
Here is the evidence about PTZ Up/Down inverted. Sorry I've been busy this few days
https://user-images.githubusercontent.com/23691755/122859353-7e037e00-d34e-11eb-83ef-10b44459d9d9.mp4

4lloyd, 2021-08-30:
Yi Dome Camera U BFUS (3K) h51ga
Up/down is inverted. Left/right is OK.

roleoroleo, 2021-08-31:
Ok, I will fix it.

roleoroleo, 2021-08-31:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/51c91f0b924ddc91a9de2cb2f8beb667ef259a47

closed 2023-01-08 by stale[bot]

tcaptalan, 2024-09-16:
Hello, I have the problem of inverted top/bottom on h60ga with firmware 0.3.2. is this normal?