roleoroleo/yi-hack-Allwinner-v2 #197: FW upgrade: No space left on SD

WalkingDot, 2021-06-19:
I tried to upgrade from 0.1.7 to 0.1.8 through the hack interface.
i am receiving this message:
`No space left on SD`

df and fdisk outputs:
```
root@yi-guard-r30gb:~# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1280      1280         0 100% /
devtmpfs                 30400      1824     28576   6% /dev
/dev/mtdblock3            2304      2304         0 100% /home
/dev/mtdblock4            1216      1176        40  97% /backup
tmpfs                    32768       124     32644   0% /tmp
/dev/mtdblock4            1216      1176        40  97% /home/app/script/wifidhcp.sh
/dev/mmcblk0p1           32768       124     32644   0% /etc/passwd
/dev/mmcblk0p1           32768       124     32644   0% /etc/shadow
/dev/mmcblk0p1        31159232     33056  31126176   0% /tmp/sd
tmpfs                    32768       124     32644   0% /etc
/dev/mmcblk0p1        31159232     33056  31126176   0% /tmp/sd/yi-hack/www/record
```

```
root@yi-guard-r30gb:~# fdisk -l
Disk /dev/mmcblk0: 30 GB, 31914983424 bytes, 62333952 sectors
122223 cylinders, 255 heads, 2 sectors/track
Units: cylinders of 510 * 512 = 261120 bytes
```

roleoroleo, 2021-06-19:
I have to check it.
It's a weird behavior.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/www/httpd/cgi-bin/fw_upgrade.sh

WalkingDot, 2021-06-19:
I replaced the fw_upgrade.sh with the new one. Now it's working.
The problem was this line:
```
df /tmp/sd/ | grep mmc | awk '{print $4}'
32644
```

closed 2021-06-19 by WalkingDot

roleoroleo, 2021-06-19:
I will fix it.

WalkingDot, 2021-06-19:
Sorry, my posting was probably not clear.
Your fixed (linked) version of fw_upgrade.sh already works very well.

Thank you ;-)