roleoroleo/yi-hack-Allwinner-v2 #200: YI Outdoor Camera 1080p YHS.3020 - Update do not start!

Johanson2020, 2021-06-21:
Hello,
I am in despair & need your help please.
I have the same problem, that the camera ignore the mod. firmware on the SD-Card
Firmware: 9.0.26.07_202104231123
FCC ID: 2AFIB-YHS3020
DID: IFUSY

what can I do to narrow down the problem?
Sorry but i have no idea...

Thanks a lot for your support!

[image]

[image]

Johanson2020, 2021-06-21:
[image]

my mistake...just decompress..

closed 2021-06-21 by Johanson2020

roleoroleo, 2021-06-22:
Too many folders in your card.
Please remove all files and try again.
You should find in your card only these:
[image]