roleoroleo/yi-hack-Allwinner-v2 #202: Cannot compile custom firmware

davidedg, 2021-06-23:
Hi,

I'm trying to follow along [https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Build-your-own-firmware](url)
I clean installed Debian 10.10 adm64 with all the repositories enabled

On the Dependencies step, it complains about not finding a candidate for `libtidy5`
I nonetheless tried installing these `libhtml-tidy-perl libtidy-dev libtidy5deb1`

On the compile step, it complains with:
> checking whether the arm-openwrt-linux-gcc linker (/opt/yi/toolchain-sunxi-musl/toolchain/arm-openwrt-linux-muslgnueabi/bin/ld) supports shared libraries... yes
> checking whether -lc should be explicitly linked in... no
> checking dynamic linker characteristics... GNU/Linux ld.so
> checking how to hardcode library paths into programs... immediate
> checking whether stripping libraries is possible... yes
> checking if libtool supports shared libraries... yes
> checking whether to build shared libraries... yes
> checking whether to build static libraries... yes
> checking for bison version... 1.21.9-1
> configure: error: **You need bison version 3.0 or greater**, or use --disable-maintainer-mode.

My main goals are:
- compile curl (I could not find a working one for the allwinner-v2, only mstar or allwinner(v1) )
- get the gdrive uploader work (hence the curl goal)
- being able to compile any further binaries I might need, namely megacmd ...
- tinker around :D

PS: is there a Discord channel ?
Bye and Thx!
Davide.

davidedg, 2021-06-23:
First 2 goals reached (via a [workaround](https://github.com/moparisthebest/static-curl) ;P)

roleoroleo, 2021-07-02:
Sorru for late reply.

> checking whether the arm-openwrt-linux-gcc linker (/opt/yi/toolchain-sunxi-musl/toolchain/arm-openwrt-linux-muslgnueabi/bin/ld) supports shared libraries... yes
> checking whether -lc should be [...]

Try to install bison/flex on your host.

> My main goals are:
> 
>     * compile curl (I could not find a working one for the allwinner-v2, only mstar or allwinner(v1) )
> 
>     * get the gdrive uploader work (hence the curl goal)
> 
>     * being able to compile any further binaries I might need, namely megacmd ...
> 
>     * tinker around :D

curl binary is the same as allwinner (v1)
You can find it here:
https://github.com/roleoroleo/yi-hack-utils

If you want to run gdrive, this repo should work:
https://github.com/roleoroleo/yi-hack-Allwinner.gdrive

> PS: is there a Discord channel ?

No, sorry.

davidedg, 2021-07-02:
> Try to install bison/flex on your host.

`apt install bison` did the job, but only to break somewhat further:
`Makefile.am:36: error: Libtool library used but 'LIBTOOL' is undefined`

which can be fixed by `apt install libtool`

May you fix the wiki (just found out that github does not support pull requests for wiki... now, that's a shame!) ?

> sudo apt-get install git unzip build-essential **libtool** **bison** bison++ bisonc++ libbison-dev ....

> curl binary is the same as allwinner (v1)
> https://github.com/roleoroleo/yi-hack-utils

I sincerely do not know how I could not find it before. Will try it (yet that static binary works anyway ;D)

> https://github.com/roleoroleo/yi-hack-Allwinner.gdrive

Yep, I am using it already (with static curl)

roleoroleo, 2021-07-02:
Wiki fixed.

d0m1n1qu3, 2022-01-25:
I just install a fresh Debian10 in a VM (VirtualBox) and try the wiki howto .. but failed with the first line, because bison and bison++ can not installed both at the same time ..

```
sudo apt-get install git unzip build-essential libtool bison bison++ bisonc++ libbison-dev autoconf autotools-dev automake libssl-dev zlibc zlib1g-dev libzzip-dev flex libfl-dev yui-compressor closure-compiler optipng jpegoptim libtidy5 node-less sassc sass-spec libhtml-tidy-perl libxml2-utils rsync
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libtidy5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libtidy5' has no installation candidate
```

than i used libtidy5deb1 ....

```
sudo apt-get install git unzip build-essential libtool bison bison++ bisonc++ libbison-dev autoconf autotools-dev automake libssl-dev zlibc zlib1g-dev libzzip-dev flex libfl-dev yui-compressor closure-compiler optipng jpegoptim libtidy5deb1 node-less sassc sass-spec libhtml-tidy-perl libxml2-utils rsync
Reading package lists... Done
Building dependency tree       
Reading state information... Done
unzip is already the newest version (6.0-23+deb10u2).
unzip set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 bison++ : Conflicts: bison but 2:3.3.2.dfsg-1 is to be installed
           Recommends: flex-old but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

may be it works without bison++ ?

`

sudo apt-get install git unzip build-essential libtool bison bisonc++ libbison-dev autoconf autotools-dev automake libssl-dev zlibc zlib1g-dev libzzip-dev flex libfl-dev yui-compressor closure-compiler optipng jpegoptim libtidy5deb1 node-less sassc sass-spec libhtml-tidy-perl libxml2-utils rsync`

but than the compile.sh fails while alsa

```
...
checking whether make sets $(MAKE)... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking whether make supports nested variables... yes
checking for arm-openwrt-linux-gcc... arm-openwrt-linux-gcc
checking whether the C compiler works... no
configure: error: in `/home/dominique/yi/yi-hack-Allwinner-v2/src/alsa-lib/alsa-lib-1.1.4.1':
configure: error: C compiler cannot create executables
See `config.log' for more details
```

can you help me with setting up a dev env .. :-)

d0m1n1qu3, 2022-01-25:
ok the last problem i could solve with resetup a fresh debian 10 as 64bit VM .. the other points like libtidy5 and bison++ still exist with 64bit .. but my described workarounds works fine .. so maybe we should correct the wiki ;-)

roleoroleo, 2022-01-26:
I don't understand the workaround you used.
Please, help me to understand.

d0m1n1qu3, 2022-01-26:
sorry for the confusion :)
it is pretty simple .. in the wiki "Get dependencies" just change it to:
```
sudo apt-get update
sudo apt-get install git unzip build-essential libtool bison bisonc++ libbison-dev autoconf autotools-dev automake libssl-dev zlibc zlib1g-dev libzzip-dev flex libfl-dev yui-compressor closure-compiler optipng jpegoptim libtidy5deb1 node-less sassc sass-spec libhtml-tidy-perl libxml2-utils rsync
```

and the hint that is should be a Debian 10 64bit would be nice ;-)

closed 2023-01-08 by stale[bot]