roleoroleo/yi-hack-Allwinner-v2 #203: Motion Events - time directory name wrong

SnOOpeRa, 2021-06-24:
Hi, with the new update 0.18 (model y211ga ) the time directory names are wrong , the time set in configuration is right and in the status page is showing the right local time utc+3 (summer time) , but in Motion Events the directory names are created with current time + 3h , e.g. current time is Thu Jun 24 12:57:25 EEST 2021 in status page , but in Motion Events the directory created are:

Date: 2021-06-24 Time: 15:00 - 2021Y06M24D15H

Ansuo, 2021-06-24:
Here is a similar problem.

Dir time is right.
The sytem creation time of dirs and videos has +1 hour.

davidedg, 2021-06-24:
I confirm the behaviour - I rolled back to not using a local time

roleoroleo, 2021-07-02:
Did you noticed this issue only with 0.1.8?

SnOOpeRa, 2021-07-03:
I can tell only for v 0.1.8 as I just receive 2 cameras y211ga and flash them with the latest firmware, will try to test with older 0.1.7 or so.

roleoroleo, 2021-07-04:
Please take a snapshot of the motion page and the output of ls command.

0-Al, 2021-09-14:
FWIW I had differences in time and fixed it by using the app to set UTC0 (GMT) and set the real time zone in Yi-Hack otherwise it's added twice.

roleoroleo, 2021-09-14:
> FWIW I had differences in time and fixed it by using the app to set UTC0 (GMT) and set the real time zone in Yi-Hack otherwise it's added twice.

Thank you for this hint. I have been looking for the origin of GMT offsets for a long time.

0-Al, 2021-09-16:
Don't know if it's any help but time offset on my firmware appears to be at 0x0x7F012c, 0x12c in "conf" IIRC. Time zone offset from UTC in seconds.

closed 2023-01-08 by stale[bot]

chrbar11, 2023-04-16:
> FWIW I had differences in time and fixed it by using the app to set UTC0 (GMT) and set the real time zone in Yi-Hack otherwise it's added twice.

Hi ghost, I did the same (set "Coordinated universal time" on Yi App, and "EST5EDT,M3.2.0,M11.1.0" on Yi-Hack), but the built-in Yi timestamp (at left top corner) is still wrong.