roleoroleo/yi-hack-Allwinner-v2 #207: Changing Configuration seems to be ignored

jamosch, 2021-06-26:
Hey Folks,
i've installed yi-hack-allwinner-v2 on my new Yi 1080p Dome.
I can access the Webinterface via http://CAM-IP:8080 - the RTSP Stream is working also in VLC.

When i try to make changes in the Webinterface e.g. setting the option "Disable cloud" - after a reboot the option is greyed out again. The same behaviour for the toggles SSH, ONVIF or NTP Server.

I change the toggle or value, submit by clicking "save configuration" and rebooting the camera afterwards.

Is this a bug or am i doing it wrong.

Best regards

---

Serialnumber: starts with IFUSY
Firmwareversion: 0.1.8
Baseversion: 9.0.05.33_202009251856
Modellsuffix: r30gb

roleoroleo, 2021-07-03:
This is not the normal behavior.
But too big to be a bug.
Please clear the cache of the browser and try again.

jamosch, 2021-07-10:
You were absolutetly right. 
I tried another Browser (Edge instead of Brave) and it worked like a charm.
Sorry for that non existing issue!

closed 2021-07-10 by jamosch