roleoroleo/yi-hack-Allwinner-v2 #208: KAMI WIRE-FREE OUTDOOR CAMERA - is this camera hackable?

gabaszo, 2021-06-27:
Hello, 

I have bought this camera : https://kamihome.com/kami-wirefree
Is it possible to hack it and allow rtsp and other stuff?
cam Id is: BFUSY273TCE1MR200102
FW ver: 5.1.0.0A_202105261538

thx for any advice

roleoroleo, 2021-07-03:
At the moment it's not supported.
Check if you are able to make a backup (see the wiki) and I will take a look.

gabaszo, 2021-07-03:
roleo - which country are you in? 
Maybe I can send one of this cam to you?

roleoroleo, 2021-07-03:
Italy.

gabaszo, 2021-07-19:
unfortunately, this backup script is not working. nothing is created on SD
;(
any other idea?

roleoroleo, 2021-07-20:
The last idea is to open the cam and take some pictures.

closed 2023-01-08 by stale[bot]

RebeliusGaming, 2024-06-21:
Hi roleo. I own one of this cameras. Would you like for me to send you the information?
If so, what would you need?

RebeliusGaming, 2024-06-22:
[wire free yi cam.zip](https://github.com/user-attachments/files/15936202/wire.free.yi.cam.zip)
This are some photos of me trying to unmount. Coulndt go further. I see no more screws, and nothing moves further. Any suggestions?

RebeliusGaming, 2024-06-22:
[image]
[image]
[image]
[image]
[image]

RebeliusGaming, 2024-06-28:
@roleoroleo just in case you have this turned off

roleoroleo, 2024-06-29:
I have never seen a camera like this one.
But without a picture of the pcb I'm not able to understand if I can support it.

RebeliusGaming, 2024-06-29:
Do you have any idea on how to open it without damaging?

erela, 2024-08-29:
Hi guys,

I think the question about the platform is answered here: https://github.com/roleoroleo/yi-hack-MStar/issues/281
Seams it's not supported for a hack for now.

RebeliusGaming, 2024-09-01:
Yes, It looks like it.
Sad news. Thank you for the heads up

roleoroleo, 2026-01-02:
Try to make a backup and I will check if they can be supported.