roleoroleo/yi-hack-Allwinner-v2 #209: Dome U Pro - Sensor camera AI Humand Detection

SmartM-ui, 2021-06-27:
Hi,
would it be possible to display the photograph also for the AI Human Detection sensor when a human movement is detected as it already happens for the motion detection sensor (camera.yi_camera_motion_detection_image)?

Thank you

roleoroleo, 2021-07-03:
At the moment it's not supported.
It requires a change in mqttv4 project.

SmartM-ui, 2021-07-05:
> At the moment it's not supported.
> It requires a change in mqttv4 project.

Thanks anyway

roleoroleo, 2021-07-05:
I will add it to my todo list.

SmartM-ui, 2021-09-02:
Hi @roleoroleo 
in these days of vacation I was able to monitor well the motion detection sensor and the AI human detection sensor.

By leaving only motion detection active, I often had alerts (both from the official app and from the Home Assistant notification on the binary_sensor.yi_camera_movement sensor) about motion, for example when an insect passed the camera.

I activated the AI Human Detect sensor from your web page and I noticed that the official YIhome app no longer sent notifications relating to insect movements, while the binary_sensor.yi_camera_human_detection sensor was also activated by recording the video and sending me the notification that the sensor had passed to the ON state.

Would it be possible to match the binary_sensor.yi_camera_human_detection sensor to the real Human Detection sensor of the YIhome app so as to detect only human movement?

Another request, since the sensor is already present in YI-HACK, can you insert it as a switch in Home Assistant, perhaps using the same system provided for the other switches? 

[image]

THANKS

roleoroleo, 2021-09-03:
> I activated the AI Human Detect sensor from your web page and I noticed that the official YIhome app no longer sent notifications relating to insect movements, while the [...]

I need to analyze this behavior.

SmartM-ui, 2021-09-03:
> I need to analyze this behavior.

Hi @roleoroleo 

I did some tests both last night and now and this is the result with AI_HUMAN_DETECT ON:

When I pass in front of the camera (therefore human detection)
MQTT explorer: I can see that the human_start command is sent and immediately afterwards a series of human_stops are sent:
[image]
[image]
[image]
[image]

Home Assistant: I can see that the binarysensor activates for a second and then automatically turns off again.

[image]

On the official YIhome app I receive the notification with video of the detection (OK)
On Home Assistant I get the Human Detect notification (OK)

When I turn on a light or an insect passes in front of the camera (so it shouldn't detect human movement):
MQTT explorer: I can see that the human_start command is sent and the human_stop command is NEVER sent.
In this case the human_start command should not be sent, so it is right that the human_stop command is not sent:

[image]

Home Assistant: I can see that the binarysensor activates and is never turned off:

[image]

On the official YIhome app I do not receive the notification with video of the detection (OK)
On Home Assistant I receive the Human Detect notification (NO)

If I can debug the check on AI_HUMAN_DETECTION or if I can help you in any way, please ask 

Thanks

roleoroleo, 2021-09-10:
Today I tried for 2 hours but I couldn't get your behavior. When I activate human detection, my cam no longer sends me human_start.
Could you send me a screenshot of the app's detection settings?

roleoroleo, 2021-09-22:
Please, could you test again human detection with retain=0?

roleoroleo, 2021-09-24:
I completely rewritten the human detection.
I think there was an error in the reverse of the cam behavior.
I removed start and stop message, now there is only a detection message (like sound).

@SmartM-ui 
Do you want to try a beta? h60ga?

SmartM-ui, 2021-09-24:
> I completely rewritten the human detection.
> I think there was an error in the reverse of the cam behavior.
> I removed start and stop messages, now there is only a detection message (like sound).
> 
> @SmartM-ui
> Do you want to try a beta? h60ga?

Hi @roleoroleo 
I would love to try and wait for this update, but as I told you I brick the Dome U Pro trying to remove your YI-HACK to update the IPcam firmware. For the moment I don't think about shooting another YI.

Thanks again for all your work and for putting my requests in your firmware. 
I enjoyed "playing" with you :-)

roleoroleo, 2021-09-25:
Do you want to try to unbrick it?
You need only an usb-serial adapter.

SmartM-ui, 2021-09-25:
> Do you want to try to unbrick it?
> You need only an usb-serial adapter.

Unfortunately I don't have a serial adapter.
I preferred to return the cam.
Thanks again for everything!

cookienet, 2021-09-27:
> I completely rewritten the human detection.
> I think there was an error in the reverse of the cam behavior.
> I removed start and stop messages, now there is only a detection message (like sound).
> 
> @SmartM-ui
> Do you want to try a beta? h60ga?

I would. What do I have to do?

roleoroleo, 2021-09-27:
I'm working on it.
I will share with you a beta asap.
Which model?

roleoroleo, 2021-09-27:
After so many attempts I think I understand why I can't get human detection to work.
I believe the algorithm is run server side by Yi's servers. 
Because if I remove the network connection between the cam and the cloud, the detection doesn't work.
So I can't get it to work.

roleoroleo, 2021-09-28:
Yes.
If the cam is able to connet to the server, I can see ipc messages between processes when a human is detected.
If I ban connections to api.eu.xiaoyi.com, I can't see them anymore.
Reading the ipc messages it seems that:

- The cam detects a motion
- The cam captures a snapshot and sends to the cloud
- The cloud reads the image
- The cloud sends the human detection event (if true)

closed 2023-01-08 by stale[bot]