roleoroleo/yi-hack-Allwinner-v2 #210: Dome U Pro - Switch / Sensor for Face Detection

SmartM-ui, 2021-06-27:
Hello,
would it be possible to provide a switch and sensor for face detection supported by the Dome U Pro? (also via MQTT)

Thank you

roleoroleo, 2021-07-03:
My cam doesn't support face detection and I can't develop it at the moment.

SmartM-ui, 2021-07-05:
> My cam doesn't support face detection and I can't develop it at the moment.

If you want I can volunteer for tests :-)

I'm at your service!
Thanks anyway

roleoroleo, 2021-07-05:
I need to know if the app sends a message to the cam when you toggle the switches on/off.
Use ipc_sniff
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/168#issuecomment-862667876
Run it manually and check if you can see a numeric code when you toggle the switch.

SmartM-ui, 2021-07-06:
> I need to know if the app sends a message to the cam when you toggle the switches on/off.
> Use ipc_sniff
> [#168 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/168#issuecomment-862667876)
> Run it manually and check if you can see a numeric code when you toggle the switch.

Sorry, how do you use ipc_sniff? Where should I run it from?

roleoroleo, 2021-07-06:
You can copy it to your sd card /tmp/sd using ftp after decompressing it.
Then login with ssh and run it manually /tmp/sd/ipc_sniff

SmartM-ui, 2021-07-06:
Hi @roleoroleo 
I copied the ipc_sniff file to the microSD (via USB), creating the tmp / sd folder, but when I try to start ipc_sniff via the mac terminal, it can't find the file.
Where am I wrong?

[image]

[image]

WalkingDot, 2021-07-07:
@SmartM-ui

WalkingDot, 2021-07-07:
@SmartM-ui

SmartM-ui, 2021-07-07:
> @SmartM-ui
> The "tmp" folder is only available if the cam is running.
> If you copy it via usb, you can place ipc_sniff into the root of your sd-card.
> Than with ssh you can find the file here: /tmp/sd/ipc_sniff

Hi,
I put the file directly into the microSD root, but when I launch it with / tmp / sd / ipc_sniff, it reports "Segmentation fault".

For info, I have formatted the microSD directly from the cam, the macbook does not read it and the partition with Windows detects that there is an error in the microSD, but I can still use it and write. I have tried several microSDs and they all have the same behavior.

PS can you recommend a free FTP software for MacOS?

[image]

WalkingDot, 2021-07-07:
> it reports "Segmentation fault".

Then try the following:
1. `killall ipc_multiplexer`
2. `/tmp/sd/yi-hack/bin/ipc_multiplexer &`

Than toggle the app settings how roleoroleo has described.
**When you are done you should restart the camera.**

> formatted the microSD directly from the cam

Maybe because the cam is formating at sector 2. My linux machine has no problems with it.

> FTP software for MacOS?

I don't have a mac but I like filezilla and it is also available for mac. (filezilla-project.org)

SmartM-ui, 2021-07-07:
> Then try the following:
>
>     1. `killall ipc_multiplexer`
>
>     2. `/tmp/sd/yi-hack/bin/ipc_multiplexer &`
>
>
> Than toggle the app settings how roleoroleo has described.
> **When you are done you should [...]

Perfect,
it worked! After I try to check the settings and bring them back to Roleoroleo

[image]

SmartM-ui, 2021-07-07:
Hi @roleoroleo 
this is the data that is detected when I press the Home and Not_Home button from the YIhome app 

[image]

Not_Home:
IPC message. Len: 20. Status: No error information!
Parsing message.
01 00 00 00 04 00 00 00 8e 00 01 00 04 00 00 00 00 dd 6d 00 

Home:
IPC message. Len: 20. Status: No error information!
Parsing message.
01 00 00 00 04 00 00 00 8e 00 01 00 04 00 00 00 00 dd 6d 00 

I hope it can be used for the switch and synchronization with Home Assistant

SmartM-ui, 2021-07-07:
Hi @roleoroleo 
this is the data that is detected when I press the face detection from the YIhome app:

Face detect ON
IPC message. Len: 20. Status: No error information!
Parsing message.
01 00 00 00 08 00 00 00 3d 10 01 00 04 00 00 00 01 00 00 00 

Face detect OFF
IPC message. Len: 20. Status: No error information!
Parsing message.
01 00 00 00 08 00 00 00 3d 10 01 00 04 00 00 00 00 00 00 00

roleoroleo, 2021-07-07:
Ok, now try to send these messages.
1 - Save the binary code (20 bytes) into a file (use hxd if you are a windows user).
2 - Use the command `ipc_cmd -x file.bin`
3 - Refresh the app and check if the switch is changed.

SmartM-ui, 2021-07-08:
Hi,
this code turns face detection on and off from the YIhome app 

Face detect ON
01 00 00 00 08 00 00 00 3d 10 01 00 04 00 00 00 01 00 00 00

Face detect OFF
01 00 00 00 08 00 00 00 3d 10 01 00 04 00 00 00 00 00 00 00

roleoroleo, 2021-08-06:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ccea78899b1e781771222a0975f9cf3fcfcb6593

SmartM-ui, 2021-09-02:
Hi @roleoroleo 
the Face Detection sensor, after the firmware update to version 0.1.9, manages to perfectly enable the sensor on the official YIHome app.

You are GREAT!

Would it be possible to implement the motion tracking switch as you did for the other sensors via MQTT?

[image]

roleoroleo, 2021-09-02:
> Would it be possible to implement the motion tracking switch as you did for the other sensors via MQTT?

I didn't implement the advertise part. It was a contribution from EnricoFerro.
Do you want to try to make a pr?

SmartM-ui, 2021-09-02:
> > Would it be possible to implement the motion tracking switch as you did for the other sensors via MQTT?
> 
> I didn't implement the advertise part. It was a contribution from EnricoFerro.
> Do you want to try to make a pr?

If you show me how to do it, gladly!
We can also ask @EnricoFerro for support?

closed 2023-01-08 by stale[bot]