roleoroleo/yi-hack-Allwinner-v2 #216: ONVIF not working on Yi Home IFUS 9.0.36

namankumar, 2021-07-01:
amazing work here roleoroleo, want to report an issue that ONVIF is not working on Yi Home IFUS 9.0.36 using y21ga. RTSP HD and SD work just fine.

namankumar, 2021-07-03:
Update: I was able to get ONVIF working using the command in https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/179#issuecomment-858400828

This also reduces latency of the video.

@roleoroleo...now video looks same as the mobile app. The question is how can I get the command to run by itself so I can avoid having to SSH into every device every time they start up

roleoroleo, 2021-07-04:
onvif must work without running it manually.
Something is wrong in the command line.

Are you using authentication?
What tcp port are you using?
What's your "ONVIF Profile" setting?
What's your "ONVIF Network Interface" setting?

namankumar, 2021-07-05:
webpage shows ONVIF is enabled but onvif did not work unless I ran the command I mentioned in the previous comment.

Are you using authentication?
> no, default setting
What tcp port are you using?
> default, so 80
What's your "ONVIF Profile" setting?
>default i believe, currently set to HIGH
What's your "ONVIF Network Interface" setting?
>Wireless (and I am using via local IP obv)

Also see that system and network info on the /index.html are missing if that helps diagnose the issue:
[image]

roleoroleo, 2021-07-05:
If the web interface doesn't work properly, probably there is another problem.
We need to investigate.
Please try to run these commands:
```
ps
lsmod
ls -laR /
```

namankumar, 2021-07-05:
```
root@yi-eaad:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  210 root         0 SW   [oom_reaper]
  211 root         0 SW<  [writeback]
  213 root         0 SW   [kcompactd0]
  214 root         0 SW<  [crypto]
  215 root         0 SW<  [bioset]
  217 root         0 SW<  [kblockd]
  258 root         0 SW   [sys_user]
  266 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  287 root         0 SW<  [spi0]
  297 root         0 SW   [kswapd0]
  380 root         0 SW<  [bioset]
  385 root         0 SW<  [bioset]
  390 root         0 SW<  [bioset]
  395 root         0 SW<  [bioset]
  400 root         0 SW<  [bioset]
  405 root         0 SW<  [bioset]
  410 root         0 SW<  [bioset]
  415 root         0 SW<  [bioset]
  420 root         0 SW<  [bioset]
  448 root         0 SW   [irq/302-sunxi-m]
  450 root         0 SW   [irq/166-sdc0 cd]
  451 root         0 SW   [irq/303-sunxi-m]
  524 root         0 SW<  [kworker/0:1H]
  532 root         0 SW<  [bioset]
  533 root         0 SW   [mmcqd/0]
  539 root         0 SWN  [jffs2_gcd_mtd4]
  545 root       980 S    -/bin/sh
  578 root       980 S    /usr/sbin/telnetd
  590 root         0 SW<  [phy0-atbm_wq]
  591 root         0 SW   [phy0-usb_atbm_b]
  634 root      1096 S    ./dispatch
 1133 root     41220 S    ./rmm
 1214 root      1432 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
 1222 root      3008 S    ./mp4record
 1223 root      1148 S    ./cloud
 1224 root      4008 S    ./p2p_tnp
 1225 root      3812 S    ./oss
 1227 root       676 S    ./watch_process
 1238 root       860 S    httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.
 1250 root       844 S    pure-ftpd (SERVER)
 1265 root       876 S    dropbear -R -B
 1267 root       668 S    ipc_multiplexer
 1292 root       980 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -
 1379 root       860 S    ntpd -p pool.ntp.org
 1414 root      5524 S<   rRTSPServer
 1446 root       988 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 1504 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --
 1505 root      2420 S    framefinder y211ga
 1514 root       860 S    /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/cr
16468 root         0 SW   [kworker/0:1]
16855 root         0 SW   [kworker/u2:0]
17480 root         0 SW   [kworker/0:2]
18558 root         0 SW   [kworker/u2:1]
20900 root       900 R    dropbear -R -B
20901 root       980 S    -ash
20934 root       980 S    sleep 10
20939 root       980 R    ps

```
```

root@yi-eaad:~# lsmod
Module                  Size  Used by
pid_list                1389  1 
ssp_ms41909             4184  2 
cpld_periph             5583  2 
sunxi_gpadc            11982  0 
vin_v4l2              106753  8 
sp2305_mipi             5720  0 
vin_io                 15334  4 vin_v4l2,sp2305_mipi
videobuf2_v4l2          7577  1 vin_v4l2
videobuf2_dma_contig     6209  1 vin_v4l2
videobuf2_memops         948  1 videobuf2_dma_contig
videobuf2_core         16939  2 vin_v4l2,videobuf2_v4l2
atbm603x_wifi_usb     368954  0 
```
```
root@yi-eaad:~# ls -laR
.:
drwxrwxr-x    2 root     root            30 Mar 13  2020 .
drwxrwxr-x   15 root     root           209 Mar 13  2020 ..
-rwxrwxr-x    1 root     root             0 Mar 13  2020 topfile
```

roleoroleo, 2021-07-05:
Please, try to run this ws:
http://IP-CAM:8080/cgi-bin/status.json

namankumar, 2021-07-05:
xxxx obfuscation by me, let me know if you need those fields

```
{
"name":"yi-hack-allwinner-v2",
"hostname":"yi-eaad",
"fw_version":"0.1.8",
"home_version":"9.0.36.00_202011101611",
"model_suffix":"y211ga",
"ptz":"no",
"serial_number":"

FFFFFFFFFFFFFFFF",
"local_time":"Mon Jul  5 21:42:30 GMT 2021",
"uptime":"29001.59",
"load_avg":"2.41 1.77 1.71",
"total_memory":"60912",
"free_memory":"33636",
"free_sd":"4%",
"local_ip":"192.168.1.133",
"netmask":"255.255.255.0",
"gateway":"192.168.1.xxx",
"mac_addr":"B4:FB:xx:xx:xx:xx",
"wlan_essid":"xxxxx",
"wlan_strength":"51" 
}
```

roleoroleo, 2021-07-06:
There is a problem with serial number detection.
And probably this issue breaks onvif command line.
I need a copy of your /tmp/mmap.info

namankumar, 2021-07-06:
Don't know how to download the file, so here are the copy pasted contents. 

Two questions for knowledge:
-there is no scp so how do you download the file wirelessly?
-how do I access the sd card via command line? I could cp the file to sd card 

```
root@yi-eaad:/tmp# cat mmap.info 
https://api.us.xiaoyi.comhttps://api.us.xiaoyi.comhttp://log.us.xiaoyi.comfamilymonitor-y211gamiusR?9.0.36.00_20201110161100000000000000000000000000000000TNPUSAM-753076-VTHVZIFUSY65B0O7SY721011300US00000000000000004MOMr5OIynN4j5QY00000000FFFFFFFFFFFFFFFFCWC2Tfj5ARVyVbbMkcq5Ctdx1GQ5P4BiB4:FB:E3:FB:EA:AD84:bb:69:6f:e8:b084:BB:69:6F:E8:B0192.168.1.133192.168.1.254255.255.255.0ThePoliceSF16011231????ddgmVAfjWFYQy7Kp4gmVAfjWFYQy7Kp4@3???cӮ?`TNPUSAM-753076-VTHVZDCSGZRMMFBJPLDIEEKKKHGOLFHPOEBGJCLFNHGHGELLDIDILBJCEBAPBICBPNBPODMAPKFOMLLCKPKMKCLBLDPAMADDHOB?
??
?QP'003000415050534400128000040097fd
root@yi-eaad:/tmp# 

```

roleoroleo, 2021-07-07:
I need the file.
It's a binary file, a cat is not enough.

namankumar, 2021-07-07:
Github didn't allow me to upload that filetype so I put it on Google Drive: https://drive.google.com/file/d/1G-zzPEDI3HEzWddJGqAcXgU-cowtIgkB/view?usp=sharing

I downloaded via scp. FTP only shows "topfile", nothing else.

roleoroleo, 2021-07-07:
Edit the file /tmp/sd/yi-hack/script/system.sh at line 255 and change
`if [ "$MODEL_SUFFIX" == "y211ga" ] || [ "$MODEL_SUFFIX" == "h60ga" ]; then`
to
`if [ "$MODEL_SUFFIX" == "h60ga" ]; then`

roleoroleo, 2021-07-07:
The same for the file /tmp/sd/yi-hack/www/httpd/cgi-bin/status.json at line 15.

namankumar, 2021-07-08:
That worked! Web interface is filled with all the details now. Onvif service is being picked up by client software and is able to connect and stream video.

That said, video transmission over onvif isn't great. Using Onvif device manager, the video keeps cutting out every few seconds. Any thoughts? 

I have another Yi Home camera on which the hack worked great. The video works fine on that one.

roleoroleo, 2021-07-08:
How is the video using vlc?

namankumar, 2021-07-08:
Significantly better today although not has good as the Yi native mac app.

[image]

roleoroleo, 2021-07-09:
This is strange because the stream sent via app and via rtsp is the same.
I don't (and I can't) change the h264 compression. The stream compressed from yi software is "copied" and forwarded to the rtsp server.

namankumar, 2021-07-09:
Could it be the resources available to the rtsp server? 
Could be that rtsp server is losing some packets  because it has a lower priority than the yi default server?

Memory, cpu cycles? Network bandwidth allocation

namankumar, 2021-07-09:
Another discovery: I don't have to change $MODEL_SUFFIX if I update the Yi firmware to Latest Version. yi-hack-Allwinner-v2 works just fine out of the box with Latest Version.

[image]

roleoroleo, 2021-07-13:
> 
> 
> Could it be the resources available to the rtsp server?
> Could be that rtsp server is losing some packets because it has a lower priority than the yi default server?
> 
> Memory, cpu cycles? Network bandwidth allocation

Open the window "Statistics" and check it you have lost frames.

[image]

closed 2023-01-08 by stale[bot]