roleoroleo/yi-hack-Allwinner-v2 #218: No records under "Events"

LotharF, 2021-07-02:
On my 2 YiHome1080p I have installed the version V0.1.7 and V0.1.8. 
All was running. Since 2 days when I open a directory under "Motion Events" there are no videos in the folder. 
I see only "No events in this folder". 
BUT when I look through the camera APP, I can see the recording. And on the SD card I have also the recording folder with videos.
How can i fix this "problem"?

roleoroleo, 2021-07-03:
Which suffix?

LotharF, 2021-07-03:
Both cameras are "Yi 1080p Home IFUS" with sw V9.0.36 and I use the "y211ga". Ore what do you mean with suffix?

roleoroleo, 2021-07-04:
Login with ssh and print the content of the folder /tmp/sd/record

LotharF, 2021-07-05:
in the folder I have:

Factory.done
newbackup
System Volume Infomation
record
log
yi-hack
lower_half_ini.sh

and if I open "record" I can see recorded files like "2021Y07M04D16H"

LotharF, 2021-07-05:
In the folder /tmp/sd/record I have this here:

2021Y07M04D16H  2021Y07M04D20H  2021Y07M05D00H  2021Y07M05D04H  2021Y07M05D08H
2021Y07M04D17H  2021Y07M04D21H  2021Y07M05D01H  2021Y07M05D05H  2021Y07M05D09H
2021Y07M04D18H  2021Y07M04D22H  2021Y07M05D02H  2021Y07M05D06H  tmp.mp4.tmp
2021Y07M04D19H  2021Y07M04D23H  2021Y07M05D03H  2021Y07M05D07H

roleoroleo, 2021-07-05:
Ok, this is ok.
Try to run the cgi:
http://IP-CAM:8080/cgi-bin/eventsdir.sh

LotharF, 2021-07-05:
then I got this:

{"records":[
{
"datetime":"Date: 2021-07-05 Time: 15:00",
"dirname":"2021Y07M05D15H"
},
{
"datetime":"Date: 2021-07-05 Time: 14:00",
"dirname":"2021Y07M05D14H"
},
.
.
.
.
.
{
"datetime":"Date: 2021-07-05 Time: 03:00",
"dirname":"2021Y07M05D03H"
},
{
"datetime":"Date: 2021-07-05 Time: 02:00",
"dirname":"2021Y07M05D02H"
}
]}

roleoroleo, 2021-07-05:
And this is ok.
Please try to update the cache of your browser.

LotharF, 2021-07-06:
I have update the cache and also I have try a other PC and mobil phone to test it.
But still the same, no records in the web page. I have also 2 other YiHome 1080p (9FUS) cameras with the MStar hack. here I can see the recordings.
Only my two "IFUS" cameras worked perfectly for days and then suddenly no more recordings were visible.

roleoroleo, 2021-07-06:
Please open the console of the browser and check if there are errors.

LotharF, 2021-07-06:
the console of my internet browser? Maybe you can tell me what exactly I have to do to give you the right informations :-)

roleoroleo, 2021-07-06:
I can't understand...

LotharF, 2021-07-06:
I do not understand that, too. If I run "http://IP-CAM:8080/cgi-bin/eventsdir.sh" I can see the single directory names also in the wep page. But if I open the directory I see only "No events in this folder" ... but there are videos in the folder :-(

roleoroleo, 2021-07-06:
Wait...
Maybe there is a misunderstanding.
What's your scenario?
1 - When you click "Motion events" page you see directories but when you click the dir you can't see the files.
2 - When you click "Motion events" page you don't see directories.

LotharF, 2021-07-06:
1

roleoroleo, 2021-07-06:
Ok.
Login with ssh and print the content of the folder /tmp/sd/record/2021Y07M05D02H
And run this:
http://192.168.1.244:8080/cgi-bin/eventsfile.sh?dirname=2021Y07M05D02H

LotharF, 2021-07-06:
SSH =>  /tmp/sd/record/2021Y07M06D10H :

E200M00S60.mp4  E213M00S60.mp4  E225M00S60.mp4  E237M00S60.mp4  E249M00S60.mp4
E201M00S60.mp4  E214M00S60.mp4  E226M00S60.mp4  E238M00S60.mp4  E250M00S60.mp4
.
.
.
E210M00S60.mp4  E222M00S60.mp4  E234M00S60.mp4  E246M00S60.mp4  E258M00S60.mp4
E211M49S11.mp4  E223M00S60.mp4  E235M00S60.mp4  E247M00S60.mp4  E259M00S60.mp4
E212M00S60.mp4  E224M00S60.mp4  E236M00S60.mp4  E248M00S60.mp4

-----------------------------

And the result of "http://192.168.1.204:8080/cgi-bin/eventsfile.sh?dirname=2021Y07M06D10H" : 

{"date":"2021-07-06",
"records":[
]}

roleoroleo, 2021-07-06:
I already fixed this problem here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/8a6f48bb69396f7334ee1c70e70b37ab724f31ca
Are you sure you are using 0.1.8?

LotharF, 2021-07-06:
Yes, I have installed it twice from here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases

roleoroleo, 2021-07-06:
Show me the file eventsfile.sh
```
cat /tmp/sd/yi-hack/www/cgi-bin/eventsfile.sh
```

LotharF, 2021-07-06:
hee the eventsfile.sh :

#!/bin/sh

printf "Content-type: application/json\r\n\r\n"

case $QUERY_STRING in
    *[\'!\"@\#\$%^*\(\)_+.,:\;]* ) exit;;
esac

CONF="$(echo $QUERY_STRING | cut -d'=' -f1)"
VAL="$(echo $QUERY_STRING | cut -d'=' -f2)"
MODEL_SUFFIX=$(cat /tmp/sd/yi-hack/model_suffix)
if [ "$MODEL_SUFFIX" == "r40ga" ]; then
    NUMCHAR=14
else
    NUMCHAR=12
fi

if [ "$CONF" == "dirname" ]; then
     DIR=$VAL
fi

printf "{\"date\":\"${DIR:0:4}-${DIR:5:2}-${DIR:8:2}\",\n"
printf "\"records\":[\n"

COUNT=`ls -r /tmp/sd/record/$DIR | grep mp4 -c`
IDX=1
for f in `ls -r /tmp/sd/record/$DIR | grep mp4`; do
    if [ ${#f} == $NUMCHAR ]; then
        printf "{\n"
        if [ $NUMCHAR == 14 ]; then
            printf "\"%s\":\"%s\",\n" "time" "Time: ${DIR:11:2}:${f:2:2}"
        else
            printf "\"%s\":\"%s\",\n" "time" "Time: ${DIR:11:2}:${f:0:2}"
        fi
        printf "\"%s\":\"%s\"\n" "filename" "$f"
        if [ "$IDX" == "$COUNT" ]; then
            printf "}\n"
        else
            printf "},\n"
        fi
        IDX=$(($IDX+1))
    fi
done

printf "]}\n"

LotharF, 2021-07-09:
And did you find the fault?

roleoroleo, 2021-07-09:
Tested on my cam, it works perfectly.
I don't know.

Try this command:
`cat /tmp/sd/yi-hack/model_suffix`

LotharF, 2021-07-09:
result: y211ga

LotharF, 2021-07-09:
result: y211ga

LotharF, 2021-07-09:
On my "Yi 1080p Home IFUS" I have the SW version 9.0.36.02_202105081647
I use the kami APP and I don't know exactly, but there was an upgrade via this APP and I think it has not worked since then.
But I'm not sure.

LotharF, 2021-07-09:
... and your mp4 recording (the structure) is just like mine? => E257M00S60.mp4

roleoroleo, 2021-07-09:
I manually created an identical structure.
But try this change on line 12.
From
`if [ "$MODEL_SUFFIX" == "r40ga" ]; then`
to
`if [ "$MODEL_SUFFIX" == "r40ga" ] || [ "$MODEL_SUFFIX" == "y211ga" ]; then`

LotharF, 2021-07-09:
Perfect this is working :-) 
So 14 was the magic number. Now all entries are back. Thank you so much :-)

stahelp, 2021-07-16:
I had exactly the same issue and this solved my problem too.

Thank you very much

soundylinz, 2021-07-21:
Read your model:
`cat /tmp/sd/yi-hack/model_suffix`

Edit your eventsfile:
`vi /tmp/sd/yi-hack/www/cgi-bin/eventsfile.sh`

Enter ":w" and ":q" and you're saved. ;)

And you are happy, me too.

Thanks for this great info!

Quinpe, 2021-07-25:
This worked for me too, 
Outdoor 1080p IFUS (H30GA)
Thanksssss

closed 2023-01-08 by stale[bot]