roleoroleo/yi-hack-Allwinner-v2 #220: How to improve RTSP latency, bitrate. fps?

namankumar, 2021-07-03:
I flashed a Yi home and now I can see the video but there is very high latency. How can I increase the bitrate and or the fps?

Stream has high lag when viewing via VLC. Stream has normal lag viewing via the mobile app.

roleoroleo, 2021-07-03:
Sorry but it's not possible.

The streams are different.
The cam uses a closed protocol to send stream to the app
And the stream is another flow.
I don't know how to improve the latency further. I've already done everything I could.

Bitrate and fps are managed by the original yi software.
I can't change the settings of the h264 codec.

namankumar, 2021-07-05:
Alright thanks, appreciate all your hard work!

closed 2021-07-05 by namankumar