roleoroleo/yi-hack-Allwinner-v2 #221: It is not working for Yi Home 1080p IFUS 9.0.19.*

george-baci, 2021-07-03:
Hi all,
my camera is Yi Home 1080p IFUS with firmware version 9.0.19.09_202010291747. It was connected succesfully to Yi app and then I followed the steps for the yi-hack using the y21ga file. The web interface at http://IP-CAM:8080 worked fine. However, nothing else worked, except the rtsp audio stream (checked via vlc). The rtsp video streams were not working (via vlc, ffplay, etc), not even the http snapshots on the web interface (when clicking on the links led to a blank tab on the browser).

The camera is now not unusable, as I pushed the reset button on its back, before running the unhack file first. Since that moment, the light is yellow without blinking/speaking. It cannot connect to either the app or yi-hack web interface. Tried the unhack and hack again, but the files do not seem to be executed as the Folder's name remains "Factory", not "Factory.done".

Can anyone please let me know (1) what was the reason it didn't work in the first place and (2) how can I make the camera usable again?

Thanks.

roleoroleo, 2021-07-04:
Never seen this behavior.
Before to investigate fw issues, please try the following steps:
- Check if you are using the original power adapter. other users noticed problems with 3rd party usb adapter and cables.
- Try to use another sd card, or try to reformat sd card.

george-baci, 2021-07-05:
Hi @roleoroleo,

thanks for your response. I do use the original usb adapter and cable. I did try to reformat the sd card multiple times. I don't own another sd card at the moment, but I don't think it is the issue, as it worked in the first place on both cameras (I could access the web interface and use the rtsp audio stream only). It is just after a while that both turned and stucked on the yellow light without responding to anything and "Factory" folder remains the same name (not "Factory.done") when attempting unhack/hack.

I should mention that, for the second cam, I executed through the config.sh file in the Factory folder the code mentioned [here](https://github.com/roleoroleo/yi-hack-Allwinner/issues/234#issuecomment-800912848) in an attempt to see if the rtsp video streams would work on the second cam (as I suspected that Internet access was the issue for them not working). It seems to me that the second cam stuck on the yellow light moments after that execution.

Thanks again.

roleoroleo, 2021-07-05:
The code you mentioned is temporary. It doesn't change permanently the cam.
If you remove the sd it's not executed.
But if the cam doesn't rename Factory folder, probably it's bricked. Weird, only 2 users managed to brick this model.

You need to unbrick the cam using the serial connection. See the wiki.
But before to try the serial, I would try another sd.

george-baci, 2021-07-05:
Hi @roleoroleo,

I suspected that they might be bricked, once I saw the Factory folder not getting renamed. But how??

I created backup files for both cameras. I am just not too familiar with the physical connection. The cam (TX, RX, GND) should connect to usb (RX, TX, GND) in that order? Am I suppose to connect the wires on the cam using solder? Could you provide me with a ttl-rs232/usb adapter on amazon, so to buy the right one?

I will try another sd first for sure, but I still don't understand what possibly got them bricked in the first place and how this could be prevented in the future.

Thanks for all the help.

roleoroleo, 2021-07-05:
Here it is a link:
https://www.amazon.com/IZOKEE-CP2102-Converter-Adapter-Downloader/dp/B07D6LLX19/ref=sr_1_3?dchild=1&keywords=usb+uart+adapter&qid=1625514527&sr=8-3

Other users have bricked the cam, but in a different way than yours.
For example trying to hack the cam with the wrong hack.
Or extracting the content of the archive with a program that translate linux LF with windows CRLF (this breaks the shell scripts).

george-baci, 2021-07-06:
Hi @roleoroleo,

thanks for your reply. Do I connect the wires as I described in the previous post (i.e., TX->RX, RX->TX, GND->GND) and using solder on the cam's board?

I extracted the archive using winrar. I still don't get it though why the audio stream worked (while the cam was functional) and video streams didn't. Is it because Internet access was blocked on my router?

I'll let you know once I attempt with a new sd and unbricking (if needed).

Thanks again.

roleoroleo, 2021-07-06:
> thanks for your reply. Do I connect the wires as I described in the previous post (i.e., TX->RX, RX->TX, GND->GND) and using solder on the cam's board?

Yes.
Don't worry if you connect RX to RX and TX to TX. This doesn't damage your cam.

> I extracted the archive using winrar. I still don't get it though why the audio stream worked (while the cam was functional) and video streams didn't. Is it because Internet access was blocked on my router?

WinRar is a problem but I don't know why the cam worked partially.
I think internet blocked is the cause. Maybe the cam doesn't work without internet access but it can't brick it.

sibero80, 2021-12-05:
> Never seen this behavior. Before to investigate fw issues, please try the following steps:
> 
> * Check if you are using the original power adapter. other users noticed problems with 3rd party usb adapter and cables.
> * Try to use another sd card, or try to reformat sd card.

I think this should be part of the README, this just helped me revive my Camera

closed 2023-01-08 by stale[bot]