roleoroleo/yi-hack-Allwinner-v2 #222: Mibao D700

ezruneko, 2021-07-03:
Hi,

I have a Mibao D700 and i have the dump file using the backup.tar.gz.

Is it possible to add compatibility with this amazing firmware?

Thank you

roleoroleo, 2021-07-04:
It depends on the platform.
Share your backup with me and I will take a look.

pcaro90, 2021-09-17:
I also have a Mibao D700 (which seems to be the same found in the issue #124, model q321br). Is there anything I could do to help adding support to this camera? Do you already have a backup? If not, I could send you mine.

pcaro90, 2021-09-18:
I forked the project and added preliminary support for this model. It seems to be working nicely, I just tried PTZ controls, RTSP and snapshots, and they all seem to be working. I'll test it more thoroughly over the following days.

roleoroleo, 2021-09-19:
> I also have a Mibao D700 (which seems to be the same found in the issue #124, suffix q321br). Is there anything I could do to help adding support to this camera? Do you already have a backup? If not, I could send you mine.

I already have the backup file if the model is q321br_lsx
If your model is q321br without _lsx then is different.
In this case, please share your backup with me.

pcaro90, 2021-09-20:
Sure thing, there you go: https://we.tl/t-bpyL3BW3id

roleoroleo, 2021-09-20:
Thank you.
File downloaded.

roleoroleo, 2021-09-20:
The exact model is q321br_lsx
Check backup/init.sh

pcaro90, 2021-09-20:
You're right, my bad; I saw the "q321br" string (without "_lsx") somewhere inside the backup files and assumed that was it.

ezruneko, 2021-11-19:
i've try to dismount my camera to check the chip but not luck and i've tried to install qg311r hack but any port was opened and the cam was soft-bricked, i've extract the sd and after 2 restart i've recovered it.

Did you need my backup? i've shared with you and screenshot of my research.

[image]
[image]

roleoroleo, 2021-11-19:
If you share with me your backup, I can try to add the support for this cam.
But I need a backup before you apply the hack.

ezruneko, 2021-11-19:
could you send me an email (my email in my profile) to send you the link to download the firmware please?
Thank you

roleoroleo, 2021-11-19:
Check the email address in the paypal link.

ezruneko, 2021-11-20:
i can't find it. i've uploaded to mega. https://mega.nz/file/75tlwAiR#h3tJuPyeliHh1SThgGX7KpmmkoRYz6IAA7ODwcSFjig

roleoroleo, 2021-11-20:
Please, try this beta:
[q321br_lsx_0.1.9.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7574244/q321br_lsx_0.1.9.tar.gz)

Could you please confirm that it's a ptz cam and the resolution is 3k?

ezruneko, 2021-11-20:
The cam is this https://www.amazon.es/gp/product/B08FCDJDZ9/ref=ppx_yo_dt_b_asin_image_o00_s00?ie=UTF8&psc=1

And yes it's a ptz cam and the resolution in the specs is 1296x 2304 (1296P).

Today asap i will test the beta and i will write my feedback. thank you!

ezruneko, 2021-11-20:
bricked, i've extracted the sd and the cam doesn't started to the default firmware.

roleoroleo, 2021-11-20:
Are you able to get a log with a serial connection?

ezruneko, 2021-11-20:
no, it's complicated to dismount the case to me and i haven't got a serial cable to do it.

roleoroleo, 2021-11-22:
Please try this binary.
Clear your SD and unzip the archive.
[backup_q321br_lsx.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7582853/backup_q321br_lsx.gz)

ezruneko, 2021-11-22:
i've tried it but it's bricked :( no motion or sound when i've started. thank you to try it with me

roleoroleo, 2021-11-22:
Weird. This solution should work.
Is the led on?
Is the power adapter/cable ok?

roleoroleo, 2021-11-22:
If led and power are ok, please another try:
[backup_q321br_lsx.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7583377/backup_q321br_lsx.gz)

ezruneko, 2021-11-23:
Yes, it's connected and the cable and charger is OK. I've supposed that not run because after the brick i've extracted the micro sd and format in the pc in fat32 to install in a rpi. last time when i've installed the firmware, i've needed to format the microsd in the cam with the yi iot app.

closed 2023-01-08 by stale[bot]

mimogh, 2023-06-09:
Hi 👋 i need firmware caméra mibao d600 please 🥺

mimogh, 2023-06-09:
Hi 👋 i need firmware caméra mibao d600 please