roleoroleo/yi-hack-Allwinner-v2 #224: RTSP only when wifi is enabled. No RTSP by ethernet.

Camaca71, 2021-07-07:
Hi, I am experiencing with a very odd behaviour.
RTSP works fine only when I connect the camera to wifi (by using the "Network" section of index.html page of webserver.
If I disable wifi the RTSP does not work. 
Thank you for you help

roleoroleo, 2021-07-08:
What's your model?

Camaca71, 2021-07-08:
Yi Dome Guard. 
Hack version r30gb.

roleoroleo, 2021-07-08:
What client are you using?

Camaca71, 2021-07-08:
VLC Media Player
Moreover the video is not visible on webserver too (localhost:8080/ptz)

Camaca71, 2021-07-08:
Note: after a few checks, I notice that the audio stream is available and running on VLC.

Camaca71, 2021-07-12:
Does anyone have any suggestion?

roleoroleo, 2021-07-13:
> VLC Media Player
> Moreover the video is not visible on webserver too (localhost:8080/ptz)

PTZ page doesn't show a stream, but a low rate sequence of snapshot.

> Does anyone have any suggestion?

No, sorry.

closed 2023-01-08 by stale[bot]