roleoroleo/yi-hack-Allwinner-v2 #225: Dome U Pro - Motion Tracking Sensor

SmartM-ui, 2021-07-08:
Hi @roleoroleo 
could you implement the motion tracking sensor present on the Dome U Pro? 

This is the tested and working code:

Motion Tracking ON
01 00 00 00 08 00 00 00 0b 40 01 00 04 00 00 00 01 00 00 00

Motion Tracking OFF
01 00 00 00 08 00 00 00 0b 40 01 00 04 00 00 00 00 00 00 00

Consequently I open the request for the switch on Home Assistant :-) 

THANK YOU

roleoroleo, 2021-08-06:
I added the face detection command to ipc_cmd (https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ccea78899b1e781771222a0975f9cf3fcfcb6593) and now I'm checking this issue.
Could you explain me the meaning of this feature?

SmartM-ui, 2021-08-06:
> I added the face detection command to ipc_cmd ([ccea788](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ccea78899b1e781771222a0975f9cf3fcfcb6593)) and now I'm checking this issue.
> Could you explain me the meaning of this feature?

Hi,
in the YI app there is the possibility to have a notification sent, with an image of the face only, when a face is recognized.

I am attaching screenshots of the setting on the app

[image]

roleoroleo, 2021-08-11:
There was a misunderstanding. I added the switch related to the face detection and you can try if it works adding the commit https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ccea78899b1e781771222a0975f9cf3fcfcb6593

After this, I would like you to explain to me what is "Motion tracking sensor".
My cam doesn't have this feature.

SmartM-ui, 2021-08-11:
> There was a misunderstanding. I added the switch related to the face detection and you can try if it works adding the commit [ccea788](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ccea78899b1e781771222a0975f9cf3fcfcb6593)
> 
> After this, I would like you to explain to me what is "Motion tracking sensor".
> My cam doesn't have this feature.

Hi @roleoroleo 
Can I insert the modified files in the commit directly via ftp without having to physically remove the microSD?
In which directory should I place them?

As for the motion tracking sensor, by activating this sensor, the webcam if it detects a movement, follows the subject by rotating the webcam 

Thank you

roleoroleo, 2021-08-16:
> Hi @roleoroleo
> Can I insert the modified files in the commit directly via ftp without having to physically remove the microSD?
> In which directory should I place them?

I will send you a compressed archive.

> As for the motion tracking sensor, by activating this sensor, the webcam if it detects a movement, follows the subject by rotating the webcam

Ok, thanks.

roleoroleo, 2021-08-16:
Here it is:
[patch.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6993269/patch.tar.gz)

Back up your old files.

SmartM-ui, 2021-08-17:
> Here it is:
> [patch.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6993269/patch.tar.gz)
> 
> Back up your old files.

Hi,
I tried to insert the four files you sent me, I see the possibility to activate / deactivate the face detection, but it goes into error when I try to save the configuration:

[image]

Thanks

roleoroleo, 2021-08-17:
Check the permissions of the files.
camera_settings.sh must be 755.
If you don't fix this problem, I will send a full hack image.

roleoroleo, 2021-08-17:
And when the "save" operation runs successfully, add a line to the file etc/camera.conf:
`FACE_DETECTION=no`
Otherwise the setting will not saved (but without return errors).

roleoroleo, 2021-08-18:
Motion Tracking switch added:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/3558620f7b1567a18084ff51b3fd481d9425ecb2

SmartM-ui, 2021-08-19:
> Motion Tracking switch added:
> [3558620](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/3558620f7b1567a18084ff51b3fd481d9425ecb2)

hi, 
I left for the holidays.  I try everything on my return!  
Thanks again

SmartM-ui, 2021-09-02:
Hi @roleoroleo 
I returned last night from vacation and today, the first thing I did was to update the YI-HACK firmware (excellent possibility to update directly from the web page) and update the custom on Home Assistant.
What to say? PERFECT WORK!
The Motion Tracking sensor, from your web page, perfectly enables the sensor on the official YIHome app.
Great Roleoroleo

Would it be possible to implement the motion tracking switch as you did for the other sensors via MQTT? 
[image]

roleoroleo, 2021-09-02:
> Would it be possible to implement the motion tracking switch as you did for the other sensors via MQTT?

I didn't implement the advertise part. It was a contribution from EnricoFerro.
Do you want to try to make a pr?

SmartM-ui, 2021-09-02:
> > Would it be possible to implement the motion tracking switch as you did for the other sensors via MQTT?
> 
> I didn't implement the advertise part. It was a contribution from EnricoFerro.
> Do you want to try to make a pr?

If you show me how to do it, gladly!
We can also ask @EnricoFerro for support?

closed 2023-01-08 by stale[bot]