roleoroleo/yi-hack-Allwinner-v2 #226: Disable snapshots

SmartM-ui, 2021-07-08:
Hi,
unfortunately with the Dome U Pro, even by enabling the swap file on the SD, when a movement is detected and the photograph is then taken, the camera freezes for a few seconds and then restarts, but I noticed that it remains "slow" anyway.

Is it possible to avoid taking the photograph upon detection or to have it saved only in low resolution? 

Thank you

roleoroleo, 2021-07-09:
Remove the topic motion_detection_image and the cam will not take a snapshot.

SmartM-ui, 2021-07-09:
> Remove the topic motion_detection_image and the cam will not take a snapshot.

Hi,
I try to remove it from the MQTT section of your YIHACK webpage.

[image]

However, for information, when only detection is active (disabling human detection), the live view, when it does not freeze to take the image, is slowed down and often goes forward two seconds at a time.

It is not possible to shoot it in low resolution, just to try if the problem is less obvious.
Now, perhaps, it should take 2 images at a time, one in high resolution and one in low resolution:

High res snapshot | http://IP:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
-- | --
Low res snapshot | http://IP:8080/cgi-bin/snapshot.sh?res=low&watermark=yes

roleoroleo, 2021-07-09:
If you disable the snapshot, the cam should works better.
It's not possible to change the resolution in the motion snapshot, it's hardcoded:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/mqtt/dir.patch

SmartM-ui, 2021-07-09:
> If you disable the snapshot, the cam should works better.
> It's not possible to change the resolution in the motion snapshot, it's hardcoded:
> https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/mqtt/dir.patch

I haven't tried to disable the option yet, but I have seen that it works much better, and without any freeze, by also activating the AI Human Detect from your WebPage. Obviously in this way, the snapshot is not taken.

I have to try to check if it is possible to take an image directly from Home Assistant when the detection sensor is activated.

closed 2023-01-08 by stale[bot]