roleoroleo/yi-hack-Allwinner-v2 #227: Save video when a motion is detected is not working correctly

SmartM-ui, 2021-07-09:
Hi @roleoroleo ,
I have seen that even activating the function to record video only when motion is detected, a video file is saved every minute (I also set the sensitivity level to low). it's normal? Perhaps for this I also find a slowness in live view, as I mentioned in the other topic? ( https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/226#issuecomment-877060976 )

[image]

[image]

Thanks

roleoroleo, 2021-07-09:
Did you reboot the cam?
Are you sure that there isn't a motion? For example, my cam detects a motion every minute when my phone receives a message and the led blinks (when the phone is charging it is facing the camera).

SmartM-ui, 2021-07-09:
> Did you reboot the cam?
> Are you sure that there isn't a motion? For example, my cam detects a motion every minute when my phone receives a message and the led blinks (when the phone is charging it is facing the camera).

I had restarted the camera after making some changes.

After rebooting it did not record continuously, but after the first pass in front of the cam, it started recording continuously every minute even though no one passed in front of the camera.

I try to better identify the problem. 
Sorry for all this work :-)

roleoroleo, 2021-07-09:
It could be that the cam doesn't send the motion stop event.
Similar to the problem with motion start when you enable ai human detection.

SmartM-ui, 2021-07-09:
> It could be that the cam doesn't send the motion stop event.
> Similar to the problem with motion start when you enable ai human detection.

For this purpose, as you advised me, I created two automations to stop the detection.
In this way the sensors (also AI Human detect) seem to work correctly from Home Assistant, deactivating after the set seconds. If it can be useful to someone, I share the code:

- id: 'TurnOffMotionYi'
  alias: TurnOffMotionYi
  trigger:
  - platform: state
    entity_id: binary_sensor.yi_camera_movement
    to: 'on'
    for: 00:00:15
  action:
  - service: mqtt.publish
    data:
      topic: yicam/motion_detection
      payload: motion_stop
      retain: true
  initial_state: true
  mode: single

- id: 'TurnOffHumanMotionYi'
  alias: TurnOffHumanMotionYi
  description: ''
  trigger:
  - platform: state
    entity_id: binary_sensor.yi_camera_human_detection
    to: 'on'
    for: 00:00:15
  condition: []
  action:
  - service: mqtt.publish
    data:
      topic: yicam/ai_human_detection
      payload: human_stop
      retain: true
  mode: single
  initial_state: true

SmartM-ui, 2021-07-12:
Hi @roleoroleo 
sorry for the inconvenience, but I have not found a solution to not automatically record a minute of video every minute.

I tried to deactivate the swap file, I tried to activate AI_Human, I tried to activate and deactivate the MQTT publish automations to deactivate the detection, I tried to set the timezone, I tried to deactivate and then reactivate "Save video when a motion is detected ", I tried to disable MQTT, but nothing video is automatically recorded every minute.

What other setting could I try?

roleoroleo, 2021-07-13:
Recording of video is a task of the original yi app.
I cant' change this behavior, I can only kill the process and completely disable the recording.
It's not related to MQTT or to swap file.

Try to reconfigure the app with motion detection enabled, ai and face disabled and recording only with motion detected.
And set these:
[image]
if "Disable Cloud" option is enabled.

SmartM-ui, 2021-07-13:
> recording only with motion detected.
> And set these:

Hi,
However, I have seen that the videos it saves every minute are displayed only through your web panel and not through the YIhome app (in fact there is no trace even in the time axis of the YIhome app).

I try to reconfigure the YIhome app by disabling and re-enabling the "recording only with motion detected" parameter.

I would like to leave the cloud enabled because it is convenient to receive the notification with the 6 seconds of video it records.

Thanks, keep checking

WalkingDot, 2021-07-13:
The new camera models/revisions are at the moment somehow strange...
Isn't the recording stopped by your home assistant automation action?

roleoroleo, 2021-07-14:
> The new camera models/revisions are at the moment somehow strange...
> 
> Isn't the recording stopped by your home assistant automation action?

As I wrote above, recording of video is a task of the original yi app.
I cant' change this behavior, I can only enable/disable recording killing the app.
Home assistant doesn't start or stop recording.

> Or if you sending "01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00" it should stop, right?

Yes. Or simply using `ipc_cmd -T`

Both hack and ha custom component, don't start/stop recording.
There is a web service that can do this but the hack does not use it 
`http://IP-CAM:8080/cgi-bin/record.sh`

SmartM-ui, 2021-07-14:
> The new camera models/revisions are at the moment somehow strange...
> 
> Isn't the recording stopped by your home assistant automation action?
> Or if you sending "01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00" it should stop, right?

Hi @WalkingDot @roleoroleo 
on home assistant I only created the automation to bring motion detection back to the off state.

I will try to run the command as soon as I see the continuous saving of the videos again (lasting one minute each).

For the moment the best setting seems to be that of activate the AI human detect from the YI-HACK web page (as it is also seen by the original YIhome app) without further automation to be launched from Home Assistant.
Only drawback, the AI human detect sensor on Home Assistant is active only for 1 second and from MQTT explorer I can see that the human_stop command is executed several times immediately after detection.

I continue with the tests

SmartM-ui, 2021-07-14:
> As I wrote above, recording of video is a task of the original yi app.
> I cant' change this behavior, I can only enable/disable recording killing the app.
> Home assistant doesn't start or stop [...]

I tried the command (/tmp/sd/yi-hack/bin/ipc_cmd -T), I have no errors, but the videos continue to be saved (and I have no motion notifications).

How can I see if the command actually works?

[image]

SmartM-ui, 2021-07-14:
hi @roleoroleo @WalkingDot 

Could the "1970" date affect the continuous recording of data?
The time is always two hours behind

Sometimes the date is not correct. Sending sample photos

{
"start":"1970-01-01T00:00:00+0000",
"end":"2021-07-14T14:59:47+0000",
"files":[ ]
}

{
"start":"1970-01-01T00:00:00+0000",
"end":"2021-07-14T14:53:40+0000",
"files":[ ]
}

{
"start":"2021-07-14T12:52:12+0000",
"end":"2021-07-14T12:53:12+0000",
"files":[ ]
}

[image]

[image]

WalkingDot, 2021-07-14:
> As I wrote above, recording of video is a task of the original yi app.

Sorry, my post shouldn't sound like it's caused by your hack. I'm sure it's a yi "thing"... (same with the 64mb r30gb)

> How can I see if the command actually works?

You could start ipc_sniff or ipc_multiplexer (with &) to retrieve the message/code. It should be "01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00".
It would also be interesting to know which code triggers the infinite recording.

> Could the "1970" date affect the continuous recording of data?

I don't know... it's the unix epoch (start time). Maybe a problem with the NTP server? But this is just a guess.

SmartM-ui, 2021-07-15:
> > As I wrote above, recording of video is a task of the original yi app.
> 
> Sorry, my post shouldn't sound like it's caused by your hack. I'm sure it's a yi "thing"... (same with the 64mb r30gb)
> 
> > How can I see if the command actually works?
> 
> You could start ipc_sniff or ipc_multiplexer (with &) to receive the message/code. The output should be "01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00".
> It would also be interesting to know which code triggers the infinite recording.
> 
> > Could the "1970" date affect the continuous recording of data?
> 
> I don't know... it's the unix epoch (start time). Maybe a problem with the NTP server? But this is just a guess.

Hi @WalkingDot 

I noticed that during the continuous recording, every minute a code different from the usual ones was reported:
11:59
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 bf 05 00 00 

12:00
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 dc 05 00 00 

12:01
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 08 06 00 00 

12:02
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 a6 05 00 00 

12:03
01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 f5 03 00 00 

......
..... ......

[image]

During the minute, however, the following codes were shown continuously:
01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
01 00 00 00 04 00 00 00 22 40 01 00 00 00 00 00 
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00 
01 00 00 00 02 00 00 00 f6 00 f6 00 00 00 00 00 

When I issue the command:
/tmp/sd/yi-hack/bin/ipc_cmd -T

I receive:
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00 
Sending MOTION_STOP to queue #1
Sending MOTION_STOP to queue #2
Sending MOTION_STOP to queue #3
Sending MOTION_STOP to queue #4
Sending MOTION_STOP to queue #5
Sending MOTION_STOP to queue #6
Sending MOTION_STOP to queue #7
Sending MOTION_STOP to queue #8
Sending MOTION_STOP to queue #9

To carry out the test, I set the parameters as recommended by @roleoroleo , i.e. I disabled the AI Human Detect from YI-HACK and then from the YIhome app

Thanks as always

WalkingDot, 2021-07-15:
Sorry.. I was hoping there is maybe a special start message,

The "ipc_cmd -T" stop message looks ok.

is normal on each recorded file:
`01 00 00 00 10 00 00 00 e8 00 e8 00 00 00 00 00 xx xx xx xx`

Only these two I don't know (cloud or app?):
`01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00`
`01 00 00 00 04 00 00 00 22 40 01 00 00 00 00 00`

SmartM-ui, 2021-07-15:
> Only these two I don't know (cloud or app?):
> `01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00`
> `01 00 00 00 04 00 00 00 22 40 01 00 00 00 00 00`

Sorry I did not understand.
In what sense from the app or from the cloud?
I ran the command from the MAC terminal and I had the YIhome app closed and I had opened the YI-HACK browser and maybe Home Assistant.
However, those codes come out as soon as I run the command:
/ tmp / sd / yi-hack / bin / ipc_multiplexer &
and they follow each other quickly, one code per second or perhaps even faster

[image]

It's not normal?
Can I carry out some other tests?

WalkingDot, 2021-07-16:
> and they follow each other quickly, one code per second or perhaps even faster

I'm also pretty new here. But when the codes are coming so fast, it doesn't seem normal to me.
I think @roleoroleo  can surely explain this.

SmartM-ui, 2021-07-16:
> > and they follow each other quickly, one code per second or perhaps even faster
> 
> I'm also pretty new here. But when the codes are coming so fast, it doesn't seem normal to me.
> I think @roleoroleo can surely explain this.

Hi,
I recorded a video to make it clear how fast the data arrives.

Please note that the camera switch is OFF, the YIhome app closed, the YIHACK web page closed, Home Assistat closed.
 I've tried setting both home and away mode on the YIhome app, but the result hasn't changed

We await the response of @roleoroleo 

https://user-images.githubusercontent.com/63554131/125949949-07e09767-8a14-4bf2-8e71-a3db91cf2060.mov

roleoroleo, 2021-07-16:
> Could the "1970" date affect the continuous recording of data?

This is a know bug but it's not a problem.

> The time is always two hours behind

This could be a problem, try to remove the "Timezone" info and test again the cam.

> I'm also pretty new here. But when the codes are coming so fast, it doesn't seem normal to me.
> I think @roleoroleo can surely explain this.

This behavior is not normal but it depends on the configuration.
With cloud disabled, I receive this message every 10/20 seconds:
`01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00`

> 01 00 00 00 08 00 00 00 81 00 01 00 00 00 00 00
> 01 00 00 00 04 00 00 00 22 40 01 00 00 00 00 00

And I don't know what those codes mean, sorry.

I know that you want to use cloud but also try a test with cloud disabled.

SmartM-ui, 2021-07-16:
> This is a know bug but it's not a problem.
>
>
> This could be a problem, try to remove the "Timezone" info and test again the cam.
>
>
> This behavior is not normal but it depends on the configuration. [...]

Hi,
I have no references to the timezone, I tried to insert it a few days ago, but then I noticed that the time of the event recording was not correct.

[image]

I will try to disable the cloud and try new tests. I'll update you as soon as possible
Thanks

roleoroleo, 2021-07-16:
When you say

> The time is always two hours behind

where do you see this?

SmartM-ui, 2021-07-16:
> I think that your cam saves a mp4 file every minute beacause it starts recording but doesn't receive the stop message.
> But if you use ipc_cmd to send the stop message it should stop (until the next start).

I tried sending the command:
/ tmp / sd / yi-hack / bin / ipc_cmd -T
but still the videos were saved every minute.

I specify that automatic saving every minute does not always happen. 
On some occasions only 3 files are recorded for one minute and then it stops correctly, but I could not understand when it works correctly to be able to give you more information

WalkingDot, 2021-07-16:
> I tried sending the command:
> / tmp / sd / yi-hack / bin / ipc_cmd -T
> but still the videos were saved every minute.

Maybe try the "steam hammer method"?
Create or upload this file to your camera, execute it and wait a few minutes.
**But wait until roleoroleo gives his okay.**
```
#!/bin/sh

while true; do
  /tmp/sd/yi-hack/bin/ipc_cmd -T
  sleep 1
done
```

SmartM-ui, 2021-07-16:
> When you say
> 
> > The time is always two hours behind
> 
> where do you see this?

I can see it in the "System" home screen.
Without the time zone, the time is two hours back on the home screen, but the time is correct in the events section.

[image]

If, on the other hand, I set the timezone, in the initial "System" screen the time is correct (as in Italy), while in the events section, the time is two hours ahead

SmartM-ui, 2021-07-16:
> > I tried sending the command:
> > / tmp / sd / yi-hack / bin / ipc_cmd -T
> > but still the videos were saved every minute.
> 
> Maybe try the "steam hammer method"?
> Create or upload this file to your camera, execute it and wait a few minutes.
> **But wait until roleoroleo gives his okay.**
> 
> ```
> #!/bin/sh
> 
> while true; do
>   /tmp/sd/yi-hack/bin/ipc_cmd -T
>   sleep 1
> done
> ```
> 
> If the test is done kill it with:
> `killall TheScriptName`

OK,
we await @roleoroleo 

For info, to kill all processes, just restart the cam from YI-HACK via the "maintenance / reboot" page?

roleoroleo, 2021-07-17:
> If, on the other hand, I set the timezone, in the initial "System" screen the time is correct (as in Italy), while in the events section, the time is two hours ahead

Yes, this is not ok. I have to check it.

> we await @roleoroleo

You can try it.

> For info, to kill all processes, just restart the cam from YI-HACK via the "maintenance / reboot" page?

Yes.

SmartM-ui, 2021-07-19:
Hi @roleoroleo @WalkingDot 

On Saturday and Sunday I was able to perform two "actual" tests without anyone at home.

**Actual rehearsal without anyone at home - Saturday 17 July**

16:31 - activated the webcam through the switch status on Home Assistant
From 16.31 to 16.59 he automatically and continuously recorded videos every minute. The first recording does not show the movement of the lid / sensor.

At 17.00 he created, as happens at the time change, a new directory in events and recorded automatically from 17.00 to 17.04, automatically recording every minute.

It was about 50 minutes without recording and then recorded a new video at 17.53 and 17.54 and again at 20.17 - 20.18 - 20.22 - 20.23

At 20.32 I returned home without deliberately turning off the webcam, which sent telegram notification via Home Assistant and notification via its YIhome App and recorded the video correctly. Then I turned off the webcam via the status switch on Home Assistant.

I specify that from 16.31 to 20.32 I have not received telegram notifications or alerts from the YIhome motion detection app, so no false positives should have been received

**Actual test without anyone at home - Sunday 18 July leaving the same settings as the previous day**

19:33 activated the Webcam through the switch status on Home Assistant

At 19.33 and 19.34 it recorded two videos (I noticed that the webcam recorded the video during the vertical rotation of the sensor, so it could be a good condition to save the video)

At 20.31 I returned home without deliberately turning off the webcam, which sent telegram notification via Home Assistant and notification via its YIhome App and recorded the video correctly. Then I turned off the webcam via the status switch on Home Assistant.

This day the webcam performed without any anomalies

SmartM-ui, 2021-07-19:
> OK,
> we await @roleoroleo
>
> For info, to kill all processes, just restart the cam from YI-HACK via the "maintenance / reboot" page?

Hi @WalkingDot ,
Since @roleoroleo  has given the green light I try!

The script I save it as text from notepad and rename it .bin and, after loading it on the webcam, I wait for the webcam to start recording automatically and after a few minutes I launch the script from the MacBook terminal and see if the recordings are interrupted. Then I try to pass in front of the sensor and verify that the registration is done correctly, stopping after a minute.

SmartM-ui, 2021-07-19:
> > If, on the other hand, I set the timezone, in the initial "System" screen the time is correct (as in Italy), while in the events section, the time is two hours ahead
> 
> Yes, this is not ok. I have to check it.

Did I explain the time problem well or do you prefer it to be more precise and I open a topic with reference screenshots?

WalkingDot, 2021-07-19:
> The script I save it as text from notepad and rename it .bin

the file extension should be `.sh` e.g. `stoploop.sh`.
good luck ;-)

SmartM-ui, 2021-07-20:
> > The script I save it as text from notepad and rename it .bin
> 
> the file extension should be `.sh` e.g. `stoploop.sh`
> good luck ;-)

Thank you!

roleoroleo, 2021-07-20:
Is the room in dark?
Is the IR led switched on?
Some time ago, I noticed a problem with a similar cam: the IR led was unstable and it increased and decreased the light.
So the cam was detecting motion detection continuously.

SmartM-ui, 2021-07-20:
> Is the room in dark?
>> It happens both when there is light in the room during the day, and at night in the dark

> Is the IR led switched on?
>> yes

> Some time ago, I noticed a problem with a similar cam: the IR led was unstable and it increased and decreased the light.
>> In the various automatic one-minute recordings, I did not notice any changes in brightness

> So the cam was detecting motion detection continuously.
>> This is the strangest part, because the one-minute "automatic" videos are recorded, but the motion sensor never goes to the on state, in fact I don't receive notifications from the official YIhome app, much less from Home Assistant. I find only the videos saved in the microSD.

When I deliberately pass in front of the webcam, the sensor in the home assistant takes the status of on and the notification arrives, also the notification with 6 seconds video arrives on the YIhome app.

WalkingDot, 2021-07-20:
> I placed the file, via filezilla, in /tmp/sd
> 
> To launch the command, do I proceed as usual?

only;
`/tmp/sd/stoploop.sh`

SmartM-ui, 2021-07-20:
> > I placed the file, via filezilla, in /tmp/sd
> > To launch the command, do I proceed as usual?
> 
> only;
> `/tmp/sd/stoploop.sh`

Thanks!

This afternoon, without having touched anything compared to the previous day (which had not recorded automatically) from its activation at 18.51 and until its deactivation at 20.07, it has always recorded every minute!

I have never received a motion detection message either through the YIhome app or through the YI-Hack

Now, I've re-enabled the webcam to test the script, but instead it seems to work fine without saving the video every minute. 

Let's see what happens tomorrow. Good night

roleoroleo, 2021-07-21:
> This is the strangest part, because the one-minute "automatic" videos are recorded, but the motion sensor never goes to the on state, in fact I don't receive notifications from the official YIhome app, much less from Home Assistant. I find only the videos saved in the microSD.

Never goes or goes only at the start of the 1st video?
Example:
- The cam records from 18.51 to 20.07
- If I'm not wrong you have in the sd card a file for each minute (approx 76 files).
But if you open the app do you see an event at 18.51?
If yes, how long does it last?

SmartM-ui, 2021-07-21:
> Never goes or goes only at the start of the 1st video?
> Example:
>
>     * The cam records from 18.51 to 20.07
>
>     * If I'm not wrong you have in the sd card a file for each minute (approx 76 [...]

Hi,
last night I deleted the files from the microSD (and also those that had uploaded to FTP).

I had no event due to motion detection, no notification arrived both in the official YIhome app and notifications that are sent by Home Assistant when it detects a change of state on the sensors created by your integration

In the YI-HACK browser, he had created 3 directories, divided by time (e.g. 2021-07-21 18-00; 2021-07-21 19-00; 2021-07-21 20-00) and inside as many files as the minutes recorded, in total, approximately 76 files.

Tonight, when I leave the house, I activate the webcam and carefully check everything, if it should record again continuously.

SmartM-ui, 2021-07-22:
Hi,
yesterday the webcam didn't automatically record every minute when i was away from home so i couldn't do any tests.

I was thinking, would it be possible to create a recording status sensor or send a notification or email when YI-Hack creates a new file within the events section?

As the recording is disconnected from the motion sensors, it would be useful to know if the cam is recording video.

roleoroleo, 2021-07-23:
If I'm not wrong, when the cam is recording, a temporary file is created:
/tmp/sd/record/tmp.mp4.tmp

SmartM-ui, 2021-07-23:
> If I'm not wrong, when the cam is recording, a temporary file is created:
> /tmp/sd/record/tmp.mp4.tmp

That's right, I seem to have seen it a few times.
Is it possible to create a sensor in HA / YI-Hack that checks the presence of this file so as to know if it is recording even without having received detection notifications?

SmartM-ui, 2021-08-17:
> > I placed the file, via filezilla, in /tmp/sd
> > To launch the command, do I proceed as usual?
> 
> only;
> `/tmp/sd/stoploop.sh`

Hi @roleoroleo @WalkingDot 
I'm back,
it was a few days that when I turned on the cam, through the switch status, the videos were saved automatically and continuously every minute.

I tried to give the command via terminal, but it seems not to find the file if I enter the path as you indicated (/tmp/sd/stoploop.sh), while writing the path as for the other tests performed (/ tmp / sd / yi-hack / bin / ipc_cmd -f /tmp/sd/stoploop.sh), did not give any errors, but nothing changed, the video kept recording automatically every minute.

[image]

Also, I tried giving the mqtt payload:
      topic: yicam / motion_detection
      payload: motion_start

      topic: yicam / motion_detection
      payload: motion_stop
      
      topic: yicam / ai_human_detection
      payload: human_start

      topic: yicam / ai_human_detection
      payload: human_stop
      

I tried to set the "at home" mode through the official YIHome app

I tried to enable and disable ai_human_detect from the YI-HACK web page

But the video kept being recorded continuously every minute.

In order for the video to stop recording automatically, I noticed that I have to walk in front of the camera.

After two minutes from the actual passage, after further recording of two videos lasting one minute, the recording is finally stopped.

The only difference, I think, is in the motion_files which is filled with real values.

The next time it goes in the loop, I have to check what date is displayed in the motion_files. From the historian, I can see that the above referred to day 15 (today is 17).

[image]

[image]

Do you think it is possible to do something to avoid having to physically pass in front of the camera to remedy the problem?

Thanks!

roleoroleo, 2021-08-17:
> Do you think it is possible to do something to avoid having to physically pass in front of the camera to remedy the problem?

If `ipc_cmd -T` doesn't work there is no remedy.

If you have time to spend you could try to uninstall the hack completely to understand if the problem is also present with the original cam. I don't remember if you've already tried.

SmartM-ui, 2021-09-02:
Hi @roleoroleo 

I did one more test to try to solve the problem of recording videos continuously.

After several days that I had no abnormal recordings (I was away from home so I did not restart HA / CAM), I saw that after updating the firmware of YI-HACK, resulting in a restart of HA / YI-HACK, the cam started to record video continuously in one-minute clips (as happened randomly even previously).

I verified through MQTT that the start of the recording was in the year 1970.

[image]

Removing the retain from the MQTT and restarting the cam from your web page, I saw that it stopped recording all the time. From Filezilla I could see that the temporary file in the / tmp / sd / record folder had stopped recreating (and is no longer present).

[image]

[image]

In summary, to stop recording videos continuously, as per other tests carried out, I have to either physically go to the camera (in order to update the motion_files date or remove the string with the motion_files retain via MQTT explorer and restart the camera.

Now I tried to remove the Retain for video list messages by setting it to 0 from the YI-HACK MQTT web page
"A retained message is a normal MQTT message with the retained flag set to true. The broker stores the last retained message and the corresponding QoS for that topic. (1 = enabled, 0 = disabled)"

[image]

Let's see if with this setting I will no longer have this problem.

It remains to be understood why the registration start date is entered with the year 1970.

SmartM-ui, 2021-09-06:
Hi @roleoroleo ,

**finally, after many tests, I was able to understand why the problem of infinite video recording occurs, even without any movement:**

if during the video recording phase due to a movement the cam is turned off with the "Switch on / off the camera" button on the we YI-HACK page or through the switch in Home Assistant, the video recording is stopped, but when the cam is turned on again, again via the "on / off the camera" switch, the video resumes saving itself without stopping after two minutes of recording, recording the video endlessly with one-minute clips.

Understanding the problem from where it was generated, I was able to do several tests:

By running the command from terminal / tmp / sd / yi-hack / bin / ipc_cmd -T (I tried to do it several times consecutively), the video continued to be saved even after the two minutes and I saw, through MQTT Explorer, the motion_files is ranged from:
{"start": "1970-01-01T00: 00: 00 + 0000", "end": "2021-09-05T23: 01: 24 + 0000", "files": []}
to {"start": "1970-01-01T00: 00: 00 + 0000", "end": "1970-01-01T00: 00: 00 + 0000", "files": []}

I tried to manually launch the human_start and human_stop payload and the motion_start and motion_stop payload, to no avail, the video kept saving infinitely.

I tried to delete, through filezilla, the temporary file tmp.mp4.tmp, but at the stroke of the minute, it was automatically regenerated and the infinite video recording restarted.

I tried to completely delete the motion_files payload, to no avail.

I tried to send the motion_files payload with the end two minutes after the current time, nothing.

I tried to activate and deactivate the new privacy switch, but the video continued to be recorded, even though it did not stream through webrtc.

The only solution to stop the continuous recording of the video is to physically pass in front of the webcam or actually trigger the movement of the cam (for example by turning on a light bulb). Another solution is to reboot the camera.

Now that you have more information on the problem, do you think it can be solved?
What is the command to not have the temporary file automatically regenerated? 

I did not understand why if the human_start payload is sent and then human_stop the video does not stop, while if a person really passes in front of the webcam, the video stops recording infinitely. What other command does your firmware receive to understand that a person actually passed in front of the cam?

Thanks in advance!

roleoroleo, 2021-09-10:
> It remains to be understood why the registration start date is entered with the year 1970.

This happens when the mqttv4 process detects a motion_stop message but doesn't detect a motion start message.
So the date is 0 (01-01-1970).

> Removing the retain from the MQTT and restarting the cam from your web page, I saw that it stopped recording all the time. From Filezilla I could see that the temporary file in the / tmp / sd / record folder had stopped recreating (and is no longer present).

I think it's not the best setup, to use retain.
Set all retain values to 0 except "Retain for birth and will messages".
But I'm pretty sure that retain is not related to your problem.

After your tests and this result:

> if during the video recording phase due to a movement the cam is turned off with the "Switch on / off the camera" button on the we YI-HACK page or through the switch in Home Assistant, the video [...]

I could try to remove the temporary file when I kill the processes.
Now, I will try to follow your hints to replicate the problem.

SmartM-ui, 2021-09-21:
Hi @roleoroleo 
were you able to replicate the problem?
Most importantly, were you able to fix it? :-)

roleoroleo, 2021-09-22:
I haven't had time to work on it yet.
Sorry.

closed 2023-01-08 by stale[bot]

kyutov, 2023-07-27:
Hi @roleoroleo, I’m having exactly the same issue even with the latest version of the hack. I have noticed that the issue occurs when the camera is turned off while it is still in motion detected state. Any suggestions what I can try in order to fix it?

roleoroleo, 2023-08-18:
Probably it's a bug related to some Yi process.
And it depends on the model, my cam is not affected.

Just to understand where is the problem, you should try to kill mp4record and restart it.