roleoroleo/yi-hack-Allwinner-v2 #233: GUI only works on iOS mobile, not on PC

d-ostrovskiy, 2021-07-26:
First, excellent work with this!
I recently got a 4-pack of Yi cams, 3 of them arrived as y20 but one was y21ga, and it malfunctioned. In my search for a solution, I discovered this great project. Flashed the hack without issue, and I'm able to connect to the GUI via Opera on my iPhone 11. I can adjust settings and save my settings without issue. However, the same won't work on any PC on this network - in all cases, it simply times out.
I've tried this in Opera on my Win 10 Pro x64 laptop and on my Win 10 Home x64 workstation, in both Chrome and Internet Explorer - same bad result across the board.
I also have Ubuntu on my workstation, which is my primary OS, but my wife is playing games on there right now, which is why I have only tested in Windows and I haven't yet had a chance to try out Linux. I don't expect any different result though, to be honest.
Any idea what could be causing this?

d-ostrovskiy, 2021-07-26:
While I'm posting anyway, I just want to highlight how unbelievably sensitive y21ga with respect to power. I thought this one was hopelessly bricked for sure, but all it needed was the OEM cable. Since the y21ga much farther than 10ft away from its power source, I fitted a USB-A-f to USB Micro-f adapter (got two of them on Amazon for $7) to the other end of the OEM cable and joined that with a long aftermarket USB Micro-m to USB-A-m cable I had been trying to use originally. It's worth noting that none of my y20 have this issue - I've got two of them hooked up to the same power source (AC USB hub) via aftermarket cables without any issue.

d-ostrovskiy, 2021-07-26:
Update: ran this on my laptop, and it seems to indicate the laptop and y21ga are on distinct networks. how can this be?
My other PC (the workstation) is physically connected to the modem/router, so there's no way that one could be having the same issue

C:\>ping 192.168.1.71

Pinging 192.168.1.71 with 32 bytes of data:
Reply from 192.168.1.71: bytes=32 time=92ms TTL=64
Reply from 192.168.1.71: bytes=32 time=103ms TTL=64
Reply from 192.168.1.71: bytes=32 time=103ms TTL=64
Reply from 192.168.1.71: bytes=32 time=113ms TTL=64

Ping statistics for 192.168.1.71:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 92ms, Maximum = 113ms, Average = 102ms

C:\>ping 192.168.1.87

Pinging 192.168.1.87 with 32 bytes of data:
Reply from 192.168.1.70: Destination host unreachable.
Reply from 192.168.1.70: Destination host unreachable.
Reply from 192.168.1.70: Destination host unreachable.
Reply from 192.168.1.70: Destination host unreachable.

Ping statistics for 192.168.1.87:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),

d-ostrovskiy, 2021-07-26:
Issue resolved.
The problem was with the WiFi password. The modem/router was using the ISP's default password, which included proscribed characters (specifically, two #). Changed it to a long alpha password and now it works in all browsers, PC and mobile. Still haven't had a chance to test in Linux, but I expect no problems.
Gonna be a pain resetting the two AC units and four AC switches that dropped off the network, but oh well.
As soon as I've got the SD cards for it (tomorrow), I'm gonna flash hacks to the other cams as well, then disable Yi cloud on all of them and properly secure them all.
Thanks again for making that possible for this one camera!

d-ostrovskiy, 2021-07-26:
Meant to close with previous comment, my bad.
Closing with this.

closed 2021-07-26 by d-ostrovskiy

roleoroleo, 2021-08-04:
Sorry for the delay in the response.
Glad you solved.