roleoroleo/yi-hack-Allwinner-v2 #234: Turn Microphone off and on

tunerooster, 2021-07-29:
When playing audio through the camera speaker, I am getting feedback because the microphone is on and is picking up the audio from the speaker.  I noticed there is a feature in the `YI app` to turn the `microphone off/on`.  Can a feature be added to `ipc_cmd` to turn the mic off and on?  E.g.,
```
ipc_cmd -c on                  # -c MIC switch on or off the mic
```
Another idea, in addition to the above feature...  Would it be a good idea to _always_ turn off the mic when playing through the camera speaker?

As always. thanks for a very well done hack!

roleoroleo, 2021-08-04:
So, you would like to switch off the mic when the speaker is playing.
And switch on again when the audio is finished.
Right?

tunerooster, 2021-08-06:
Yes, that would be fine (at least I can't think of a downside to that
strategy for my purposes), but I also think adding an option to ipc_cmd
would be useful.

Thanks so much for your great hack.  It has *totally* enhanced and extended
my camera automation capabilities!

roleoroleo, 2021-08-11:
I tried to sniff the system when I swith on the mic from the app but I didn't find a suitable message linked to this action.
I also tried to find a gpio but nothing.
I will study it again the next week.

tunerooster, 2021-08-11:
On my YI App, for this camera, there is a switch to turn off the mic, so
apparently it is possible.  I don't see the option on my other camera
models.

Outdoor 1080p IFUSY59UOS Model: YHS.3020

 Thanks for your efforts!

roleoroleo, 2021-08-16:
Could you take a snapshot of this switch?

tunerooster, 2021-09-14:
Did you ever get the snapshot of the MIC switch?  I sent it weeks ago to you by email.  I.e., I "shared" it and so it wasn't clear to me how to post it here.

I am attaching it again here...
[image]

Does this feature request look doable?

roleoroleo, 2021-09-14:
Sorry, I didn't receive your previous message.
This is the code when the mic is switching off:
02 00 00 00 08 00 00 00 2a 10 01 00 04 00 00 00 65 00 00 00
and on:
02 00 00 00 08 00 00 00 2a 10 01 00 04 00 00 00 64 00 00 00

I think it's possible to add a command to ipc_cmd

tunerooster, 2021-09-14:
Outstanding!  Thanks!

Do you have any preference for the option letter to use for MIC on/off.  I see you used "-c" for face detection.

tunerooster, 2021-09-14:
I just realized, I don't have the tools to build the ipc_cmd binary.  I will gladly add the code to ipc_cmd.h and ipc_cmd.c and send you a patch if you will build it?

roleoroleo, 2021-09-14:
No problem, I will edit the files myself.

roleoroleo, 2021-09-14:
Here it is:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/fe5a18b91744ce4d8914e87c5c3a155d97b275f5

tunerooster, 2021-09-14:
You're the **best**!   Looking forward to a new ipc_cmd, speak.sh and speaker.sh

tunerooster, 2021-09-16:
Were you going to post the updated ipc_cmd and speak scripts?  Or did I miss something?

Sorry for the confusion on my part...

roleoroleo, 2021-09-16:
I'm testing them but something is wrong.

tunerooster, 2021-09-17:
No problem.  I just wanted to make sure I didn't miss something.

Thanks as always...

tunerooster, 2021-09-23:
Don't mean to push, but have you had any luck with this yet?

roleoroleo, 2021-09-23:
No.
I believe there is a conflict between the yi firmware and my command.
Probably yi switch on and off the microphone when I play a sound.
I'm trying to understand.

tunerooster, 2021-09-23:
I much appreciate your effort and thanks for the update!

roleoroleo, 2021-09-24:
I tested the command but it works only with the app.
If I disable the mic, rtsp stream contains audio.
I don't know how to disable it in a different manner.

tunerooster, 2021-09-24:
I understand.  It sounds like that's all the option does.  Too bad...  Do
you also hear feedback from the mic when the camera speaker plays?

closed 2023-01-08 by stale[bot]