roleoroleo/yi-hack-Allwinner-v2 #242: Motion detection doesn't work in general -- y21ga -- allwinnerV2 0.1.8

Pulsar981, 2021-08-29:
y21ga -- allwinnerV2 0.1.8
offline use, no Yi app, no internet, no other app, just with web interface

Hi,
unfortunately I can't get my motion detection running, no matter which combination of settings I try (I've tried a lot). When I set the option "save video when a motion is detected" to off, the cam records constantly 1min-videos - this is correct behaviour as described there. I see the videos in the web interface in the Events section. When I activate this option, it should only record when a motion is detected, but no videos are recorded. Directing the cam on a near TV to ensure constant motion. I don't see new records in the Events section, nor on the SD card.

roleoroleo, 2021-08-30:
Try to enable motion detection from the app.
I know that you don't use the app but try to activate it  only for the startup configuration.

- Enable cloud
- Configure yuour settings from the app
- Disable cloud

mcsshs, 2021-09-15:
> Try to enable motion detection from the app.
> I know that you don't use the app but try to activate it only for the startup configuration.
> 
> * Enable cloud
> * Configure yuour settings from the app
> * Disable cloud

I have the same problem. I did your suggestions, but the problem remained. :(
Model Suffix: y21ga
Firmware Version: 0.1.9

roleoroleo, 2021-09-15:
Let's start from scratch. We need to check if the problem is mqtt or the detection of the event.
If you use the app, does motion detection works?
Do you receive notification messages when a motion is detected?

> Directing the cam on a near TV to ensure constant motion.

This is not a good idea because this is a single event, until the motion stops.
You have to alternate a still image with a movement.

Maxym77, 2022-01-04:
my y21ga send a lot of MQTT motion message every 1/2/3 minutes... is bugged

roleoroleo, 2022-01-07:
Are you sure that the image doesn't change?
My cam sends a motion message when I leave the phone in the kitchen and the led blinks.

closed 2023-01-08 by stale[bot]