roleoroleo/yi-hack-Allwinner-v2 #243: r30gb stucks after connecting to wifi, even without hack

areyoufserious, 2021-08-30:
i applied hack, everything worked for 30 mins, then stucked
i reset the camera and even can't finish the default setup, camera tells that connected to wifi, then stuck and restart

can you please help me?

areyoufserious, 2021-08-30:
also after tries i see file in logs folder log_no_wifi_dev.tar.gz
[log_no_wifi_dev.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7073712/log_no_wifi_dev.tar.gz)

roleoroleo, 2021-08-30:
Maybe the hack is not completed successfully.
Try to unhack the cam and reset with the back button.

areyoufserious, 2021-08-30:
if you mean unbricking section, unfortunately, i don't have USB flashed to do that
is there any way?

roleoroleo, 2021-08-30:
I mean "Remove the hack".
Unbrick is the last try.

areyoufserious, 2021-08-30:
sure, remove sd card was my first idea

roleoroleo, 2021-08-30:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Remove-the-hack

closed 2023-01-08 by stale[bot]