roleoroleo/yi-hack-Allwinner-v2 #246: Date / Time Problem

SmartM-ui, 2021-09-02:
Hi,
By leaving the timezone setting blank, the date in the event records have the correct date (real time 00.56 - Time reported: 00:56):

[image]

but the local time is two hours behind (real time sep 03 01:02:23 - time reported: sep 02 23:02:23

[image]

and also in the motion_files (see the end date) it is two hours behind (Real Time: 03.09.2021 00:58:22 - Time Reported: 2021.09.02 22:57:12)

[image]

If I set the timezone with the Italian value, the date in the event log is two hours ahead.

How should I set the value for the timezone to have a synchronization of all dates / times? 

Thanks

roleoroleo, 2021-09-10:
How do you set timezone in the yi app?

SmartM-ui, 2021-09-21:
> How did you set timezone in the yi app?

Hi @roleoroleo 

timezone in YIhome: UTC +1:00

roleoroleo, 2021-09-22:
Plase, try to set timezone = GMT in the app.
Check this post: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/203#issuecomment-918762012

SmartM-ui, 2021-09-22:
> Plase, try to set timezone = GMT in the app.
> Check this post: [#203 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/203#issuecomment-918762012)

Perfect!
Set UTC +0 in the YIHome app and set the correct timezone to YI-HACK.
Now the time is correct everywhere

roleoroleo, 2021-09-22:
The name should be "Coordinated universal time".

SmartM-ui, 2021-09-22:
> The name should be "Coordinated universal time".

That's right, I set Coordinated universal time! 
Thanks

Jon8RFC, 2021-12-05:
@roleoroleo Am I missing a step?  I'm getting confused because it's not working as expected...
Yi Home 1080p
IFUSY65U5W
9.0.36.02_202105081647
y211ga

Even if I disable NTPD, times don't seem to match up, universally.  There seem to be different variables.
For example, if the time is December 4, 2021 23:05:00 (GMT-6) and I've set the following and then rebooted:
Yi-Hack: CST6CDT,M3.2.0,M11.1.0
Yi-Hack NTPD: disabled
Yi app: Coordinated universal time

**I'm rounding off numbers and keeping them consistent after observing the changes, for simplicity in showing the issue...**
Bolded are changes.
I'm sorry for the wall of text, but I wanted you to see what changes.

My computer shows "December 4, 2021 23:05:00" (correct)
Yi-Hack web status shows "Sat Dec 4 23:05:00 CST 2021" (correct)
Recording folder being recorded to is "2021Y12M04D23H" (correct)
Filezilla client to Yi-Hack FTP recorded file (dynamically changes, not read/write) modified time shows "12/4/2021 17:05:00" (6 hours behind)

Now, if I remove the timezone set in Yi-Hack configuration page, save, power cycle completely (not a soft reboot):
My computer shows "December 4, 2021 23:05:00" (correct)
Yi-Hack web status shows "Sun Dec 5 05:05:00 GMT 2021" **(6 hours ahead)**
Recording folder being recorded to is "2021Y12M05D05H" **(6 hours ahead)**
Filezilla client to Yi-Hack FTP recorded file modified time shows "12/4/2021 23:05:00" **(correct)**

Now, if I change the Yi app timezone to UTC-06:00, power cycle, nothing seems to change.

Now, if I change the Yi-Hack timezone to CST6CDT,M3.2.0,M11.1.0, save, power cycle:
My computer shows "December 4, 2021 23:05:00" (correct)
Yi-Hack web status shows "Sat Dec 4 23:05:00 CST 2021" **(correct)**
Recording folder being recorded to is "2021Y12M04D23H" **(correct)**
Filezilla client to Yi-Hack FTP recorded file (dynamically changes) modified time shows "12/4/2021 17:05:00" **(6 hours behind)**

Now, if I "Disable Cloud", "Enable Recording without Cloud", "Enable NTPD", save, power cycle:
My computer shows "December 4, 2021 23:05:00" (correct)
Yi-Hack web status shows "Sat Dec 4 23:33:12 CST 2021" (correct)
Recording folder being recorded to is "2021Y12M04D23H" (correct)
Filezilla client to Yi-Hack FTP recorded file (dynamically changes) modified time shows "12/4/2021 18:05:00" **(5 hours behind)**

Now, if I change the Yi app timezone to Coordinated universal time, power cycle, nothing changes.

Now, if I re-enable "Cloud" (keeping NTPD enabled), save, power cycle:
My computer shows "December 4, 2021 23:05:00" (correct)
Yi-Hack web status shows "Sat Dec 4 23:33:12 CST 2021" (correct)
Recording folder being recorded to is "2021Y12M04D23H" (correct)
Filezilla client to Yi-Hack FTP recorded file (dynamically changes) modified time shows "12/4/2021 18:05:00" **(6 hours behind)**

pwrsoft, 2022-01-18:
I have the same issue with invalid dates in recording folders, as well as in file names.

canuckray, 2022-01-20:
@roleoroleo , I am also having issues with timestamp / time zone on my Yi 1080 Home.

BUT.... the issue only presents itself on the date/timestamp overlay that is overlaid on the rtsp stream. 

My setup is:

- Yi 1080 Home, QFUS

- Using Yi App:  Time zone set to UTC +0000
- Yi Hack web: set time zone to my correct time zone, disabled cloud, enabled NPT time sync, reboot. 

The result:

Yi Hack web reports the local time correctly - 10:14am (I am GMT-4, so GMT would be 2:14pm)
However, the timestamp overlay on the rtsp stream shows 8 hours ahead!  It's as if it has added 4 hours to GMT, rather than subtracted 4 hours. (it is showing 18:14, as shown below).

I have tried different combinations of Yi time zone, NTP on/off, Yi Hack time zone, and cannot get these to agree.

[image]

roleoroleo, 2022-01-20:
Your problem is different because is related to the overlay.
And I don't know hot to solve it because it depends on the yi firmware.

But check if the "rmm" process has the correct environment variables.
Login with ssh.
Find the process id: `ps | grep rmm`
Check the environment: `cat /proc/<PID>/environ`

canuckray, 2022-01-20:
Thanks for the quick reply - appreciate it.   The output is below.  Looks cryptic to me, but maybe something will jump out at you.

[image]

roleoroleo, 2022-01-20:
TZ variable is absent.
This is not ok.
What is yout timezone setting in the web interface?

canuckray, 2022-01-20:
pasted from web interface:  EST5EDT,M3.2.0,M11.1.0

[image]

roleoroleo, 2022-01-21:
I checked the script and I can't understand.
Try another command:
`cat /proc/1172/environ | tr '\0' '\n'`

canuckray, 2022-01-21:
[image]

SeriousPat, 2022-02-11:
di you fix it?
i habe UTC-0 in Yi app.
my timestamp up left in the video is incorret.

[image]

[image]

[image]

roleoroleo, 2022-02-11:
I don't know how to fix it, at the moment.

1-Al, 2022-02-16:
FWIW I found using the setting GMT in the Yi app and TZ in Yi-Hack breaks the seek time in the Yi android app when trying to playback video. For those using RTSP and personal recorders I guess it doesn't matter but I ended up changing both yi app and yi hack to the proper timezone and only applying the timezone inline where needed in system.sh. For example with httpd, although others might be added. 

```
export TZ=$(get_config TIMEZONE)
if [[ $(get_config HTTPD) == "yes" ]] ; then
    mkdir -p /tmp/sd/record
    mkdir -p /tmp/sd/yi-hack/www/record
    mount --bind /tmp/sd/record /tmp/sd/yi-hack/www/record
    httpd -p $HTTPD_PORT -h $YI_HACK_PREFIX/www/ -c /tmp/httpd.conf
fi
# maybe some other apps here too
export -n TZ
```

But what works for me might not work for others and I don't know if Yi use daylight savings, maybe someone with daylight savings can shed some light? :) 

Example with above and +7hrs set for both Yi Hack (where needed) and the Yi app. The r30gb needs a workaround but the other camera's b091qp/h60ga seem fine.

 
[image]

roleoroleo, 2022-02-16:
I wanted to try to do the opposite: run rmm without TZ.
But probably your solution is better.

roleoroleo, 2022-02-18:
@SeriousPat 
Try this system.sh (overwrite the original one).
Set your timezone in the app (instead of UTC).
Reboot and check:

- Date/time in status web page.
- The watermark printed on the video and on the snapshot.
- If you are able to play videos captured with a motion (both web gui and app).

[system.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8095078/system.tar.gz)

roleoroleo, 2022-02-24:
What about the date/time of the table in the "Events" page.
Is it ok? Or is it shifted by an offset?
I changed the system.sh and I reconfigured the correct time in the app.
Now, the time of the events is not ok.

1-Al, 2022-02-25:
Not sure who you are asking, if it's me then my event page is fine. Don't use MQTT, CRON and SSH seems to do it's own thing with time, as I said before I'm a noob with Linux and not that much better with Windows so don't know.

TBH just telnet/ssh and ftp would be enough for me with these camera's but it's nice to learn so I try a few things. Even learning a little arm and wrote a test grabber with piped aac and video streams from fshare. Audio was a bit hit and miss with my modification to rtsp_server_yi, the newer one you submitted works better than my attempt. Not sure what to do with videofile.cpp to get the 265 working, all those brackets and semi colons do my head in.  :/

roleoroleo, 2022-02-25:
> Not sure what to do with videofile.cpp to get the 265 working, all those brackets and semi colons do my head in. :/

Check the last commit:
https://github.com/roleoroleo/RtspServer/tree/yi

1-Al, 2022-02-26:
Forgot to mention I'm not using ONVIF either and RTSP is being used for learning purposes, no dedicated box for it's use. Cloud is enabled but no cloud storage.

> Check the last commit: https://github.com/roleoroleo/RtspServer/tree/yi

Cool, will try it out later today. Will start another thread under discussions.

kalyan02, 2022-07-09:
Thanks for the excellent software. Its magical to be able to ssh and play around into the camera.

Am facing similar date time problems.

I moved the `export TZ` to just before the `rmm` is executed, this seems to fix the folder label. However the overlay on top left is still 8 hours behind.

FYI: I am not using Yi App at all and all cloud features are disabled. Is there anyway to modify the firmware timezone offset? I am looking through #203 . Any tips?

roleoroleo, 2022-07-09:
I don't know how to set the internal timezone (I mean the same that the app changes).

rikrdo89, 2022-11-24:
I'm having the same issue. The timestamp on the rtsp feed is off by 5hrs, whenever I disable cloud. I have the yi app set to UTC. If I set the timezone to EST (which should subtract 5 hrs from UTC), it adds 5 hours more to the timestamp, so it shows a time that is 10hrs ahead. I tried multiple combinations of turning off ntdp, timezones, reboot cycles, but the timestamp seems to be always off unless I don't disable cloud.

roleoroleo, 2022-11-25:
Is your TZ positive or negative?

rikrdo89, 2022-11-25:
It's supposed to be negative... my TZ is EST5 (Eastern time, which UTC - 5 hr)

roleoroleo, 2022-11-25:
Try to overwrite 2 files contained inside this archive:
[patch.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10093421/patch.tar.gz)

/tmp/sd/yi-hack/bin/set_tz_offset
/tmp/sd/yi-hack/script/system.sh

rikrdo89, 2022-11-25:
Thanks a lot, I'll try that. Do I just ssh into the cam and overwrite those files, and reboot?, or Do I turn off the cam, take the micro sd and change the files in my computer, and then power on the cam?

roleoroleo, 2022-11-25:
As you prefer.
Normally I use ftp. No need to change permissions because the file system is a fat32.
A reboot is required.

canuckray, 2022-11-25:
I can confirm that these 2 files corrects the issue!   Beautiful!

Thanks @roleoroleo .... greatly appreciated!

rikrdo89, 2022-11-25:
the patch works!!! Thanks a lot @roleoroleo, no more incorrect timestamp!

itkfilelor, 2022-12-13:
Sadly, the system.sh file crashes both of my cameras on the current F/W (0.2.5)  (model: y211ga)

roleoroleo, 2022-12-13:
Pay attention when you copy the file to the cam.
The file must be in linux format (LF line termination).

itkfilelor, 2022-12-13:
> Pay attention when you copy the file to the cam.
> The file must be in linux format (LF line termination).

That should be easy since my daily is PopOs 😊
When it failed I also opened it up in VSCode to see what's up and I saw all \n no \r

roleoroleo, 2022-12-13:
Please, check this commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/4db939ee69d7b134313dd2610be2b71cc328bf32

itkfilelor, 2022-12-13:
> Please, check this commit: [4db939e](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/4db939ee69d7b134313dd2610be2b71cc328bf32)

Just applied, if it fails, I only have 1 more shot til much later. At work right now. 😊
Technology is great.

chrbar11, 2023-04-16:
> FWIW I found using the setting GMT in the Yi app and TZ in Yi-Hack breaks the seek time in the Yi android app when trying to playback video. For those using RTSP and personal recorders I guess it [...]

Hi ghost, did you obtain the result when Cloud was disable?
I met trouble to set the good timezone for the built-in YI timestamp at left top ([#606](https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/606#discussion-5003820)).

chuckheron, 2023-06-22:
Hi,
I applied the patch, but as soon as I set TIMEZONE to anything, the camera wouldn't boot.

I was trying the TZP_SET calculation in /bin/sh, and I kept getting a 'bad substitution' with the  ${TZP:0:1} syntax.

I got it to work using sed:
`TZP_SET=$(echo $TZP | sed 's/\(.\)/\1 /g' | awk '{ print ($1($2$3)*3600+($4$5)*60) }')
`

My camera boots now and the times seem to be correct.

Sadly, it's not doing motion detection anymore though, but that's probably a different issue.

Thanks
-chuck

roleoroleo, 2023-06-23:
> I was trying the TZP_SET calculation in /bin/sh, and I kept getting a 'bad substitution' with the ${TZP:0:1} syntax.

This is strange.
What happens if you run the followin commands?
```
root@yi-hack:~# export TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
root@yi-hack:~# export TZP=$(TZ=$TIMEZONE date +%z)
root@yi-hack:~# export TZP_SET=$(echo ${TZP:0:1} ${TZP:1:2} ${TZP:3:2} | awk '{ print ($1$2*3600+$3*60) }')
root@yi-hack:~# echo $TZP_SET
```

chuckheron, 2023-06-23:
Hi,

/bin/sh
sh$ export TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
sh$ export TZP=$(TZ=$TIMEZONE date +%z)
sh$ export TZP_SET=$(echo ${TZP:0:1} ${TZP:1:2} ${TZP:3:2} | awk '{ print ($1$2*3600+$3*60) }')
sh: 3: Bad substitution
sh$ echo $TZP_SET

sh$

That syntax seems to work in bash, not sh though.

/bin/bash
bash$ export TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
bash$ export TZP=$(TZ=$TIMEZONE date +%z)
bash$ export TZP_SET=$(echo ${TZP:0:1} ${TZP:1:2} ${TZP:3:2} | awk '{ print (
$1$2*3600+$3*60) }')
bash$ echo $TZP_SET
+7200
bash$

Thanks
-chuck

This is strange.
What happens if you run the followin commands?
***@***.***:~# export TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
***@***.***:~# export TZP=$(TZ=$TIMEZONE date +%z)
***@***.***:~# export TZP_SET=$(echo ${TZP:0:1} ${TZP:1:2} ${TZP:3:2} | awk '{ print ($1$2*3600+$3*60) }')
***@***.***:~# echo $TZP_SET

roleoroleo, 2023-06-23:
Sorry bu I don't understand, inside the cam sh and bash are symbolic links to ash.
Where are you running, these commands?

chuckheron, 2023-06-24:
Hi
Yes, sorry, I was running them on a linux machine, I tried it on the camera there is no error.  
Now I'm not sure why the camera wasn't booting with that code in there.
It's working now..

Thanks
-chuck

sirduke989, 2023-07-18:
> Try to overwrite 2 files contained inside this archive: [patch.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10093421/patch.tar.gz)
> 
> /tmp/sd/yi-hack/bin/set_tz_offset
> /tmp/sd/yi-hack/script/system.sh

I can confirm that the set_tz_offset in this patch fixes the OSD timestamp for a negative tz offset (CST/CDT) but does not appear to be included in the latest version (0.2.9). I tested this on RFUS both Indoor (y211ga) and Outdoor (h30ga) cameras running version 0.2.9. Note: I did not need to replace system.sh.

To get this to work:

1. I re-enabled cloud integration and rebooted
2. I set to timezone to Coordinated universal time in the Yi-Home app
3. I disabled cloud integration, set the correct timezone and enabled time OSD in Yi-Hack
4. Rebooted (Offset was still doubled, IE +10 instead of -5)
5. I replaced set_tz_offset from the patch on the camera
6. Rebooted (now the OSD timestamp is displaying correctly)

roleoroleo, 2023-07-19:
This is weird.
I included the fixes in the master branch and tested again in this moment.
It works with my h52ga (0.2.9) using "CST6CDT,M3.2.0,M11.1.0".

Please run the commands in this post, using your timezone and check the output.

sirduke989, 2023-07-19:
@roleoroleo I actually did try those commands prior to replacing set_tz_offset on 0.2.7 and 0.2.9 and it always gave the correct offset for me of -18000. This lead me to think that the issue was with set_tz_offset and not system.sh. I should have included this in my last post but I am also using "CST6CDT,M3.2.0,M11.1.0" as my Timezone in the UI.

Here are the commands ran on one of my cameras:
root@yicamout:~# export TIMEZONE=CST6CDT,M3.2.0,M11.1.0
root@yicamout:~# export TZP=$(TZ=$TIMEZONE date +%z)
root@yicamout:~# export TZP_SET=$(echo ${TZP:0:1} ${TZP:1:2} ${TZP:3:2} | awk '{ print ($1$2*3600+$3*60) }')
root@yicamout:~# echo $TZP_SET
-18000

I did some more testing and tried replacing set_tz_offset from different patches on one of my RFUS cameras running y211ga 0.2.9 and below are the resulting OSD:
From patch.tar.gz: OSD is UTC-5 (correct)
y211ga (0.2.9): OSD is UTC+5
h52ga (0.2.9): OSD is UTC+5
y211ga (0.2.7): OSD is UTC+5
y211ga (0.2.6): OSD is UTC-5 (correct)
y211ga (0.2.5): OSD is UTC-5 (correct)
y211ga (0.2.4): OSD is UTC-5 (correct)

roleoroleo, 2023-07-20:
Ok, the last try:
set_tz_offset writes 4 bytes to the file /tmp/mmap.info
You can check if it writes properly reading the file:
`hexdump -C -s 0x04e0 -n 4 /tmp/mmap.info`

The offset depends on the model and it's in the following table:
```
    { 0x4a0, 1 },    // Y20GA_9
    { 0x4a4, 1 },    // Y20GA_12
    { 0x4a0, 1 },    // Y25GA
    { 0x4a0, 1 },    // Y30QA

    { 0x4e0, 1 },    // H30GA_9
    { 0x56c, 0 },    // H30GA_11     Already present
    { 0x4e0, 1 },    // H51GA
    { 0x4e0, 1 },    // H52GA
    { 0x560, 0 },    // H60GA        Already present
    { 0x4e0, 1 },    // Q321BR_LSX
    { 0x4e0, 1 },    // QG311R
    { 0x4e0, 1 },    // R30GB
    { 0x56c, 0 },    // R35GB        Already present
    { 0x560, 0 },    // R40GA        Already present
    { 0x4e0, 1 },    // Y211GA_9
    { 0x570, 0 },    // Y211GA_12    Already present
    { 0x4e0, 1 },    // Y21GA_9
    { 0x564, 0 },    // Y21GA_12     Already present
    { 0x4e0, 1 },    // Y28GA
    { 0x56c, 0 },    // Y291GA_9     Already present
    { 0x570, 0 },    // Y291GA_12    Already present
    { 0x4e0, 1 },    // Y29GA
    { 0x000, 0 }     // B091QP       0x560 or 0x56c but already present
```

Try to run set_tz_offset and check if the number is ok.
For example:
```
root@yi-hack-h52ga:~# set_tz_offset -c tz_offset_osd -m h52ga -f 9 -v -18000
root@yi-hack-h52ga:~# hexdump -C -s 0x04e0 -n 4 /tmp/mmap.info
000004e0  b0 b9 ff ff                                       |....|
000004e4
```

sirduke989, 2023-07-20:
I should also mention that I am running v12 of the firmware and each test below was performed after a fresh reboot of the camera. I also ran the hexdumps (for both v9 and 12 addresses) before and after the set_tz_offset command.

Booted with set_tz_offset from y211ga_0.2.9:
```
root@YICAM1:~/bin# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  50 46 00 00                                       |PF..|
00000574
root@YICAM1:~/bin# hexdump -C -s 0x4e0 -n 4 /tmp/mmap.info
*
000004e0
root@YICAM1:~/bin# set_tz_offset -c tz_offset_osd -m y211ga -f 9 -v -18000 -d
debug on
The size of the buffer is 3756
Mapping file /tmp/mmap.info, size 3756, to b6f8d000
Closing the file /tmp/mmap.info
Unmapping file /tmp/mmap.info, size 3756, from b6f8d000
root@YICAM1:~/bin# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  50 46 00 00                                       |PF..|
00000574
root@YICAM1:~/bin# hexdump -C -s 0x4e0 -n 4 /tmp/mmap.info
000004e0  b0 b9 ff ff                                       |....|
000004e4
```

Since I am on version 12 of the firmware I also tried changing that in the command line (looks like it doesn't work when passing in version 12):
```
root@YICAM1:~# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  50 46 00 00                                       |PF..|
00000574
root@YICAM1:~# hexdump -C -s 0x4e0 -n 4 /tmp/mmap.info
*
000004e0
root@YICAM1:~# set_tz_offset -c tz_offset_osd -m y211ga -f 12 -v -18000 -d
debug on
root@YICAM1:~# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  50 46 00 00                                       |PF..|
00000574
root@YICAM1:~# hexdump -C -s 0x4e0 -n 4 /tmp/mmap.info
*
000004e0
```

Not sure if this will help but with set_tz_offset from patch.tar.gz (when OSD is showing correctly):
```
root@YICAM1:~# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  b0 b9 ff ff                                       |....|
00000574
root@YICAM1:~# hexdump -C -s 0x4e0 -n 4 /tmp/mmap.info
*
000004e0
root@YICAM1:~# set_tz_offset -c tz_offset_osd -m y211ga -f 12 -v -18000 -d
set_tz_offset: unrecognized option: c
set_tz_offset: unrecognized option: m
set_tz_offset: unrecognized option: f
debug on
Setting timezone to -18000000 ms
root@YICAM1:~# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  b0 b9 ff ff                                       |....|
00000574
root@YICAM1:~# hexdump -C -s 0x4e0 -n 4 /tmp/mmap.info
*
000004e0
```

Also thank you for all your hard work on helping make this project possible!

roleoroleo, 2023-07-21:
The problem is probably related to the "-f" parameter.
You should usd "-f 12" but it doesn't work.
I will check it.

roleoroleo, 2023-07-21:
I think I found the problem.
Based on the dumps I received from other users, I thought it was already set for this model.
So the code skips the write operation.
I will remove this limit.

roleoroleo, 2023-07-21:
Please, try this:

[set_tz_offset.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12122094/set_tz_offset.gz)

sirduke989, 2023-07-21:
Looks good, thank you.

After replacing set_tz_offset with the one in the previous post and rebooting the camera, I now see the correct OSD and the hexdump looks good.

```
root@YICAM1:~# hexdump -C -s 0x570 -n 4 /tmp/mmap.info
00000570  b0 b9 ff ff                                       |....|
00000574
```

sirduke989, 2023-07-21:
Also tested on Yi Camera Outdoor RFUS on firmware version 12 running 0.2.9 for h30ga and it looks good.

hiagocosta, 2023-10-21:
> Please, try this:
> 
> [set_tz_offset.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/12122094/set_tz_offset.gz)

i'm experiencing the same error, can I use this on my y291ga with 9.3.5 fw?
im on the latest update since now (0.2.9)...

just overwrite the bin file?

roleoroleo, 2023-10-23:
Yes, you can try it.
Rename the old binary and copy the new one.

hiagocosta, 2023-10-23:
Thanks, it worked like a charm!

hiagocosta, 2023-10-23:
updating this issue,
it does not corrected, I think that after a period of time the cam lost de access of ntp and returning back to UTC without TZ

kiddyfurby, 2024-02-02:
I saw the 0.3.1 changelog and was very excited because the long standing issue of OSD timezone is corrected!

We're getting there! but for 1 of my 3 y29ga, the timezone is incorrect.
On the one y29ga, `set_tz_offset -c tz_offset_osd -m y29ga -f 9 -v 0 -d` did correct the OSD time

The two y29ga without timezone issue had Base Version 9.0.09.06_202103291039
The one which required manual set_tz_offset has 9.0.09.**03_202006090939**

I do not have yi app so that settings is whatever value the default is.

[2 more replies]

closed 2024-06-03 by github-actions[bot]