roleoroleo/yi-hack-Allwinner-v2 #247: New Dome U Pro Ufficial Firmware update (version 9.0.27.20_202108161046)

SmartM-ui, 2021-09-03:
Hi,
The YIhome app tells me that there is a firmware update for the Dome U Pro 9.0.27.20_202108161046
I currently have version 9.0.27.19_202105131533 installed

Trying to update, the download is performed, the cam restarts, but fails to update and the previous version is always installed.

In order to proceed with the official firmware update of the cam, do I have to remove the SD with the YI-HACK hack?

roleoroleo, 2021-09-03:
It is not easy to answer this question.
I think it would be better if you remove the hack with the unhack procedure.

SmartM-ui, 2021-09-03:
> It is not easy to answer this question.
> I think it would be better if you remove the hack with the unhack procedure.

In yi-hack-Allwinner-v2, it is not enough to unplug the SDcard to remove the Hack?

roleoroleo, 2021-09-03:
Yes and no...
The hack doesn't start but there are 2 original files modified by the hack:
/backup/init.sh
/backup/tools/wifidhcp.sh

When you run unhack, these files are restore to the original version.
Probably this is not a problem when you update the firmware.
It should work without running unhack.

SmartM-ui, 2021-09-03:
> Yes and no...
> The hack doesn't start but there are 2 original files modified by the hack:
> /backup/init.sh
> /backup/tools/wifidhcp.sh
>
> When you run unhack, these files are restore to the original [...]

Thanks for the reply!
Where do I find the unhack procedure in case I need it?

roleoroleo, 2021-09-10:
> Where do I find the unhack procedure in case I need it?

In the wiki.

Rocket200, 2021-09-18:
i backup the old and new FW 
it help when i upload it ?
i send this yi back and give the eufy p24 a try ;) i have to much yi cams 
[image]

roleoroleo, 2021-09-19:
> it help when i upload it ?

Yes.

Rocket200, 2021-09-19:
[backup LFUS FW 9.0.27.14_202103082324.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7192669/backup.LFUS.FW.9.0.27.14_202103082324.zip)
[backup LFUS FW 9.0.27.20_202108161046.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7192670/backup.LFUS.FW.9.0.27.20_202108161046.zip)

@roleoroleo what u think about eufy p24 cam ? the cam have RTSP in GUI on/off

roleoroleo, 2021-09-19:
Thanks for sharing, I downloaded them.

I don't know eufy p24 cam at all.

Rocket200, 2021-09-20:
the eufy p24 cam is excellent 
RTSP work or Apple HomeKit no Coud!
all is faster than Yi App GUI usw.

SmartM-ui, 2021-09-22:
> > Where do I find the unhack procedure in case I need it?
> 
> In the wiki.

Hi @roleoroleo ,
I took the file for the unhack (unhack_h60ga) and unpacked it in an empty microSD. 
In the root I see the structure as in your example: two folders (Factory and NewBackup) with their files inside.
When I insert the microSD into the camera, after turning it on, it stays in boot mode (yellow light), but it can't go on (about 10 minutes of waiting).
I tried to restart the cam with both microSD and microSD with your previously working hack, but the cam remains in boot mode (yellow light).

I also found the original firmware backup, but it seems to be different (it has a backup folder with 7 .bin files inside). 
I tried to insert the microSD with those files, but still the cam does not restart.

Can I do something or do I have to send her for assistance?

Thanks

roleoroleo, 2021-09-22:
Weird behavior...
I don't know why this happened but other 2 or 3 users noticed this problem.
The only way to recover the cam is the unbrick procedure describerd in the wiki.
But you need a serial connection to che cam.
The backup (7 .bin files) is used in this procedure.

SmartM-ui, 2021-09-22:
> Weird behavior...
> I don't know why this happened but other 2 or 3 users noticed this problem.
> The only way to recover the cam is the unbrick procedure describerd in the wiki.
> But you need a serial connection to che cam.
> The backup (7 .bin files) is used in this procedure.

Provo a chiedere la sostituzione.
Thanks

SmartM-ui, 2021-09-22:
> Provo a chiedere la sostituzione.
> Thanks

Hi @roleoroleo 
I'm sorry I ask you for an info, if you can.
Having to buy another cam quickly, which one do you recommend that works well with your hack?
I would be interested that it has the possibility of privacy (cam shutdown), human recognition (otherwise I would have too many false alarms) and that the notification arrives with video.
Thanks

SmartM-ui, 2021-09-22:
> the eufy p24 cam is excellent
> RTSP work or Apple HomeKit no Coud!
> all is faster than Yi App GUI usw.

Hi @Rocket200 
can you tell me if this is the model that works correctly with YI-HACKS?
"Telecamera wi-fi interno smart eufy Security 2K panoramica, videosorveglianza con AI per riconoscimento persone/animali, assistenza vocale, visione notturna, non richiede HomeBase, microSD non inclusa"

Rocket200, 2021-09-22:
work not with the hack but it have RTSP already ;)
you obly need to Trust the eufy Commpany
the cam needs to stay online to work with the App !! 
no cloud all work with SD 
i try it 1-2 Weeks out now

roleoroleo, 2021-09-22:
> Hi @roleoroleo
> I'm sorry I ask you for an info, if you can.
> Having to buy another cam quickly, which one do you recommend that works well with your hack?
> I would be interested that it has the [...]

Only sonoff gk-200mp2-b is supported by my yi-hack but without human recognition.

SmartM-ui, 2021-09-22:
@Rocket200 
in addition to the rstp I was looking for something that could be integrated into the home assistant to see the sensors

@roleoroleo 
Of the YI models which is the most compatible?

roleoroleo, 2021-09-22:
> Of the YI models which is the most compatible?

y21ga.
But the problem is that you can't choose the model when you place an order with Amazon or Aliexpress.
And probably y21ga is no longer for sale.

SmartM-ui, 2021-09-22:
> > Of the YI models which is the most compatible?
> 
> y21ga.
> But the problem is that you can't choose the model when you place an order with Amazon or Aliexpress.
> And probably y21ga is no longer for sale.

OK thank you!
Could the Eufy be compatible with your hack? I saw that  @Rocket200  he posted the firmware

closed 2023-01-08 by stale[bot]