roleoroleo/yi-hack-Allwinner-v2 #257: Serial number - HELP NEEDED

roleoroleo, 2021-09-15:
In the last cams, Yi (& co) changed the structure of the file /tmp/mmap.info
And I'm not able to extract the serial number from the file.
I need help to find the offset and fix the problem.
Please, read the file with hexdump and send me the offset where serial number starts.

`hexdump -C /tmp/mmap.info`

For example, for old models like y21ga is 656.
For h60ga is 784.

roleoroleo, 2021-09-15:
y211ga
9.0.36.00_202011101611
656

roleoroleo, 2021-09-15:
h60ga
9.0.27.19_202105131533
784

roleoroleo, 2021-09-15:
h30ga
unknown firmware - https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/232
784

roleoroleo, 2021-09-15:
y211ga
9.0.36.01_202103231716
784

0-Al, 2021-09-16:
For b091qp and firmware version 9.0.41.99_202103111513 is same as h60ga, haven't looked at /tmp/mmap.info, will check later.

Mkampff, 2021-10-04:
my camera yi dome guard yrs3019 IFCNY36VR6

00000180  00 00 00 00 00 00 00 00  00 00 00 00 66 61 6d 69  |............fami|
00000190  6c 79 6d 6f 6e 69 74 6f  72 2d 72 33 30 67 62 00  |lymonitor-r30gb.|
000001a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001c0  00 00 00 00 00 00 00 00  00 00 00 00 61 77 73 00  |............aws.|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 00 00  00 00 00 00 04 75 00 00  |.............u..|
00000210  39 2e 30 2e 30 35 2e 33  33 5f 32 30 32 30 30 39  |9.0.05.33_202009|
00000220  32 35 31 38 35 36 00 00  00 00 00 00 00 00 00 00  |251856..........|
00000230  31 31 31 35 31 30 30 33  31 35 35 30 30 31 30 30  |1115100315500100|
00000240  30 30 30 30 30 30 30 30  30 30 30 30 30 30 30 30  |0000000000000000|

Mkampff, 2021-10-04:
I cant conect to yi home app, with chinesse app: error this camera can not be used outside of china. with normal app no error but cant conect.

kiddyfurby, 2022-01-06:
y29ga, 9.0.09.06_202103291039
784

y29ga, 9.0.09.03_202006090939
656

1-Al, 2022-01-31:
Just a thought. What about extracting the serial number from the manufacturing (mfg) partition instead ?

MFG_PART=$(grep  -oE  ".{0,0}mfg@.{0,9}" /sys/firmware/devicetree/base/chosen/bootargs | cut -c 5-14)
SERIAL_NUMBER=$(dd bs=1 count=36 skip=36 if=/dev/$MFG_PART 2>/dev/null | tr '\0' '0' | cut -c1-20)

Sorry if the syntax is disturbing, but hopefully provides an example?

roleoroleo, 2022-02-01:
I need to check my mtdblock archive.
Do you think it's in the same position for all models?

1-Al, 2022-02-02:
> Do you think it's in the same position for all models?

It does for my cameras but cannot say if it does for all. Even if it doesn't, at least the mfg partition should stay the same (I would hope) when updates are applied.

roleoroleo, 2022-02-02:
I checked "my archive".
For all backups made before setting up the cam the block is empty.
For all other files, the offset is the same as you wrote above.
I can extract the serial number in this manner.

xorguy, 2022-02-21:
YiCam Outdoor (IFUSY...) h30ga with firmware 9.0.26.07_202104231123 --> 310

vielfarbig, 2022-09-20:
Yi Dome Guard YRS r35gb with firmware 9.0.05.33_202009 --> 210

closed 2022-12-21 by roleoroleo