roleoroleo/yi-hack-Allwinner-v2 #260: Can not hack/ upgrade serial LFUSY6073L

Luxjacek, 2021-09-16:
Please advice am I doing something wrong. I extract tha latest version to a root folder, and switching the cam on. But nothing is changing. Is my cam hackable ? FCC ID: 2AFIB-YHS6020. I would appreciate.

roleoroleo, 2021-09-16:
It depends on the model.
What's the firmware version?

Luxjacek, 2021-09-16:
Hi, 
Its 9.0.27.20_202108161046

Luxjacek, 2021-09-17:
On SD Card should there be h60ga_0.1.9.tar or should i extract this file to the root ?

roleoroleo, 2021-09-18:
I updated the readme with more details.

Luxjacek, 2021-09-18:
It works ! Thank You ! But, what should i do, because after reboot, the cam is back to factory each time?

roleoroleo, 2021-09-19:
Did you remove the sd card?

Luxjacek, 2021-09-23:
Dear roleoroleo, 
No, I do everything like in readme. And still the cam gets back to fatory after update your hack. Its like when I put fresh card with your hack, tha cam starts, i get the website view of your software, but after 1 minute or so, i hear a click in the cam and it means the cam restarts, and than i have again factory .... It is very irritating. I tryed some like 50 times....

roleoroleo, 2021-09-24:
Maybe the sd card is not fully compatible.
If a restart occurs, the card is not recognized.
2 hints:

- Change the sd card.
- Enable swap space from web interface when the hack is running.

alexfranke, 2021-09-24:
This also happened to me with a new SD card. I realized that the 64GB SD card was formatted as exFAT and not FAT32. Windows is unable to format some larger cards as FAT32, so I used a Linux machine to do it. Is it possible that the card might be formatted wrong?

roleoroleo, 2021-09-24:
Yes, it's possible.
Try a smaller card, if you can.

Luxjacek, 2021-09-24:
So, thank you ! It works ! Problem was, probably in swap space on a card. In the short ammount of time after insserting a card a switched to enable swap on card, and than everything was ok. Thanks again great work you have done.

closed 2023-01-08 by stale[bot]