roleoroleo/yi-hack-Allwinner-v2 #263: LW-OSC202Y Yi Iot Cam

crispyduck00, 2021-09-17:
Hi,

I just got the following cam and thought it would maybe be possible to use yi-hack on it:
[LW-OSC202Y](https://www.yiiothome.com/products/cm-osc252d.html) 

The Chip is a QG2101B, I have already done a backup. The whole SD content from the backup is attached.
[cam backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7185891/cam.backup.zip)

Would be great if this cam would also support the hack.

FW Version is 9.0.41....

Br,
crispyduck

0-Al, 2021-09-17:
Yep. It's same number as mine 'b091qp'. Only one rf connection for the 2 antenna, the other is just an open wire. The box I got had 3MP on the label but was advertised as 2MP however the camera sensor is JX-H63 which is 1MP and that's interpolated to 1920x1080 at about 10fps!. It cost around 16 euro so cant complain but wish they had just used the native resolution. As it is I've been trying to compile v4l2-ctl for it but ended up trashing Debian that I used to compile the firmware. Never mind though, Roleo is on the ball and has already posted a hack for it under the discussion thread. 

The video setting I used in case it's of use.
`#define BUF_OFFSET_B091QP 300
#define BUF_SIZE_B091QP 524588
#define FRAME_HEADER_SIZE_B091QP 26
#define DATA_OFFSET_B091QP 4
#define LOWRES_BYTE_B091QP 8
#define HIGHRES_BYTE_B091QP 4
#define SPS_TYPE_B091QP 1`

roleoroleo, 2021-09-18:
I'm working on it.

roleoroleo, 2021-09-20:
Please check this discussion:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/249

closed 2023-01-08 by stale[bot]