roleoroleo/yi-hack-Allwinner-v2 #264: Help identifying the TTL pads in Yi Dome Pro 2K (H60GA)

ngutman, 2021-09-20:
Not sure how but my camera got bricked (not loading with/without SD card, status light stuck on orange), I did a backup before but I'm having trouble identifying the TTL uart pads to try and unbrick it (awesome project!!!)

I attached several photos of the board, I couldn't find TX/RX/GND labels anywhere :/

[image]
[image]
[image]

roleoroleo, 2021-09-20:
Same problem here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/254

ngutman, 2021-09-20:
Thanks I’ll close this issue and update there. Getting a logic analyzer so hopefully I’ll identify the pads soon.