roleoroleo/yi-hack-Allwinner-v2 #265: Dome U Pro - LFUS - MQTT sound_detection not working

egditch, 2021-09-20:
Hi, I've got a Yi Dome U Pro 2K LFUS with firmware 0.1.9 (Base: 9.0.27.19_202105131533).

I've had some truble making the MQTT work, i'm trying to get the sound_detection messages but the cam does't publish them.
I've set every other mqtt retention flag to 0 but i'm still getting all the other messages except for the sound_detection and the baby_crying one.

I've had the same problem on the previous version of the firmware ( 0.1.8 ) and i've already tried reformatting the sd and "clean-flashing" the hack

In the RTSP stream I can hear sounds.

Anyway here is my config:

```
camera.conf
SWITCH_ON=yes
SAVE_VIDEO_ON_MOTION=no
SENSITIVITY=low
AI_HUMAN_DETECTION=no
SOUND_DETECTION=yes
SOUND_SENSITIVITY=80
LED=no
ROTATE=no
IR=yes
```

```
###############################################################################
#                       Configuration file for mqttv4                         #
###############################################################################

# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=192.168.0.252
MQTT_PORT=1883

MQTT_CLIENT_ID=yi-cam

MQTT_USER=
MQTT_PASSWORD=

MQTT_PREFIX=yicam
MQTT_PREFIX_CONFIG=

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_BIRTH_WILL=status
TOPIC_AI_HUMAN_DETECTION=ai_human_detection
TOPIC_MOTION=motion_detection
TOPIC_MOTION_IMAGE=motion_detection_image
TOPIC_MOTION_FILES=motion_files
TOPIC_BABY_CRYING=baby_crying
TOPIC_SOUND_DETECTION=sound_detection

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

BIRTH_MSG=online
WILL_MSG=offline
AI_HUMAN_DETECTION_START_MSG=human_start
AI_HUMAN_DETECTION_STOP_MSG=human_stop
MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop
BABY_CRYING_MSG=crying
SOUND_DETECTION_MSG=sound

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN_BIRTH_WILL=0
MQTT_RETAIN_AI_HUMAN_DETECTION=0
MQTT_RETAIN_MOTION=0
MQTT_RETAIN_MOTION_IMAGE=0
MQTT_RETAIN_MOTION_FILES=0
MQTT_RETAIN_BABY_CRYING=1
MQTT_RETAIN_SOUND_DETECTION=1
```

```
system.conf
HTTPD=yes
TELNETD=no
SSHD=no
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=no
MQTT=yes
RTSP=yes
RTSP_STREAM=low
RTSP_AUDIO=alaw
SPEAKER_AUDIO=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=low
ONVIF_NETIF=wlan0
ONVIF_WM_SNAPSHOT=no
NTPD=no
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
```

Maybe i'm dumb and i've configured it in the wrong way, any suggestions?

Cheers

roleoroleo, 2021-09-22:
I will check it.
Meantime set all retain=0 except MQTT_RETAIN_BIRTH_WILL.

egditch, 2021-09-22:
Set it like this:

```
# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN_BIRTH_WILL=1
MQTT_RETAIN_AI_HUMAN_DETECTION=0
MQTT_RETAIN_MOTION=0
MQTT_RETAIN_MOTION_IMAGE=0
MQTT_RETAIN_MOTION_FILES=0
MQTT_RETAIN_BABY_CRYING=0
MQTT_RETAIN_SOUND_DETECTION=0
```

Rebooted the cam, MQTT explorer still reports:
ai_human_detection
motion_detection
motion_detection_image
status

Let me know if i can do anything else!

Cheers

WalkingDot, 2021-09-23:
Hi,
is the normal sound detection wotking? (record video on sound event)

egditch, 2021-09-23:
> 
> 
> Hi,
> is the normal sound detection working? (record video on sound event)

I can't find that option in the web page, I only got record video on motion, is it in the official app? Cause I don't have it

WalkingDot, 2021-09-23:
> I can't find that option in the web page

Sorry, the exact name is "Sound Detection" and you can find it at "Camera Settings".
The testing is a little more difficult... because the sound trigger needs a 10 minute cooldown.

egditch, 2021-09-24:
The testing was pretty difficult so I might've fucked it up but from my trials the record on sound detection doesn't seem to be working either.

I enabled "Recording without Cloud", "Save video when a motion is detected" and "Sound Detection", rebooted the cam making it face a wall, waited 10 min, made a lot of noise and then checked with "/cgi-bin/eventsdir.sh" wich showed no directories.

After that I wiggled a pen in front of the cam and checked again with "/cgi-bin/eventsdir.sh" wich showed a directory with the videos of the wiggling pen.

WalkingDot, 2021-09-25:
It could be possible that your camera has a rare sound detection problem.
You could test if sound events are detected with the app or if you are familiar with ssh use [ipc_sniff](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6665498/ipc_sniff.gz).
A code should then appear at a sound event (e.g.: 04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00)

egditch, 2021-09-27:
Sorry for the delay in the response, tried with the ipc_sniff was getting "Segmentation fault" so I tried things you posted here: #210

I got a lot of messages like:

```
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 02 00 00 00 60 00 80 00 00 00 00 00
```

Some of the were longer (Len: 20), but I couldn't know what was a sound one so to be sure I downloaded the app and tried with it, sound detection is working within the app, I get notifications of sound alerts!

roleoroleo, 2021-09-27:
I tested now my cam (y21ga) and I can confirm that sound detection is working.
Try to disable baby crying (in the app) and try to set sound sensitivity to 50.

egditch, 2022-10-03:
Updated today to 0.2.3 (Base: 9.0.27.19_202105131533), baby crying is off and sound sensitivity set to 50 in the app, sadly nothing changed, MQTT sound_detection not working.

roleoroleo, 2022-10-05:
Try to run mqttv4 manually and check if the event is detected:
```
root@yi-hack-allwinner-v2:~# . /tmp/sd/yi-hack/script/env.sh
root@yi-hack-allwinner-v2:~# killall mqttv4
root@yi-hack-allwinner-v2:~# /tmp/sd/yi-hack/bin/mqttv4
```

roleoroleo, 2022-10-06:
Please, try this library.
[ipc_multiplex.so.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9722773/ipc_multiplex.so.gz)
Uncompress it to /tmp/sd/yi-hack/lib and reboot the cam.

egditch, 2022-10-07:
No event is detected both before and after switching the library.

I only get:

"Client yi-cam sendind PUBLISH (... , 'yicam/status' , ...)"
"Client yi-cam sendind PUBLISH (... , 'yicam/motion_detection' , ...)"
"Client yi-cam sendind PUBLISH (... , 'yicam/motion_detection_image' , ...)"
"Client yi-cam sendind PUBLISH (... , 'yicam/motion_files' , ...)"

even though motion detection is disabled in the official app

roleoroleo, 2022-10-16:
I checked again the fix and it seems working.
Are you sure that your cam supports sound detection?
Do you receive this type of event in the app?

closed 2023-02-13 by stale[bot]