roleoroleo/yi-hack-Allwinner-v2 #266: How can I do a timelapse with yi hack?

kaizenGG, 2021-09-21:
I am currently using: yi-hack-Allwinner-v2 0.1.4

And I would like to make a timelapse of some plants. I found this post: https://github.com/roleoroleo/yi-hack-MStar/issues/291

But it didn't help me, maybe I did something wrong. I'm not sure. I added this to the crontab:

#! / bin / sh

           date >> / tmp / sd / cron_test
sleep 10; date >> / tmp / sd / cron_test
sleep 20; date >> / tmp / sd / cron_test
sleep 30; date >> / tmp / sd / cron_test
sleep 40; date >> / tmp / sd / cron_test
sleep 50; date >> / tmp / sd / cron_test

but no he save nothing

roleoroleo, 2021-09-21:
You have to create a script with the content you want to run.
And then add this script to the crontab.
```
* * * * * /tmp/sd/yi-hack/my_script.sh
```

kaizenGG, 2021-09-21:
would that be okay? I have it in this path: /yi-hack/my_script.sh

# crontab -l
# m h  dom mon dow   command
* * * * * sleep 5; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 10; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 15; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 20; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 25; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 30; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 35; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 40; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 45; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 50; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 55; date >> /tmp/yi-hack/my_script.sh
* * * * * sleep 60; date >> /tmp/yi-hack/my_script.sh

roleoroleo, 2021-09-22:
Use absolute path in the crontab box:
```
* * * * * /tmp/sd/yi-hack/my_script.sh
```

Jon8RFC, 2021-12-05:
Yi Home 1080p
IFUS
9.0.36.02_202105081647
y211ga

I, too, am struggling to get it to function in that manner, and I'm also not a linux guy.
It will screenshot once per minute, on the minute (why is that?), if I put this into the crontab box:
``* * * * * /tmp/sd/yi-hack/bin/imggrabber -r high -w > /tmp/sd/screenshots/`/bin/date +'%Y-%m-%d_%H.%M.%S'`.jpg``

But, even if I strip down a .sh to just:
```
#!/bin/sh

/tmp/sd/yi-hack/bin/imggrabber -r high -w > /tmp/sd/screenshots/`/bin/date +'%Y-%m-%d_%H.%M.%S'`.jpg
```
And put into the crontab box:
`* * * * * /tmp/sd/yi-hack/timelapse_simple.sh`

It will not function.

Sorry to ask for tech support on this, but thank you for helping all of us!

closed 2023-01-08 by stale[bot]