roleoroleo/yi-hack-Allwinner-v2 #268: Support Request: Yi/Xiaoyi Camera Y4 (1296p)

tony1ee, 2021-09-29:
I have a Y4 in hand, it seems to have the same model# as the KIMI MINI (YYS.2919)

roleoroleo, 2021-09-29:
What's the fw version?
9.0.*?

tony1ee, 2021-09-29:
I don't know. Can't set it up because it won't allow me to pair with my phone. "this camera can only be used within China".

tony1ee, 2021-09-29:
> What's the fw version?
> 9.0.*?

I managed to activate it. It has the firmware version of 9.0.09.06 202103291039

roleoroleo, 2021-09-30:
It's not supported at the moment.
I could take a look but I need a backup.

kiddyfurby, 2021-10-30:
Thank you so much, here's 
[fshare_frame_buf](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7445990/fshare_frame_buf.zip)

roleoroleo, 2021-10-30:
The file is not correct.
It contains only "00".

kiddyfurby, 2021-10-30:
Not quite sure why, did it again and verified its not 00 this time...
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7447299/fshare_frame_buf.zip)

roleoroleo, 2021-10-31:
Ok, try this:
[y29ga_0.1.9.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7448535/y29ga_0.1.9.tar.gz)

kiddyfurby, 2021-10-31:
No, RTSP is not working. Any log file I should help gather?

roleoroleo, 2021-10-31:
Both? Low and high res?
Check the environment of the process.
`cat /proc/<PID>/environ`

kiddyfurby, 2021-10-31:
Everything worked, for once.  High res, low res, audio, snapshot...

When I said it didn't work I was referring to the high res stream, but I also noticed that snapshot wasn't working
After seeing your comment, I tested it again to confirm if low res is working or not, with the sane power supply, usb cable, Wi-Fi network, and it worked this time.

But then I have a new issue: it works intermittently.
I am trying to find a way to reproduce the issue 100%.
For now I know that if I do not hear the IR cut sound, the cam is never going to connect to Wi-Fi and the LED is going to stay solid yellow forever.

roleoroleo, 2021-10-31:
Are you using the original power adapter and cable?

kiddyfurby, 2021-11-01:
No. I was powering it via my Macbook.
I have the original power adapter but I am no longer sure which cable is original.

After testing for 20+ times, the only sensible conclusion I can draw is I have intermittent cablesssss.
As soon as the LED turn blue, it is safe to bend the cable and move the camera around.

kiddyfurby, 2021-11-01:
Thank you very much roleoroleo
It appears everything is working when one has the right cable!

roleoroleo, 2021-11-01:
Good news!
I will release it asap.

kiddyfurby, 2022-01-02:
Is there any chance we can have a unhack file for y29ga?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Remove-the-hack

roleoroleo, 2022-01-07:
Yes.
Already pushed:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

hengfeb, 2022-03-18:
Hi All,
<<If you want to use the original Yi app, please install it and complete the pairing process before installing the hack.>>

my camera is  Yi Home Y4 IFCN | 9.0.09* | y29ga
I always cant compelete the pairing, help me!

Henry

hengfeb, 2022-03-19:
Hi All, 
How to can let compelete the pairing process?
(Yi home Y4 IFCN camera)

hengfeb, 2022-03-19:
[image]

hengfeb, 2022-03-19:
[image]
[image]

roleoroleo, 2022-03-19:
Are you using the correct app?
Your serial number is chinese.

hengfeb, 2022-03-19:
<Are you using the correct app?>
No, this is limmit to region for chinese area,
Other area will cant success on pairing.

So, have the way to hack the limmit?

henry

hengfeb, 2022-03-19:
Like as “ I don't know. Can't set it up because it won't allow me to pair with my phone. "this camera can only be used within China".”

I dont know how to he can fix this problem?

hengfeb, 2022-03-19:
@kiddyfurby @tony1ee ,
Could you tell how to can success on pairing? 
( I always got the message "This camera can only be used in China.".

Help me, plz.

roleoroleo, 2022-03-20:
You could try to use proxychains.
It's installed on the cam but you need to configure it manually.

hengfeb, 2022-03-21:
@roleoroleo ,
<<You could try to use proxychains，you need to configure and run it manually.>>
You mean I need to connect serail console to run the script manually?
Could you tell me what the proceduse can let me to follow the steps?

Henry

hengfeb, 2022-03-21:
@roleoroleo,

My problem is I cant to pair my camera device，so I cant hack to anything.
Do you can teach me, to pair my device manually? I will donate to you! 
because I have four different type camera with the same problem ( cant to pair normally).

Henry

roleoroleo, 2022-03-21:
Read the wiki:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Install-the-hack-without-the-Yi-app

hengfeb, 2022-03-21:
1) Install the hack as described on the main page of this repo.
　>> How I know my camera has been hacked?
2)Follow the wiki https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials to connect the cam to the wifi network
  >> How I know my carmera has been paired?

Henry

roleoroleo, 2022-03-21:
If you reach the web page of the cam, it's hacked.
To check if the cam is paired, open the list of connected client in your router.

hengfeb, 2022-03-21:
@roleoroleo 
<<If you reach the web page of the cam, it's hacked.>>
The camera cant on pairing, so how I can reach the web page find it?
It must be paired, so I can find it by the web page, right?
So, the step 1 has the problem.

Henry

hengfeb, 2022-03-21:
@roleoroleo 
for step 1:

I download the hack file "y29ga_0.2.1", and put  these to one empty sd card, and insert to my camera，and turn "ON".
then I search my router is nothing for "yi-hack".
if I  wont to use the original Yi app, the script how to know which my wifi and my password? 
Henry

roleoroleo, 2022-03-21:
The hack is not enough to connect to your wifi.
After the hack you need to follow this:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials

- Remove the sd card and clean it.
- Follow the wiki about wifi credentials.
- Create again the sd card with the content of y29ga_0.2.1.

hengfeb, 2022-03-21:
Ok, I make a mistake, I try it now.

hengfeb, 2022-03-21:
I create one folder "yi-hack" and put that three files in that folder.

roleoroleo, 2022-03-21:
Sorry but I gave you the wrong instructions.

After the hack, follow the wiki (https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials) WITHOUT cleaning the sd card.
So add the three files to the sd card where the hack is already there.

hengfeb, 2022-03-21:
I will three add the "yi-hack" folder, then insert my camera device, and turn "ON".
the same result, my router is not "yi-hack" appear.

hengfeb, 2022-03-21:
https://controsensi.blogspot.com/2019/08/solved-xiaomi-xiao-yi-home-cn12-this.html

roleoroleo, 2022-03-22:
This thread refers to another model. You cannot apply this method to your cam.
Let's go one step at a time.
Check if the cam is hacked.
If you remove the sd card from the cam, is there Factory.done folder?

hengfeb, 2022-03-22:
<<If you remove the sd card from the cam, is there Factory.done folder? >>
Yes, I confirm. I  saw it "Factory.done".

roleoroleo, 2022-03-22:
So, the hack worked.
Now you have to change wifi configuration.
Try again to copy the 3 files in the yi-hack folder of your sd card.
Edit the file configure_wifi.cfg to your needs (wifi user and password).
Overwrite startup.sh with this content, so we can see the log:
```
#!/bin/sh

/tmp/sd/yi-hack/configure_wifi.sh > /tmp/sd/yi-hack/configure_wifi.log
```

hengfeb, 2022-03-22:
I want to donate 10 EUR dollars to you by paypal, but I got the message "No support donate at this area".
other account I can transfer to you?

Henry

hengfeb, 2022-03-22:
@roleoroleo 

I excute you say procedure about three times, and get nothing log file to generate it.
see my photo as the following:
[image]

roleoroleo, 2022-03-22:
Maybe the file are in windows format (I mean CF-LF) and are not executed when you start the cam.
What compression program did you use to decompress the archive?
Try 7zip.
Check and edit them converting to LF only. For example, you could use notepad++ and set "Unix (LF)" in the bottom-right corner.
[image]

roleoroleo, 2022-03-22:
> I want to donate 10 EUR dollars to you by paypal, but I got the message "No support donate at this area". other account I can transfer to you?
> 

No sorry, but don't worry.

hengfeb, 2022-03-22:
OK, I try it again.

hengfeb, 2022-03-22:
@roleoroleo 
Redo the procedure, and I confirm the file format is only "LF".
the result is the same, nothing log file to generate. see the photo as the followings.
[image]

roleoroleo, 2022-03-23:
I noticed a problem.
In your sd there is the file fw_upgrade_in_progress.
This means that the hack installation doesn't compete successfully.
Please check if the file lower_half_init.sh is ok about LF and content.

hengfeb, 2022-03-23:
@roleoroleo 
<<In your sd there is the file fw_upgrade_in_progress >>
yes, I redo it again, the result is the same "fw_upgrade_in_progress",

<<Please check if the file lower_half_init.sh is ok about LF and content.>>
Of course, I sure all script is "Only LF" format, see the above photos.

Henry

hengfeb, 2022-03-23:
@roleoroleo 
<<In your sd there is the file fw_upgrade_in_progress >>
yes, I redo it again, the result is the same "fw_upgrade_in_progress",

<<Please check if the file lower_half_init.sh is ok about LF and content.>>
Of course, I sure all script is "Only LF" format, see the above photos.

Henry[
[log_wifi_connected.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8335822/log_wifi_connected.tar.gz)
](url)

hengfeb, 2022-03-23:
@roleoroleo 
<In your sd there is the file fw_upgrade_in_progress >
yes, I redo it again, the result is the same "fw_upgrade_in_progress",

<Please check if the file lower_half_init.sh is ok about LF and content.
[log_wifi_connected.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8335829/log_wifi_connected.tar.gz)
>
Of course, I sure all script is "Only LF" format, see the above photos.

Henry

hengfeb, 2022-03-23:
@roleoroleo 

I back to see my hacked camera "b091qp", the same as "fw_upgrade_in_progress" in the yi-hack folder.
but I saw the some log files on it.

Henry
[log_bind_success.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8336088/log_bind_success.tar.gz)
[log_first_login.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8336090/log_first_login.tar.gz)
[log_login.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8336091/log_login.tar.gz)
[log_no_wifi_dev.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8336092/log_no_wifi_dev.tar.gz)
[log_start_reset.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8336093/log_start_reset.tar.gz)
[log_wifi_connected.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8336094/log_wifi_connected.tar.gz)

roleoroleo, 2022-03-23:
The file wpa_supplicant.conf contained in the log is configured properly.
I think that the connection with your wifi is ok.
Try to scan your local network to find the ip of your cam.

hengfeb, 2022-03-23:
@roleoroleo 
No, I use the orginal app to generate it.
I think just one way, I connect the serial console of the comera, and give you remote to my computer to try it, I dont know as if you will to do it?
Or I do it myself first.

Henry

roleoroleo, 2022-03-25:
> I think just one way, I connect the serial console of the comera, and give you remote to my computer to try it, I dont know as if you will to do it? 

Ok.

hengfeb, 2022-03-28:
@roleoroleo 
I make a serial connection for you.
could you give me a mail, let me tell you anydesk account and pwd.

[Henry](
[YI_home_hack2.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8360329/YI_home_hack2.log)
[YI_home_hack.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8360327/YI_home_hack.log))

roleoroleo, 2022-03-28:
The log shows:
```
mount: mounting /dev/mmcblk0p1 on /tmp/sd failed: No such file or directory
---update file(home_y29gam) Not exist---
mount: mounting /dev/mmcblk0 on /tmp/sd failed: No such file or directory
```

It doesn't mount the sd card.
Unsupported sd card?
Wrong partition/format?

hengfeb, 2022-03-28:
@roleoroleo 
<< It doesn't mount the sd card. >>
If it does, but I run "Dump your backup firmware" script, It's done(no problem), why?
I have the backup folder on SD card.
Henry

roleoroleo, 2022-03-28:
For other people who may have the same problem.
The SD was not properly formatted and probably the cam took a long time to mount it.
SD repartitioned and reformatted using a linux machine.
Now it works.

[6 more replies]

closed 2023-01-08 by stale[bot]