roleoroleo/yi-hack-Allwinner-v2 #269: Building Yi-Hack-Allwinner-v2 with Fedora 34

0-Al, 2021-10-05:
Dependencies needed for Fedora 34:
autoconf
automake
patch
openssl-devel
zlib-devel
bison
flex
libtool
yuicompressor

Note yui-compressor name is used in the Yi-Hack source so modify as necessary, example "sudo ln /bin/yuicompressor /bin/yui-compressor" or by other means, whatever is suitable.

Bug in gsoap 2.8.92 used by Yi Hack source 0.1.9. Note gsoap.zip is downloaded twice during compiling! Can fix by modifying soapcpp2_lex.l line 53 to **extern YYSTYPE yylval;** as per [https://sourceforge.net/p/gsoap2/bugs/1269/](url) or can a later fixed version of gsoap be used?

roleoroleo, 2021-10-06:
> Bug in gsoap 2.8.92 used by Yi Hack source 0.1.9. Note gsoap.zip is downloaded twice during compiling! Can fix by modifying soapcpp2_lex.l line 53 to **extern YYSTYPE yylval;** as per [https://sourceforge.net/p/gsoap2/bugs/1269/](url) or can a later fixed version of gsoap be used?

Do you mean "downloaded twice", 1st for onvif and 2nd for wsdd?
I tried to update gsoap to a newer version but I can't compile it. I have a lot of errors during linking.

0-Al, 2021-10-07:
Yes, like that but probably 3 times including alsa-lib? Makes it awkward for us non-linux people to try and fix problems, well did for me at any rate. I posted this mainly in the hope it might at least help someone who might be running into the same issues. 

Thanks for the heads up with later gsoap versions.

closed 2021-10-07 by 0-Al