roleoroleo/yi-hack-Allwinner-v2 #281: Can you make the same for Escam pt202????

carlosvelosa, 2021-10-20:
Can you make the same for Escam pt202????

roleoroleo, 2021-10-20:
I don't know.
If it's based on the same platform, I could try.
Please, take  some pictures of the board.

roleoroleo, 2021-10-30:
I will add to the next release.

roobbb, 2021-12-23:
Hello.
I even bought one and hacked it successfully. It works very well. Thank you very much!
One question: does the performance advice even apply to this device?

roleoroleo, 2021-12-24:
> One question: does the performance advice even apply to this device?

Yes, It's the same cpu/ram architecture.

kirch-cam, 2022-01-22:
I have an almost identical (unbranded) model, the only difference is the wifi module
Main board: QG2101b and xmc chip
B102QP KB V1 board has Realtek chip instead of SV6255P. The Realtek wifi chip is marked 8188FTV  KCF03G1 GK53Q1
Firmware is 9.0.41.99_202103301400
I assume they use the wifi module just as a serial modem, so the b091qp should work?
[image]
[image]

roleoroleo, 2022-01-22:
I don't know if it works.
Send me a backup copy and I will check.

kirch-cam, 2022-01-22:
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7918751/backup.zip)

roleoroleo, 2022-01-23:
It's different from other supported b091qp.
I could try to support your cam but I need some time.

kirch-cam, 2022-01-23:
Thank you very much. That would be great.

GqSteve, 2022-04-03:
@carlosvelosa Is the hack working also, when the cam is connected through the network interface?

carlosvelosa, 2022-04-03:
> @carlosvelosa Is the hack working also, when the cam is connected through the network interface?

Yes it's working without problems @GqSteve

GqSteve, 2022-04-07:
@carlosvelosa I set up one with WiFi. I connected the camera per wire and disabled the WiFi AP. The cam is not requesting any IP from DHCP over the wired connection. How can I get it working? Btw, if the wired connection is going to work, how can I disable the WiFi on the cam?

roobbb, 2022-04-07:
@GqSteve 
I started with LAN first and had to pair my cam with my LAN via the app. But was not able to connect it to Wifi. Therefore I paired my Cam again with Wifi via app. Both connections are working fine since. My Cam has two MAC-ID's and separate IP's.
Maybe you could get it work with LAN if you try to pair your Cam with LAN too. 
This "Cable" option is very small and easy to miss 
[image]
Maybe it's worth a try 😀

GqSteve, 2022-04-07:
@roobbb I bought 2 cameras from banggood and none of them is retrieving an IP from DHCP. I tried the wired connection with two different networks so far. The manual refers to Yi IoT app.  Maybe is the cam a fake one?!

Which app do you use? Where did you buy yours?

GqSteve, 2022-04-07:
My bad...the network interface supports 100M port speed...my port was fixed to 1000M...grrrrr
But I can confirm that the hack works! Thanks!

napalmpt, 2022-04-15:
Hi. I have one of these cams that somehow got bricked. it now only lights up the led and stays like that. I followed instructions for soft recovery, converted the backup mtdblock4.bin into backup_b091qp but didn't work. I tried to find the rs232 pads, but no luck or nothing happens on boot... should I return it?

orion131313, 2022-04-22:
> > @carlosvelosa Is the hack working also, when the cam is connected through the network interface?
> 
> Yes it's working without problems @GqSteve

Yi YoT android app still working?  Are there any better apps?

kirch-cam, 2022-08-10:
> It's different from other supported b091qp. I could try to support your cam but I need some time.
@roleoroleo have you had a chance yet to support it?

roleoroleo, 2022-08-17:
Yes, I think it's possible.
Please, send me 3-4 copies of the file /dev/shm/fshare_frame_buf

closed 2023-01-07 by stale[bot]

elsie940313, 2023-04-30:
Good day guys!

I have the exact same camera ( b091qp) as @carlosvelosa but flashed it with the the wrong firmware (h51ga) , as seen below in the "init.sh" file:

root@(none):/backup# cat init.sh
#!/bin/sh
echo "--------------------------home app init.sh--------------------------"
**SUFFIX=h51ga**
enable_4g=

PATH="/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/ba
LD_LIBRARY_PATH="/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp
export PATH
export LD_LIBRARY_PATH

NETWORK_IFACE=

if [ "${enable_4g}" = "y" ];then
    NETWORK_IFACE=eth1
else
    NETWORK_IFACE=wlan0
fi

echo "NETWORK_IFACE=${NETWORK_IFACE}"

ulimit -c unlimited
ulimit -s 1024
#echo /tmp/sd/core.exe[%e].pid[%p].sig[%s] > /proc/sys/kernel/core_pattern

echo 100 > /proc/sys/vm/dirty_writeback_centisecs
echo 50 > /proc/sys/vm/dirty_expire_centisecs
#echo 2048 > /proc/sys/vm/extra_free_kbytes

echo 400 > /proc/sys/vm/vfs_cache_pressure
echo 0 > /proc/sys/vm/drop_caches
echo 256 > /proc/sys/vm/min_free_kbytes

sysctl -w vm.dirty_background_ratio=2
sysctl -w vm.dirty_expire_centisecs=500
sysctl -w vm.dirty_ratio=2
sysctl -w vm.dirty_writeback_centisecs=100

sysctl -w fs.mqueue.msg_max=256
mkdir /dev/mqueue
mount -t mqueue none /dev/mqueue

mount tmpfs /tmp -t tmpfs -o size=32m
mkdir /tmp/sd
mkdir /dev/shm
mkdir /tmp/var
mkdir /tmp/var/run
mkdir /tmp/run

checkdisk
rm -fr /tmp/sd/FSCK*.REC
umount -l /tmp/sd
mount -t vfat /dev/mmcblk0p1 /tmp/sd

#----insmod hardware cyption------

#extract files
mkdir /tmp/ko/
mkdir /tmp/app/
mkdir /tmp/tools/

if [ -f /home/home_${SUFFIX}m ]; then
    echo "---/home/home_${SUFFIX}m exist, update begin---"
        dd if=/home/home_${SUFFIX}m of=/tmp/newver bs=22 count=1
        newver=$(cat /tmp/newver)
        if [ -f /home/homever ]; then
                curver=$(cat /home/homever)
        else
                curver=0
        fi
        echo check version: newver=$newver, curver=$curver
        if [ $newver != $curver ]; then
                ### cipher ###
                sleep 1
                mkdir /tmp/update
                cp -rf /home/base/tools/extpkg.sh /tmp/update/extpkg.sh
                /tmp/update/extpkg.sh /home/home_v429m
                rm -rf /tmp/update
                rm -rf /home/home_${SUFFIX}m
                #sync
                echo "update finish"
                reboot -f
        fi
        echo "---same version ? update fail---"
        rm -rf mv /home/home_${SUFFIX}m
elif [ -f /tmp/sd/home_${SUFFIX}m ]; then
        echo "---tmp/sd/home_${SUFFIX}m exist, update begin---"
        dd if=/tmp/sd/home_${SUFFIX}m of=/tmp/newver bs=22 count=1
        newver=$(cat /tmp/newver)
        if [ -f /home/homever ]; then
                curver=$(cat /home/homever)
        else
                curver=0
        fi
        echo check version: newver=$newver, curver=$curver
        if [ $newver != $curver ]; then
                ### cipher ###
                sleep 1
                mkdir /tmp/update
        cp -rf /tmp/sd/home_${SUFFIX}m /tmp/update/
                cp -rf /backup/tools/extpkg.sh /tmp/update/extpkg.sh
                /tmp/update/extpkg.sh /tmp/sd/home_${SUFFIX}m
                rm -rf /tmp/update
                mv /tmp/sd/home_${SUFFIX}m      /tmp/sd/home_${SUFFIX}m.done
                sync
                echo "update finish"
                reboot -f
        fi
        echo "---same version ? update fail---"
        mv /tmp/sd/home_${SUFFIX}m      /tmp/sd/home_${SUFFIX}m.done
else
        echo "---update file(home_${SUFFIX}m) Not exist---"
fi

if [ -f /backup/url ];then

    if [ "${enable_4g}" = "y" ];then
        /home/app/4g &
        # sleep 40
    else
        #when ota upgrade failure, will enter this
        echo "--------------------------insmod sensor--------------------------"
        if [ -f /backup/ko/icplus.ko ];then
            insmod /backup/ko/icplus.ko
        fi

        if [ -f /backup/ko/8188fu.ko ];then
            insmod /backup/ko/8188fu.ko
        fi

        if [ -f /backup/ko/8189fs.ko ];then
            insmod /backup/ko/8189fs.ko
        fi
        sleep 1
        ifconfig lo up
        ifconfig ${NETWORK_IFACE} up
        ethmac=d2:`ifconfig ${NETWORK_IFACE} |grep HWaddr|cut -d' ' -f10|cut -d:
        ifconfig eth0 hw ether $ethmac
        ifconfig eth0 up
    fi

    echo "============================================= begin to upgrade...... =

Attached photo shows my cam I have as confirmation:  
[image]
[image]

Can someone please send me the stock firmware mtdblock4.bin backup file, as i don't have a backup one to unbrick my cam.

Thanks Guys

roleoroleo, 2023-05-01:
b091qp is used for different hardware models.
Try to make a backup copy (if the cam reaches the backup script during the boot) and send me your current mtdblock4.
I will create a new mtdblock4.bin for you.

roleoroleo, 2023-05-04:
Your mtdblock4 is completely wrong.
I don't understand...
Did you restore the mtdblock4 of the h51ga?
This should be the right file:
[backup_b091qp.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11394778/backup_b091qp.gz)

elsie940313, 2023-05-04:
I restored the backup you sent me. 

Good news is I can connect to the cam using the YI IOT app. 

Below is the current firmware version.
[image]

Should I try the b091 hack? 

Thanks for the help so far, much appreciated!

roleoroleo, 2023-05-04:
Yes, b091qp.

AditaNi, 2023-05-04:
can you please improve this firmware, bucase for example after one week of functioning i cannot logon onto camera. And in HA i have  alot of errors.
It will be very nice if you have some time to improve the b091qp firmware for pt202 camera.

elsie940313, 2023-05-04:
tried it, its not working, after loading the hack I tried "http://ip-cam:8080/" (replacing the ip-cam part with the ip assigned to the camera via my router.

I do have a serial coms with the board, connected straight to the serial pins on the board so i can troubleshoot if need be.

roleoroleo, 2023-05-05:
> I do have a serial coms with the board, connected straight to the serial pins on the board so i can troubleshoot if need be.

Yes, please.
Send me a log of the boot.

elsie940313, 2023-05-05:
Logged the boot sequence to the attached log.txt file as well as copied below.

156]HELLO! BOOT0 is starting!
[160]BOOT0 commit : aacdf0e
[162]board init start
[165]set pll start
[167]set pll end
[169][pmu]: bus read error
[171][pmu]: bus read error
[174]board init ok
[176]chip id check OK
[178]DRAM BOOT DRIVE INFO: V0.41
[181]DRAM CLK = 528 MHz
[183]DRAM Type = 2 (2:DDR2,3:DDR3)
[187]DRAMC read ODT  off.
[189]DRAM ODT off.
[191]DRAM SIZE =64 M
[199]DRAM simple test OK.
[201]rtc standby flag is 0x0, super standby flag is 0x0
[206]dram size =64
[210]spinor id is: 20 70 17, read cmd: 0b
[214]Succeed in reading toc file head.
[217]The size of toc is 4c000.
[305]Entry_name        = optee
[308]Entry_name        = u-boot
[312]Entry_name        = dtb
[316]Jump to second Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-27-g7f6d3375 #1 Thu Dec 10 03:28:16 UTC 2020 arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:127: no pmu node

U-Boot 2018.05 (Mar 11 2021 - 15:12:01 +0800) Allwinner Technology

[00.359]DRAM:  64 MiB
[00.361]Relocation Offset is: f9f94000
[00.373]secure enable bit: 0
[00.375]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.382]gic: sec monitor mode
[00.384]flash init start
[00.386]workmode = 0,storage type = 3
SF: Detected XM25QH64A with page size 256 Bytes, erase size 4 KiB, total 8 MiB
[00.403]sunxi flash init ok
[00.406]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.467]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.475]sprite_led_gpio start
[00.478](weak)update dtb dram start
[00.482]update dtb dram  end
[00.484]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
[... 296 lines cut ...]
dispatch.c(check_wifi_connect-3067)[20:00:43.468]:mask(255.255.255.0)

dispatch.c(check_wifi_connect-3077)[20:00:43.484]:mac(0C:CF:89:98:E0:B2)

dispatch.c(check_wifi_connect-3091)[20:00:43.561]:signal_quality(61)
cmd(rm /tmp/sd/log_wifi_connected.tar.gz;rm /tmp/sd/wifi_connected -r;mkdir /tmp/sd/wifi_connected;ps > /tmp/sd/wifi_connected/ps1.txt;cp /home/homever /tmp/sd/wifi_connected;lsmod > /tmp/sd/wifi_connected/lsmod.txt;ls -laSR /home > /tmp/sd/wifi_connected/file.txt;cp /tmp/wpa_supplicant.conf  /tmp/sd/wifi_connected;cp /tmp/log.txt /tmp/sd/wifi_connected;cp /tmp/log_oss.txt /tmp/sd/wifi_connected;df /home > /tmp/sd/wifi_connected/df.txt;df / >> /tmp/sd/wifi_connected/df.txt;tar -cf /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/wifi_connected;mkdir /tmp/sd/log; mv /tmp/sd/log_wifi_connected.tar.gz /tmp/sd/log;rm /tmp/sd/wifi_connected -r &)
rm: can't remove '/tmp/sd/log_wifi_connected.tar.gz': No such file or directory
rm: can't remove '/tmp/sd/wifi_connected': No such file or directory
cp: can't stat '/tmp/log.txt': No such file or directory
cp: can't stat '/tmp/log_oss.txt': No such file or directory
tar: removing leading '/' from member names
mkdir: can't create directory '/tmp/sd/log': File exists

dispatch.c(check_wifi_connect-3109)[20:00:44.863]:signal_quality(62)
now ptz_info----x:1000,y:0

ptz_ctl.c(ptz_stop-170)[20:00:47.264]:PTZ stop OK
now.y:0,dst_y:420

~~~~~~~dir = 2, speed = 10, run time = 8400~~~~~~

now ptz_info----x:1000,y:420
ptz_ctl.c(ptz_stop-170)[20:00:48.169]:PTZ stop OK
 
[log.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11410196/log.txt)

elsie940313, 2023-05-05:
[log.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11410200/log.txt)

roleoroleo, 2023-05-08:
> checkdisk.c(main-450)[00:00:05.615]:FS cluster size is too little. cluster_size = 8192 

Maybe the sd card is not properly formatted.

elsie940313, 2023-05-08:
I swapped out sd cards, does not give me the error anymore. 

I can seen the unit on the network but still no access to the webpage. 

MY LOG FOR THE NEW sd CARD:

[134]HELLO! BOOT0 is starting!
[137]BOOT0 commit : aacdf0e
[140]board init start
[142]set pll start
[144]set pll end
[146][pmu]: bus read error
[149][pmu]: bus read error
[151]board init ok
[153]chip id check OK
[155]DRAM BOOT DRIVE INFO: V0.41
[158]DRAM CLK = 528 MHz
[161]DRAM Type = 2 (2:DDR2,3:DDR3)
[164]DRAMC read ODT  off.
[166]DRAM ODT off.
[169]DRAM SIZE =64 M
[176]DRAM simple test OK.
[178]rtc standby flag is 0x0, super standby flag is 0x0
[184]dram size =64
[187]spinor id is: 20 70 17, read cmd: 0b
[191]Succeed in reading toc file head.
[194]The size of toc is 4c000.
[279]Entry_name        = optee
[282]Entry_name        = u-boot
[286]Entry_name        = dtb
[289]Jump to second Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-27-g7f6d3375 #1 Thu D
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:127: no pmu node

U-Boot 2018.05 (Mar 11 2021 - 15:12:01 +0800) Allwinner Technology

[00.333]DRAM:  64 MiB
[00.335]Relocation Offset is: f9f94000
[00.347]secure enable bit: 0
[00.349]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.356]gic: sec monitor mode
[00.358]flash init start
[00.360]workmode = 0,storage type = 3
SF: Detected XM25QH64A with page size 256 Bytes, erase size 4 KiB, total 8 MiB
[00.377]sunxi flash init ok
[00.380]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.441]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.449]sprite_led_gpio start
[00.452](weak)update dtb dram start
[00.456]update dtb dram  end
[00.459]update dts
root_partition is rootfs
[... 631 lines cut ...]
dispatch.c(check_wifi_connect-3091)[14:37:22.562]:signal_quality(63)
cmd(rm /tmp/sd/log_wifi_connected.tar.gz;rm /tmp/sd/wifi_connected -r;mkdir /tmp                                                     cp /tmp/log.txt /tmp/sd/wifi_connected;cp /tmp/log_oss.txt /tmp/sd/wifi_connecte                                                     ed -r &)
rm: can't remove '/tmp/sd/log_wifi_connected.tar.gz': No such file or directory
rm: can't remove '/tmp/sd/wifi_connected': No such file or directory
cp: can't stat '/tmp/log.txt': No such file or directory
cp: can't stat '/tmp/log_oss.txt': No such file or directory
tar: removing leading '/' from member names

dispatch.c(check_wifi_connect-3109)[14:37:23.863]:signal_quality(63)
now ptz_info----x:1000,y:0

ptz_ctl.c(ptz_stop-170)[14:37:26.589]:PTZ stop OK
now.y:0,dst_y:420

~~~~~~~dir = 2, speed = 10, run time = 8400~~~~~~

now ptz_info----x:1000,y:420

ptz_ctl.c(ptz_stop-170)[14:37:27.494]:PTZ stop OK
hue, ptz_x_runtime get (13) to (13)
hue, ptz_y_runtime get (3) to (3)

dispatch.c(p_worker-3865)[14:38:15.722]:rcd may crashed

killall: mp4record: no process killed

roleoroleo, 2023-05-14:
The hack process seems to be ok:
```
############## Starting Firmware Dump ##############
############## Starting Hack ##############
```
Are you using port 8080 to access the webpage?

roleoroleo, 2024-03-09:
I think it's not supported.
But try to make a backup.