roleoroleo/yi-hack-Allwinner-v2 #303: Auto Restart rRTSPServer

tunerooster, 2021-11-20:
I have had  rRTSPServer die occasionally (not often).  But it might be useful to start it in a loop so if it does die it automatically restarts.  ...just a minor feature improvement.  :)

Thanks for a job well done!

roleoroleo, 2021-11-20:
There should already be a watchdog: wd_rtsp.sh
Why doesn't it work in your scenario?

tunerooster, 2021-11-20:
Good question.  I'll investigate.

tunerooster, 2021-11-21:
wd_rtsp.sh was/is not running.  I rebooted and it is now running, but it apparently died in addition to rRTSPServer, thus no restart...

I have enabled the wd_rtsp.sh LOG messages.  The initial messages from the log are as follows:
```2021-11-21 09:23:59 - Starting RTSP watchdog...
2021-11-21 09:29:45 - No running processes, restarting...
2021-11-21 09:29:57 - Detected possible locked process (1)
2021-11-21 09:30:33 - No running processes, restarting...
```
If/when rRTSPServer is again not running, I will send you the log.

closed 2023-01-08 by stale[bot]