roleoroleo/yi-hack-Allwinner-v2 #304: Disassemble u-boot

roleoroleo, 2021-11-21:
During the boot phase, u-boot tries to load a file image to overwrite a flash partition.
It looks for:
- one_y21ga.bin (probably the whole image)
- boot_y21ga (kernel)
- rootfs_y21ga
- home_y21ga
- backup_y21ga

When I copy a file in the SD it tries to load it but without success.

In the first attempt I got an error about the magic, so I added a u-boot header.
And it seems to work better.
But now I have a "wrong type" error.
I think it refers to the type field in the header, but I can't find the right value.

Is anyone able to disassemble the raw binary and help me to find what header I have to prepare?

1-Al, 2021-11-22:
I don't have a suitable disassembler but did use mkimage for home and backup, seemed to work. Used on unknown Yi-iot camera.

Example:
  
`mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-home" -d /media/sf_FedShare/yihome.bin home_b091qp`

Where yihome.bin is the home partition binary and home_b091qp is the output file. This basically creates a 64-byte header for the partition. File was placed on root of FAT32 32GB SDCard.

Guess work so might not be fully correct, never checked if the xiaoyi names were actually necessary or  not.

Debug data for 'home' flash

[160]HELLO! BOOT0 is starting!
[163]BOOT0 commit : 4f02397
...
SF: Detected XM25QH64A with page size 256 Bytes, erase size 4 KiB, total 8 MiB
...
card_caps:0x3000000a
host_caps:0x3000003f
[00.955]unable to read ssr
[00.958]unable to read ssr
[00.960]the mode SD Legacy (freq : 25 MHz)
check_file [/one_b091qp.bin] exsit 
check_file [/one_b091qp.bin] exsit 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:boot start_offset:0x20 part_size:2031616 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:rootfs start_offset:0xfa0 part_size:1179648 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
**part_name:home start_offset:0x18a0 part_size:3145728 
 (11.3 MiB/s)
file_name /home_b091qp read image[3] size:3145792 
========== start: 0x1b80, len: 0x1800 ==============
_sunxi_flash_spinor_write 188 erase_size 4096
Saving Environment to SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
Writing to env...
========== start: 0x3e00, len: 0x100 ==============
_sunxi_flash_spinor_write 188 erase_size 4096
OK
update image[3] success
update finishing, will be reboot** 
[41]HELLO! BOOT0 is starting!
[44]BOOT0 commit : 4f02397
...
card_caps:0x3000000a
host_caps:0x3000003f
[00.838]unable to read ssr
[00.840]unable to read ssr
[00.843]the mode SD Legacy (freq : 25 MHz)
check_file [/one_b091qp.bin] exsit 
check_file [/one_b091qp.bin] exsit 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:boot start_offset:0x20 part_size:2031616 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:rootfs start_offset:0xfa0 part_size:1179648 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
**part_name:home start_offset:0x18a0 part_size:3145728 
 (11.3 MiB/s)
file_name /home_b091qp read image[3] size:3145792 
image[3] version same**
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
**part_name:backup start_offset:0x30a0 part_size:1376256 
 (11.1 MiB/s)
file_name /backup_b091qp read image[4] size:1376320 
image[4] version same**
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:env start_offset:0x3b20 part_size:131072 
Nothing updated, start normal boot
Hit ctrl + q  to stop autoboot:  1  0 
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
[02.373]partinfo: name boot, start 0x20, size 0xf80
start: 0x300, len: 0x40
start: 0x340, len: 0xf3e
[02.875]android.hardware = sun8iw19p1
## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2030224 Bytes = 1.9 MiB
   Load Address: 40008000
   Entry Point:  40008000
[02.943]Starting kernel ...

Note that the data checksum in the 64-byte header is added to the 'env' partition as a version number such as "backup_ver=2b62465a" which it uses as a check whether to update or not, so probably not a straight forward recovery method if that is what you intend. Also in this particular case the environment (env) name is back to front, backup is called home and home is called backup (bug).

What I did find more interesting is that the eGON loader at sector 16 will run straight from the SDCard but don't know about the chain loading after that.

Anyway, hope that helps a little.

roleoroleo, 2021-11-22:
Thank you.
Exactly what I'm trying to do.
I would like to use it as a recovery when the init.sh is broken.
About the checksum I could change a byte somewhere... :)

I will study your post and I will test it on my cam.

roleoroleo, 2021-11-22:
Mistery solved!
The name is very important. If you change "xiaoyi-home" it doesn't work.
Thank you very much, you helped a lot of users with a bricked cam.

closed 2021-11-22 by roleoroleo