roleoroleo/yi-hack-Allwinner-v2 #305: New unbrick procedure

roleoroleo, 2021-11-22:
Thanks to @1-AI I found a new unbrick procedure useful if your cam doesn't boot anymore.
Unlike the procedure described here, it doesn't require serial connection.
Let me know if you need it.

Kipper641, 2021-11-25:
Could do with the info please.

RandomItalianGuy, 2021-11-25:
I need it, i have some bricked cameras. Would also love a copy of the y21ga firmware to try and reflash them.

roleoroleo, 2021-11-25:
Take a look here: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/304
There is a method to overwrite a partition from the bootloader without starting the os.
If you prepare a file with the right name and the right content you can unbrick the cam.
Normally the brick happens because the backup partition is full and a file is truncated during a write (i.e. init.sh).
If you restore the backup partition the problem is solved.
To do this:
- take mtdblock4.bin from your backup
- apply a new header with this command (you need a linux os) `mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-backup" -d your_mtdblock4.bin backup_y21ga`
- copy the file backup_y21ga to the root folder of your sd card
- switch on the cam

Depending on your cam, change y21ga with your model.

@RandomItalianGuy
Which partition do you need?

Kipper641, 2021-11-26:
Thanks for the reply, however I have failed to find my backup :)

                      Keith

roleoroleo, 2021-11-26:
> Thanks for the reply, however I have failed to find my backup :)
> 
> ```
>                   Keith
> ```

What's your model?

Kipper641, 2021-11-26:
Y21.

Kipper641, 2021-11-26:
Yi home 1080p :)

roleoroleo, 2021-11-26:
I will send you a file for your y21ga.

roleoroleo, 2021-11-26:
Here it is:
[backup_y21ga.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7610487/backup_y21ga.gz)

roleoroleo, 2021-11-26:
Committed:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/d36146b42689f1f5226da85e7ea989b0c3e8baa5
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

1-Al, 2021-12-01:
@roleoroleo I wrote a Windows app to generate recovery files if you want to try Alpha version. No compiling or installing, just copy to and run from backup folder containing mtd bin files. Kernel on it's own doesn't work, idk what name to use in the header but the 'one' should cover it. If it isn't of interest please delete this message and attachment, perhaps the discussion I started too on sensor resolution. I'm surprised there's no interest in that but the lack of response does answer a question for me so not totally useless but maybe taking up unnecessary space now :)

[mkimg.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7630376/mkimg.zip)

uklatt, 2021-12-01:
Hello roleoroleo,

I bricked my cam a long time ago when I tried to undo the hack. 8-)
I tried to solder a serial and could see boot messages but was not able to use a shell...

Now I found this new unbrick solution and tried it with my mtdblock4.bin backup. - It doesn't work.

Is it possible to flash all mtd partitios with this method? (I have a backup of all partitions)

Uwe

roleoroleo, 2021-12-01:
Yes.
There is a procedure that writes the whole flash but I didn't test it (onebin).
I suggest you to write one partition at a time.
The supported partitions are:

| Filename | Header label |
| uboot_y21ga | xiaoyi-uboot |
| kernel_y21ga | unknown |
| rootfs_y21ga | xiaoyi-rootfs |
| home_y21ga | xiaoyi-home |
| backup_y21ga | xiaoyi-backup |
| env_y21ga | xiaoyi-env |
| one_y21ga.bin | xiaoyi-onebin |

I tested only backup and home.
You need to change this line according to your needs:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/4bbb5b485ae93f09a9d61de8c8854d92e192ae53/unbrick/build.sh#L30

As explained in a previous post, u-boot saves on a variable the hash of the file when writes it.
It you try to write it again, u-boot ignores the file.
If you have a serial connection, the log is clear.

uklatt, 2021-12-01:
I have changed the last bytes of mtdblock1..4 and created the corresponding xxx_y21ga files.
At boot the yellow LED blinks 3 times (only at the first boot with the created files).
But the camera still does not work.

Do you have a whole flash image to try?
Is there any solution to catch the logs/errors without serial connection (serial connection will never work again because of damaged board).

Uwe

roleoroleo, 2021-12-01:
> Do you have a whole flash image to try?

No, and I don't know how to build it.
I didn't try it yet. Maybe @1-Al ?

> Is there any solution to catch the logs/errors without serial connection (serial connection will never work again because of damaged board).

AFAIK no.

1-Al, 2021-12-02:
The onebin isn't the whole image but for allwinner-v2 the first 6 partitions. For instance y21ga v9.0.19.02_202007141512 it's u-boot, kernel(boot), rootfs, home, backup and env. It excludes the mfg and conf blocks. For allwinner-v1 mkimg produces a onebin image of 4 partitions. For instance y25ga v8.3.0.0A_201912271707 it's kernel(boot), rootfs, home and env. I don't have an Allwinner v1 so don't know if it would work. Running the allwinner v2 onebin rcovery appears to work but I haven't really changed enough to be thorough and mkimg could do with some further checks on the first mtd partition such as SPL checksum, optee checksum, u-boot checksum, flattened device tree and there's even a UEFI MBR and partition table embeded at the end, all in this first mtd partition called uboot.

Secondly I have no documentation for this, mkimg makes assumptions and could easily break if the Company makes changes.

uklatt, 2021-12-02:
I tried images created with mkimg.exe but my camera seems to be dead...

Uwe

1-Al, 2021-12-02:
> I bricked my cam a long time ago when I tried to undo the hack. 8-) I tried to solder a serial and could see boot messages but was not able to use a shell...

> I tried images created with mkimg.exe but my camera seems to be dead...

1. Once recovery images are flashed the data crc of them is stored and they are not flashed again unless that number is different or non-existent.
2. If you have shorted your debug port then the serial data will likely be looped back and result in a 'any key press' being detected and halting the boot. Do you have any pics of the damage? Perhaps it can be cleaned up or the serial port redirected as is done on the h60ga? I don't have a y25ga camera.

I wouldn't recommend using onebin or uboot recovery except in rare circumstances as flashing uboot may break it and it has to already be working to run this recovery method. Also the env partition isn't a straight recovery but only recovers existing settings and leaves the others intact. For instance if the recovery has loglevel=8 and recovery has loglevel=4 then that will become loglevel=8 while if env has backup_ver=12345678 and the recovery doesn't have an entry for that then it will remain as backup_ver=12345678. Well, that's how it works for my Yi-Iot camera.

roleoroleo, 2021-12-03:
Thanks for your explanation.

roleoroleo, 2021-12-07:
> @roleoroleo I wrote a Windows app to generate recovery files if you want to try Alpha version. No compiling or installing, just copy to and run from backup folder containing mtd bin files. Kernel on [...]

Tested only with mtd4.
It works properly.

1-Al, 2021-12-09:
Sorry for the late reply. 

Couple of things, just because I can put some code together doesn't make me a programmer by a long shot, there's still some unresolved points such as extra data integrity checks and the kernel boot partition name/type. 

I could give you my source code. It's in assembler x86 fasm style and then you would be free to use and make changes as you see fit, maybe even a C/C++ version?

roleoroleo, 2021-12-10:
> I could give you my source code. It's in assembler x86 fasm style and then you would be free to use and make changes as you see fit, maybe even a C/C++ version?

Assembler?!?
It's not for me, thanks.
:) :) :)

1-Al, 2021-12-11:
Okay, the offer stands incase you change you mind later. There's even a Linux asm version :grinning:

swiatus, 2022-01-18:
> Yes. There is a procedure that writes the whole flash but I didn't test it (onebin). I suggest you to write one partition at a time. The supported partitions are:
>
> Filename	Header label
> uboot_y21ga [...]

Hello @roleoroleo 

Iam a beginner for this all hack things etc, but i have a little different problem.
I have Yi Homce Camera 3 (y25ga v8.3.0.0A_201912271707) with YI firmware and some days camera just stop working.
I can not do hard reset, yellow light blinking and lens adaptations voice coming.
I bought the same camera to make a successfully backup.  (for the while blue light blinking).

Could you please explain more step about new header? Or maybe do you have original firmware to flash it?

roleoroleo, 2022-01-19:
y25ga is Allwinner, not Allwinner-v2.
Check this wiki:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam

HolgerOnGithub, 2022-03-22:
Hallo ich habe die IFUSY mit YA212ga mein Backup zuvor war jedoch schon die Version 11 hat jemand ein Backup mit der Version 9 damit ich auch die BEwegungserkennung wieder in Funktion bekomme?

CodeKlopfer, 2022-03-31:
@HolgerOnGithub, there seems to be one here: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388#issuecomment-1062645815. However, I have not tried it myself.

HolgerOnGithub, 2022-03-31:
> @HolgerOnGithub, there seems to be one here: [#388 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388#issuecomment-1062645815). However, I have not tried it myself.

Thanks, a not wor QFUSY Backup. i have a IFUSY Cam

meluvalli, 2022-08-09:
I think I'm lost.  How does sticking a file named "backup_y21ga" on the SD card make it restore?  I tried this, and it did nothing.

roleoroleo, 2022-08-17:
The bootloader checks if a file with this name exists.
It reads the file and overwrites the whole partition.
Then name must be PARTITION_MODEL where partition is rootfs, home, backup, etc... and MODEL is y21ga, y29ga, etc...

lohacker84, 2022-12-13:
hi good morning I would need you to send me the firmware 9.0.19.02_202007141512 as since I updated to the firmware New y21ga_9.0.19.12_202102241808 everything is blurry at night.  I hope you can help me because this really does not look good.

lohacker84, 2022-12-13:
please help me because i don't know where to find it to be able to go back.

roleoroleo, 2022-12-13:
Here it is:
[backup.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10218851/backup.tar.gz)

lohacker84, 2022-12-13:
really thanks for the help.  I've never done the hack.  is it ok to use this one you sent me?

roleoroleo, 2022-12-13:
If your cam is a y21ga, it's ok.

lohacker84, 2022-12-13:
sorry if I ask for more information but how should I proceed, just unzip into the sd?

roleoroleo, 2022-12-13:
Read the wiki.

lohacker84, 2022-12-13:
do i have to use the unbrick procedure?

lohacker84, 2022-12-13:
I don't have linux to be able to send the command. if I install the yi-hack-Allwinner-v2 hack do I find the downgrade from firmware 9.0.19.12_202102241808 to 9.0.19.02_202007141512 ?

roleoroleo, 2022-12-13:
To install a different fw version (I mean yi original fw) you need to follow the unbrick procedure using the file I posted above.
This operation is not available as a web feature in the hack.
So you need a linux machine (a virtual machine is enough).

The procedure is detailed at this page: https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick
and you don't need to send any command to the cam, only prepare a special file and copy it to the sd card.

lohacker84, 2022-12-13:
I honestly don't think I'm capable of it.  i have never used linux and repo.  could you create me the file ready to be inserted in the sd card?  I am willing to pay you the trouble and time.  I don't want to have to throw it away.  let me know and sorry for the mountain of questions.

roleoroleo, 2022-12-14:
Here it is:
[backup_y21ga.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10225303/backup_y21ga.gz)
[home_y21ga.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10225304/home_y21ga.gz)

Decompress them in your sd card.

lohacker84, 2022-12-14:
Hi good morning!!  first of all thank you for creating the files for me to put on the camera.  which will really help me.  but i would like to share a big surprise this morning, as soon as i opened the app a notification warned me that there is a new firmware update for y21ga.  they brought the firmware from 9.0.19.12_202102241808 to 12.1.19.0_202212141335 amazing!  but I imagine that with the latter you can no longer install the hack. Thanks again!
[image]

roleoroleo, 2022-12-14:
I don't know if the hack works with version 12.
I need to investigate.

keecy, 2023-01-03:
Hi, 
I have followed the unbrick procedure, I have created the 3 files based on my model.
`─$ sudo ./build.sh hacked y211ga
Image Name:   xiaoyi-rootfs
Created:      Tue Jan  3 09:46:29 2023
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1179648 Bytes = 1152.00 KiB = 1.12 MiB
Load Address: 00000000
Entry Point:  00000000
Image Name:   xiaoyi-home
Created:      Tue Jan  3 09:46:29 2023
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    3407872 Bytes = 3328.00 KiB = 3.25 MiB
Load Address: 00000000
Entry Point:  00000000
2432+0 records in
2432+0 records out
1245184 bytes (1.2 MB, 1.2 MiB) copied, 0.00904506 s, 138 MB/s
Successfully mounted y211ga/mtdblock4.bin on y211ga/mnt
### Updating y211ga/backup/init.sh
Image Name:   xiaoyi-backup
Created:      Tue Jan  3 09:46:29 2023
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1245184 Bytes = 1216.00 KiB = 1.19 MiB
Load Address: 00000000
Entry Point:  00000000
`
rootfs_y211ga.gz, home_y211ga, and backup_y211ga.

But now i should unzip all 3 files and transfer into a formatted sd card or a sd with the yihack files ?

Br,

roleoroleo, 2023-01-04:
It does not matter.
The files are processed before the cam starts.
But do one file at a time.
Start with backup and check if the cam works, then home and finally rootfs.

keecy, 2023-01-04:
Thanks roleoroleo, 
But it seems there is no reaction from the camera, the yellow led keeps on.
Shall i try something more before to try the hardway ?

the uart config would be like that?
the speed - 9600 or 115200 ?

[image]

Br,

roleoroleo, 2023-01-04:
> But it seems there is no reaction from the camera, the yellow led keeps on.

Strange, are you sure your cam is a y211ga?

> the speed - 9600 or 115200 ?

115200

keecy, 2023-01-04:
[image]
I don't say yes, because i bought a pack of 2. The other  is working without an issue.
Do you have a file y211ga, so I can try?

Br,

roleoroleo, 2023-01-06:
Here it is, version 9.
[rootfs_y211ga.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10360061/rootfs_y211ga.gz)
[home_y211ga.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10360062/home_y211ga.gz)
[backup_y211ga.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/10360063/backup_y211ga.gz)

keecy, 2023-01-06:
Thanks, already made it work.
the camera is a y291ga. :)

twillist, 2023-02-12:
So the problem I'm having is that if I have the wires soldered for a serial connection I can't get any output to putty (but that's probably my inexperience with serial connections) but I can get the camera to boot when applying the unbrick fixed backup files and can even get the hack to boot and I can access the webpage. But a soon as I take the wires off it won't boot. Any ideas?

vla268, 2023-03-03:
Hello.
I would like to recover my old camera Yoosee Techage Novatek/Grain Media GM8135S 1080p, 8135s_8M_firmware 21.00.01.05
Resetting doesn't work. Camera starts but can't be flashed with IPCTest tool. Everytime I try camera flashes and goes offline, no lan access for about a minute, then comes back online, but it doesn't let you do everything with it.
I have been reading your method using a file in the sdcard. Could I get a file for my camera? 
I contacted manufacturer and said camera is broken with no solution, but I don't think so.
The problem with the camera can be fixed with your method, I just need the right file.
Thanks for any assistance.
Regards.

closed 2023-07-09 by stale[bot]

luckyfreddy, 2025-02-10:
Thanks a lot. My bricked cam was unbricked. With serial-communication I was not be able to do that; no inputs was accepted. With the method of this thread it worked.