roleoroleo/yi-hack-Allwinner-v2 #308: Yi Dome - grey screen

michalbogacz, 2021-11-26:
Firmware Version | 0.1.9
Base Version | 9.0.05.33_202009251856
Model Suffix | r30gb

I have a problem with rtsp. Often screen turns grey (at least once every 30s). I don't see any artifacts, it's just grey. 

It doesn't seem to be problem with software I have to view rtsp. I have second (older) camera with 
yi-hack-Allwinner and there is no such issue. 

Can I somehow help with fixing this?

roleoroleo, 2021-11-27:
Try with another client and check if you have the same problem.
And check if the wifi signal is poor.

michalbogacz, 2021-11-27:
I moved cam and connected with VLC but that didn't helped. WIFI signal strength is 88%.

roleoroleo, 2021-11-27:
r30gb is the only model that sends the stream coded in h265.
Probably it's a performance issue but I can't test it because I don't own that model.

michalbogacz, 2021-11-28:
How it works? Your code is converting h265 to h264? 

Is there anything I can do to help you with that cam?

roleoroleo, 2021-11-28:
No, my code doesn't decoder or encode the stream.
Simply it extracts the frames from the framebuffer and sends them to the rtsp server.

michalbogacz, 2021-11-28:
From performance perspective it doesn't seem that there is high load on CPU
```
Mem: 59864K used, 1044K free, 1804K shrd, 1920K buff, 11376K cached
CPU: 40.0% usr  9.9% sys  0.0% nic 49.2% idle  0.0% io  0.0% irq  0.7% sirq
Load average: 1.54 1.42 1.24 1/115 16185
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1118     1 root     S    42180 69.0   0 42.5 ./rmm
 1388     1 root     S<    5484  8.9   0  2.1 rRTSPServer
 1211     1 root     S     3476  5.6   0  0.5 ./p2p_tnp
 1209     1 root     S     2724  4.4   0  0.2 ./mp4record
  598     1 root     S     1084  1.7   0  0.2 ./dispatch
```

roleoroleo, 2021-11-28:
Does VLC show any error?
[image]

michalbogacz, 2021-11-29:
I don't see any errors in VLC. Here is one of moments when screen turned grey: 

[image]

michalbogacz, 2021-12-04:
I tried to debug rRTSPServer with
`RRTSP_AUDIO=no RRTSP_MODEL=r30gb RRTSP_RES=high RRTSP_DEBUG=3 rRTSPServer 2> log.txt` 

I was also watching stream. It was working for a few seconds, then broke and after next few seconds I stopped rRTSPServer (that's why log is not long).

Here is [log.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7654363/log.txt)

In log I found:
```
1638640094487: warning - unexpected NALU header
1638640094487: warning - incorrect frame counter - frame_counter: 6552 - frame_counter_last_valid: 6946
``` 
Is it possible that this is connected to issue?

roleoroleo, 2021-12-05:
This warning is not an error. Let's analyze the log:
```
1638640094487: frame detected - frame_len: 28 - frame_counter: 7529 - frame_counter_last_valid: 7529 - resolution: 1080
1638640094487: frame_len: 28 - cb_current->size: 262144
1638640094487: frame_len: 37 - frame_counter: 7529 - resolution: 1080
1638640094487: frame_write_index: 22/42
```
Everything OK above.
Below the grabber looses the sync. It waits for the frame number 6947 but detects frame number 6552.
Same problem for both: low res and high res.
```
1638640094487: warning - unexpected NALU header
1638640094487: warning - incorrect frame counter - frame_counter: 6552 - frame_counter_last_valid: 6946
1638640094487: frame detected - frame_len: 834 - frame_counter: 6552 - frame_counter_last_valid: 6946 - resolution: 360
1638640094487: warning - incorrect frame counter - frame_counter: 7132 - frame_counter_last_valid: 7529
1638640094487: frame detected - frame_len: 3496 - frame_counter: 7132 - frame_counter_last_valid: 7529 - resolution: 1080
1638640094487: warning - incorrect frame counter - frame_counter: 6553 - frame_counter_last_valid: 6946
1638640094488: frame detected - frame_len: 860 - frame_counter: 6553 - frame_counter_last_valid: 6946 - resolution: 360
1638640094488: warning - incorrect frame counter - frame_counter: 7133 - frame_counter_last_valid: 7529
1638640094488: frame detected - frame_len: 2925 - frame_counter: 7133 - frame_counter_last_valid: 7529 - resolution: 1080
1638640094488: warning - incorrect frame counter - frame_counter: 6554 - frame_counter_last_valid: 6946
1638640094488: frame detected - frame_len: 841 - frame_counter: 6554 - frame_counter_last_valid: 6946 - resolution: 360
1638640094488: warning - incorrect frame counter - frame_counter: 7134 - frame_counter_last_valid: 7529
1638640094488: frame detected - frame_len: 3190 - frame_counter: 7134 - frame_counter_last_valid: 7529 - resolution: 1080
1638640094488: frame detected - frame_len: 827 - frame_counter: 19090 - frame_counter_last_valid: 7529 - resolution: 0
1638640094488: warning - incorrect frame counter - frame_counter: 6555 - frame_counter_last_valid: 6946
1638640094488: frame detected - frame_len: 805 - frame_counter: 6555 - frame_counter_last_valid: 6946 - resolution: 360
1638640094488: warning - incorrect frame counter - frame_counter: 7135 - frame_counter_last_valid: 7529
1638640094488: frame detected - frame_len: 2806 - frame_counter: 7135 - frame_counter_last_valid: 7529 - resolution: 1080
1638640094488: warning - incorrect frame counter - frame_counter: 6556 - frame_counter_last_valid: 6946
1638640094488: frame detected - frame_len: 843 - frame_counter: 6556 - frame_counter_last_valid: 6946 - resolution: 360
1638640094488: warning - incorrect frame counter - frame_counter: 7136 - frame_counter_last_valid: 7529
1638640094488: frame detected - frame_len: 2728 - frame_counter: 7136 - frame_counter_last_valid: 7529 - resolution: 1080
1638640094488: warning - incorrect frame counter - frame_counter: 6557 - frame_counter_last_valid: 6946
1638640094488: frame detected - frame_len: 806 - frame_counter: 6557 - frame_counter_last_valid: 6946 - resolution: 360
```
Now the grabber finds frame number 6947 and ths restore the sync.
Same thing for high res. It waits for frame 7530, it detects frame 7531 but it's normal: not all frames in the framebuffer are h264 frames.
There can be a jump of a number in the counter.
```
1638640094513: frame detected - frame_len: 791 - frame_counter: 6947 - frame_counter_last_valid: 6947 - resolution: 360
1638640094514: frame_len: 791 - cb_current->size: 49152
1638640094514: frame_len: 791 - frame_counter: 6947 - resolution: 360
1638640094514: frame_write_index: 19/42
1638640094514: frame detected - frame_len: 814 - frame_counter: 19196 - frame_counter_last_valid: 7529 - resolution: 0
1638640094514: frame detected - frame_len: 3655 - frame_counter: 7531 - frame_counter_last_valid: 7531 - resolution: 1080
```

So, this behavior is not a problem: if the grabber is able to restore the sync. no frames are lost.

closed 2023-01-08 by stale[bot]