roleoroleo/yi-hack-Allwinner-v2 #314: Yi 1080p Home RFUS

fernandolcardoso, 2021-12-01:
Hello everyone,

I recently bought some cameras from Yi and am looking for an alternative firmware for them.

The problem is I'm not finding a compatible version or I don't know which version to use.

- Camera: Yi 1080p Home RFUS
- Firmware: 9.0.36.02
- Model: YYS.2016

I appreciate if anyone can tell me the correct version.

roleoroleo, 2021-12-01:
Probably it's a y211ga.
To be sure I need a dump of the flash.

bigeazy000, 2021-12-04:
HeyFernando, 

I just got the same cameras today, and was able to load y211ga firmware without issue, its working fine.

AndriyLviv1, 2021-12-05:
> Probably it's a y211ga. To be sure I need a dump of the flash.
Hi
here is backup of camera 
RFUS
FW 9.0.36.02_202105081647
Y211GE
Can i load y211ga firmware?
[backup_RFUS_9.0.36.02_Y211GE.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7655625/backup_RFUS_9.0.36.02_Y211GE.zip)

roleoroleo, 2021-12-05:
Yes, it's a y211ga.

fernandolcardoso, 2021-12-06:
Confirmed, the version I questioned is compatible with the y211ga.

closed 2021-12-06 by fernandolcardoso

Al3eRtO, 2022-08-29:
Hi, i just got the YI RFUSY625HE, is it possible to install the custom firmware?
Thanks!

Marcin685, 2022-12-04:
I upload y211ga firmware version to y211ge camera and all works OK!