roleoroleo/yi-hack-Allwinner-v2 #315: y28 bootloop

beamwerks, 2021-12-01:
hooray for this project.  just tried on my kami home with y28ga_0.1.9 and having the same boot loop as described in this issue

https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/5

roleoroleo, 2021-12-03:
Please try the unbrick procedure described here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick
or here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam

beamwerks, 2021-12-03:
should have been clearer.  camera boots fine when i remove the sd card and works as normal pre hack.

it's just when i try the y28 software, it goes into the yellow light / short blue light and briefly working / then yellow light loop

roleoroleo, 2021-12-03:
Try to enable swap space.

beamwerks, 2021-12-11:
boom.  just had to catch it that setting in while the url was briefly working.    works great now.  donation sent.  cheers!

1-Al, 2021-12-11:
If the camera is too unstable for applying settings AFAIK you could manually change it by editing the setting in the 'yi-hack/etc/system.conf' SDCard file on a different device. Not much help to you now but might help someone else later.

roleoroleo, 2021-12-11:
@beamwerks 
Thank you for your donation!

closed 2023-01-08 by stale[bot]