roleoroleo/yi-hack-Allwinner-v2 #316: Imposible Hack Yi 1080p Home IFUS 9.0.36*	y211ga

julianfj, 2021-12-02:
I have a newly purchased camera and I am not able to do the hack. when I put the card it ends up turning the blue light but the web does not respond
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7644739/backup.zip)
.

roleoroleo, 2021-12-03:
The hack you are using is ok.
Try again witha a different sd card.

julianfj, 2021-12-03:
I have tried many cards and none of them work
After insert the sdcard Screenshot_2
[image]
.
Before 
[image]

When start the web is not avalible but the camera is connected to network

[image]

Could you help me?

roleoroleo, 2021-12-03:
Factory.done means that the hack was performed.
Does the cam boot correctly?
Is it in boot loop?
What's the sequence of the led?

julianfj, 2021-12-03:
orange led --> blue led blinked -> blue led
[image]

roleoroleo, 2021-12-03:
The led is ok, the cam isn't looping.
Did you remove the sd card after the hack?

Svartskog, 2021-12-03:
192.168.1.217**:8080** will solve your problem

1-Al, 2021-12-04:
Please check 192.168.1.217 is the camera (camera MAC address) and not another device. What do you get for windows command ftp 192.168.1.21? If it connects try running an SSH and ps command, can use putty https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html

julianfj, 2021-12-04:
without the SD:
[image]
[image]

with SD
[image]
[image]
[image]
with SD the app dont work but the web no load

1-Al, 2021-12-05:
> with SD the app dont work but the web no load

The app should also work by default with the SDCard and hack unless the cloud app is disabled. 

Your backup shows init.sh already changed so chances are there's something going bad after that. You could try checking the "log" folder on the SDCard for clues, you can rename it to say "log1" and see if you get a new "log" folder with the hack. Also try "telnet 192.168.1.217"

What size and format of SDCard are you using?

julianfj, 2021-12-05:
hi, news, telnet works.
in sd log folder no whow any relevant...

when open telnet show this error

(none) login: root

BusyBox v1.27.2 () built-in shell (ash)

**mount: mounting none on /sys/kernel/debug failed: Resource busy**

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

a ps
root@(none):~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  210 root         0 SW   [oom_reaper]
  211 root         0 SW<  [writeback]
  213 root         0 SW   [kcompactd0]
  214 root         0 SW<  [crypto]
  215 root         0 SW<  [bioset]
  217 root         0 SW<  [kblockd]
  258 root         0 SW   [sys_user]
  265 root         0 SW   [kworker/0:1]
  266 root         0 SW<  [cfg80211]
  272 root         0 SW<  [watchdogd]
  287 root         0 SW<  [spi0]
  297 root         0 SW   [kswapd0]
  380 root         0 SW<  [bioset]
  385 root         0 SW<  [bioset]
  390 root         0 SW<  [bioset]
  395 root         0 SW<  [bioset]
  400 root         0 SW<  [bioset]
  405 root         0 SW<  [bioset]
  410 root         0 SW<  [bioset]
  415 root         0 SW<  [bioset]
  420 root         0 SW<  [bioset]
  448 root         0 SW   [kworker/0:2]
  449 root         0 SW   [irq/302-sunxi-m]
  451 root         0 SW   [irq/166-sdc0 cd]
  452 root         0 SW   [irq/303-sunxi-m]
  495 root         0 SW   [kworker/0:3]
  518 root         0 SW   [kworker/0:4]
  520 root         0 SW<  [bioset]
  521 root         0 SW   [mmcqd/0]
  533 root         0 SW<  [kworker/0:1H]
  546 root         0 SWN  [jffs2_gcd_mtd4]
  553 root       980 S    -/bin/sh
  574 root         0 SW   [kworker/u2:2]
  585 root       996 S    /usr/sbin/telnetd
  597 root         0 SW<  [phy0-atbm_wq]
  598 root         0 SW   [phy0-usb_atbm_b]
  641 root      1172 S    ./dispatch
  691 root      1424 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211
  727 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
  778 root       984 S    -ash
 1888 root       980 R    ps

root@(none):~# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1280      1280         0 100% /
devtmpfs                 30400         0     30400   0% /dev
/dev/mtdblock3            2816      2816         0 100% /home
/dev/mtdblock4            1216       960       256  79% /backup
tmpfs                    32768        28     32740   0% /tmp
/dev/mmcblk0p1          253952        35    253917   0% /tmp/sd

:(

1-Al, 2021-12-05:
Good.

> when open telnet show this error
That's okay, nothing to worry about

What you should worry about is your SDCard

```
Filesystem 		1024-blocks 	Used 	Available 	Use% 	Mounted on
/dev/mmcblk0p1 		253952 		35 	253917 		0% 	/tmp/sd
```

According to this the first partition is about 260MB and only about 35KB used. Y-Hack on my card uses around 18MB. Please check the details of your SDCard.

julianfj, 2021-12-05:
by sd card only have the hack, and this is only some kbs...

1-Al, 2021-12-06:
> by sd card only have the hack, and this is only some kbs...

https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases

y211ga_0.1.9.tgz
-----------------
Packed size
8,802133 Bytes = 8.39MiB
[image]

Unpacked size
19568640 Bytes = 18.66MiB
[image]

julianfj, 2021-12-06:
oh!! this is the problem!!
i downloaded the files of the main page
[image]
this repo no hace these files!!!

thanks for the support!

1-Al, 2021-12-07:
Cool, good to hear you got it to work. :+1:

xshaomin, 2022-05-21:
Hi, I have the same issue. Could you please kindly let me know how you resolve this? 
The fw_upgrade_in_progress is an empty file.

Thank you very much!

> oh!! this is the problem!! i downloaded the files of the main page
> 
> [image]
> 
> this repo no have these files!!!
> 
> thanks for the support!

roleoroleo, 2022-05-21:
Download the archive from the release section:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases

xshaomin, 2022-05-21:
Thank you very much! I downloaded the y211ga_0.2.1.tgz from the release. I extracted the contents of the archive to the root of the SD card.  There is an empty file named  fw_upgrade_in_progress, size is 0.

Basically, I have the same issue "julianfi" had. I could not access the camera's web configuration page by using cam-ip:8080 (I replaced the cam-ip by the ip address of the camera)
Later, I checked the sd card and the Factory.done folder was there.

> Download the archive from the release section: https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases

roleoroleo, 2022-05-21:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/316#issuecomment-986369022

xshaomin, 2022-05-21:
> [#316 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/316#issuecomment-986369022)

Thank you. I have a 32 GB sd card, reformat by the camera and removed all the non-hidden files.
Then I copy the files and folders to the sd card using a win 10 computer
|-- Factory/
|-- yi-hack/
|-- lower_half_init.sh
 
Anything is wrong here?

roleoroleo, 2022-05-21:
It's ok.
If it doesn't work, use another sd card.

xshaomin, 2022-05-21:
> It's ok. If it doesn't work, use another sd card.

thanks. Have tried a few sd cards from different brand, but no luck.

i checked again, the camera is Yi 1080p Home RFUS, the firmware is 9.0.35, instead of 9.0.36, does this matter?

Thank you very much!

roleoroleo, 2022-05-21:
If the firmware version is different, the hack is not for your cam.
Please make a backup of the flash and share with me:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)

roleoroleo, 2022-07-16:
I will check it asap.

roleoroleo, 2022-07-16:
Please try this beta:
[y291ga_0.2.2.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/9126220/y291ga_0.2.2.tar.gz)

Naffto, 2022-07-17:
wow, thank you. this was really fast.
I've tested it on 2 cams with RFUS and firmware 9.0.35 and it worked on both.

RuiDoGolf, 2022-07-24:
Thanks, I have successfully hack two RFUS 9.0.35.00.
do some one know how to disable time stamp?

roleoroleo, 2022-07-25:
> Thanks, I have successfully hack two RFUS 9.0.35.00. do some one know how to disable time stamp?

If you mean the timestamp in the upper-left corner I think it's not possible.
It's managed by yi processes.

RuiDoGolf, 2022-07-25:
> > Thanks, I have successfully hack two RFUS 9.0.35.00. do some one know how to disable time stamp?
> 
> If you mean the timestamp in the upper-left corner I think it's not possible. It's managed by yi processes.

ok, thanks for the information. keep the good work :)

ethempekin, 2022-08-27:
Hi,

ref: y291ga. I got the hacks for new set of cameras and added them to the homekit. While I can access them on Yi Home, homekit fails to respond. On my homebridge Log I see error code #1.

Any suggestions? Thanks again for this amazing work.

fernandolcardoso, 2022-09-01:
I have a curious situation, I was using the y211ga version normally until yesterday the blue LED started blinking non-stop and I couldn't access the yi-hack services anymore.

After several failed attempts, changing card, trying to restore. I decided to open it and I noticed that I don't have the y211ga, but a y291ga with version 9.0.36*

[image]

[image]

[image]

[image]

[image]

I tested the beta version here, the LED stopped blinking, however, I can't open the web interface.

fernandolcardoso, 2022-09-01:
Sorry, I changed the card, tested it again and now it works again!

[image]

ethempekin, 2022-09-02:
Can you share the fm link here? 

> Sorry, I changed the card, tested it again and now it works again!
> 
> [image]

roleoroleo, 2022-09-02:
y211ga and y291ga are very similar.

ethempekin, 2022-09-02:
I have no difficulties in setting up camera with y211ga or y291. 
I am getting for this error:
`FFmpeg exited with code: 1 and signal: null (Error)`

Probably related to something else then?

roleoroleo, 2022-09-03:
Where do you run ffmpeg?
Which parameters?
Please, explain me better.

ethempekin, 2022-09-03:
It is on RPi.

I used a similar approach to my older cameras when copying over the IP address.

` {
                    "name": "Baby Camera",
                    "unbridge": true,
                    "videoConfig": {
                        "source": "-i rtsp://192.168.1.43/ch0_0.h264",
                        "stillImageSource": "http://192.168.1.43:8080/cgi-bin/snapshot.sh?res=low&watermark=yes",
                        "maxStreams": 2,
                        "maxWidth": 800,
                        "maxHeight": 600,
                        "forceMax": false,
                        "vcodec": "libx264",
                        "audio": true,
                        "debug": true
                    } `

roleoroleo, 2022-09-06:
Video source is ok.
Try to force the frame rate to 20 fps.

closed 2023-01-08 by stale[bot]

tutsisback, 2024-09-12:
Hello, I have exactly the same camera as @fernandolcardoso. But I did a big mistake. I erased the content of my SD Card and did not do a backup. So now, my cam seems to be bricked as the led light keeps shinning yellow even I try to flash it again with last firmware for y291ga model.
Have you any solution ?
Thanks

fernandolcardoso, 2024-09-12:
Try the y211ga and y291ga variants

tutsisback, 2024-09-13:
If you mean format SD card, and unzip y291ga or y211ga files on Root of sd card and power UP the cam, that's what I've been doing multiple times without results.

fernandolcardoso, 2024-09-13:
@tutsisback Exactly, it's already occurred to me to use different firmware and work. In this case, try to make the recovery. There is documentation on this.

roleoroleo, 2024-09-14:
You need to identify your cam model.
Check the fw version in the app.

tutsisback, 2024-09-20:
@roleoroleo As I said, my cam seems to be bricked, as the yellow light is always on. SD Card is empty and I cannot see the firmware in the app as the cam is not connecting.

fernandolcardoso, 2024-09-20:
So once, this happened to me, what I remember doing was replacing the card. Another note, try using a 32GB card