roleoroleo/yi-hack-Allwinner-v2 #317: How to reset if hw button is broken

Mrivilla, 2021-12-07:
Hi,
I have a new YI Dome U. Hack is working perfectly, but reset button doesn’t work. It seems is not present or broken. I’m not eagle enouth to dismantle the camera.
Is there any possibility to force factory hard reset executing a boot command or though ssh access?
Thanks.
Marcos

roleoroleo, 2021-12-07:
Why do you need to do a reset?
What type of reset do you mean?
Reset of the yi config or hack config?

Mrivilla, 2021-12-07:
I have the camera with the hack and works, but unfortunately I need to enroll again the camera to the mobile application. To do that, it would be necessary press the reset button, but the button is broken or doesn't exit, so I cannot initiate the process to enroll the camera to the mobile app.  Just wondering what are the files that the hard reset to initiate the enrollment process again.
Then after the added to the mobile can add the hack again. The backup that I've taken  just was after the first enrollment before the hack, so I'm still have the same problem.
Maybe a factory backup restoration it would be enough.

roleoroleo, 2021-12-07:
To reconnect the cam to the app, simply switch off "Disable Cloud".

closed 2021-12-07 by Mrivilla

LarryMassey, 2023-01-16:
Just took my camera apart & the reset button was laying inside the unit… I needed to reset because TMobile replaced my modem & 9 cameras later… this one wouldn’t reset pics included… So I Won’t Be Able to Use this on…
[image]
[image]