roleoroleo/yi-hack-Allwinner-v2 #321: Yi Outdoor 1080p IFUS

emadwealthsimple, 2021-12-17:
Hey, there I have been trying to install the hack on the camera for the past 2 days with no luck.

I have tried different sd cards and files and versions. 

I installed it from https://github.com/TheCrypt0/yi-hack-v4/releases 
- yi hack outdoor zip 

But it only has two files the home_h30 file and roots_h30 file, I was looking at other issues and other people were uploading four files two of them being folders.

Can someone please help me out this is my first time trying something like this?

roleoroleo, 2021-12-17:
It depends on the model of your cam.
Check your fw version in the mobile app.

emadwealthsimple, 2021-12-17:
Model is YHS.3020
Version Number is 9.0.26

roleoroleo, 2021-12-17:
See here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2
h30ga

emadwealthsimple, 2021-12-18:
Is it just these two files?

[image]

roleoroleo, 2021-12-18:
No, download the archive from the release page.

emadwealthsimple, 2021-12-21:
Just did that now. Camera had yellow light for around 30 seconds and then blue light flashed then blue light on.

Getting an error when I try to access from IP address and from http://IP-CAM:8080

roleoroleo, 2021-12-24:
Check the content of the sd card.
Is the Factory folder renamed to Factory.done?

closed 2023-01-08 by stale[bot]