roleoroleo/yi-hack-Allwinner-v2 #322: Yi 1080p Home QFUS 9.0.36

iBL1nK, 2021-12-17:
I think this might be a new version. I bought this off Aliexpress 6 December 2021.
Firmware version is 9.0.36.03_202109231441
Serial starts with QFUSY
I've uploaded my backup dump here: https://www.mediafire.com/file/2h1llae2xt3nuzt/backup.rar/file

I have no clue how to proceed, please let me know what to do. Happy to test and be the guinea pig.

iBL1nK, 2021-12-17:
I risked it for the biscuit and can now confirm that y211ga_0.1.9.tgz works for this new model. Might want to update wiki/readme for future reference

Venerdi3, 2022-01-21:
Help me, please ...
I need explanations for dummies ...
My YI 1080p Home Camera AI + also has Firmware version is 9.0.36.03_202109231441 and the Serial starts with QFUSY.
I tried y211ga_0.1.9.tgz, but nothing happens.
I also have some other questions ...
1) In the microSD should I put rootfs_y20 and home_y20 only or should I also put y211ga_0.1.9.tgz?
2) Should I put y211ga_0.1.9.tgz as the only file?
3) Should I put y211ga_0.1.9.tgz, or should I unzip this file?
4) How do I find login and password to create the RTSP string and use the IPCam with software like iSpy?
...thank you

roleoroleo, 2022-01-22:
You should decompress the archive.
But the archive doesn't contain rootfs_* and home_*
Username and password are blank if you don't change them.

Venerdi3, 2022-01-22:
Thank you for your answer.
Unfortunately nothing happens.
I don't even understand if the firmware has been flashed or not.
What can I understand from?
If I check with the YI app on my smartphone I don't see any changes. On the smartphone it works perfectly, but the firmware release I see is always the same.
I tried to do what you told me, and then I used the Contacam software on the pc where I have other IPCam, but all the RTSP protocols don't work. I left login and psw blank.
Do you have any other suggestions for me?

Venerdi3, 2022-01-23:
I have succeeded!
I had formatted my SDcard in FAT32 with Easeus Partition, instead I had to format it with the internal software of the IPCam
Now everything is OK, the RTSP works.
Thanks roleoroleo!

jffz, 2022-01-27:
Can't achieve to make it works with `QFSUY` serial and `9.0.36.03_202109231441` firmware. Tried `y211ga_0.2.0.tgz` and `y211ga_0.1.9.tgz`
Can't even achieve to dump the original firmware, just nothing happen.

Any help would be appreciated.

roleoroleo, 2022-01-27:
Is the cam working with the yi app?
Is it online?
If yes, please try to change the sd card.

jffz, 2022-01-27:
It's all good, just double checked my SD card and it was wrongly formatted using an extended partition.
Thanks :)

closed 2023-01-08 by stale[bot]