roleoroleo/yi-hack-Allwinner-v2 #323: YI Dome Guard QFUS r35gb

Rushmed, 2021-12-18:
Hi,

my new cam YI Dome Guard QFUS r35gb FW: 9.0.46 arrived today.
Tried flashing r30gb and r40ga without success.
Is there any chance to get the cam hacked?

roleoroleo, 2021-12-19:
I need a backup of the flash to check if I can support it.

Rushmed, 2021-12-19:
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7740610/backup.zip)

Prego!

roleoroleo, 2021-12-19:
This file should grant telnet/ssh access.
[r35gb_0.1.9.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7741219/r35gb_0.1.9.tar.gz)

Now I need a copy of the framebuffer file /dev/shm/fshare_frame_buf

Rushmed, 2021-12-19:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7741301/fshare_frame_buf.zip)

Here it is!

roleoroleo, 2021-12-19:
Please, test this archive.
[r35gb_0.1.9.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7741518/r35gb_0.1.9.tar.gz)
It should work.
You could find problem with rtsp, snapshot and/or ptz.

Rushmed, 2021-12-20:
Thanks soo much!

Shoud I give feedback here or should I open issues if I find some?

roleoroleo, 2021-12-20:
Here.
Probably we will need to change a few more bits.

Rushmed, 2021-12-20:
- Serial number is not shown:
[image]

- RTSP is working fine. Tested both streams wit VLC and tinyCam.

- ONVIF PTZ is working for pan and tilt. The zoom is not working. Tested with tinyCam Pro

- Snapshot is working but there is an additional timestamp at the top left corner which does not show the correct time. The cam is not a european model so I needed to choose American server at YiHomeApp to get the cam logged in. Its possibly a DST issue. Timezones are set up correct in the app and in the cams web UI. 
[image]

Thats what I figured out so far. I tested the points you mentioned and set up the cam the way I did with my other hacked Yi cams. Every thing else seem to be fine. Of course this was not an in depth test.

roleoroleo, 2021-12-20:
Zoom? Does the camera have a zoom?

You should set GMT in the app and set the correct time in the configuration web page.

The serial number is not ok. Please read the file /tmp/mmap.info with hexdump and tell me the offset of the serial number "QFUS..."

Rushmed, 2021-12-20:
Zoom is menitoned on the package and [here](https://www.yitechnology.com.sg/products/yi-dome-guard/).
[image]

How to set GMT? This is the App setting:
[image]
and this is the web UI setting:
`CET-1CEST,M3.5.0,M10.5.0/3`
Should be correct for Germany.

I do not know/own hexdump. Tried wit NPP:
[image]
Cursor is after QFUS at position 789.

roleoroleo, 2021-12-20:
> Zoom is menitoned on the package and [here](https://www.yitechnology.com.sg/products/yi-dome-guard/). [image]
>

I think that this is a feature of the app. Not available on the cam.
But I could investigate.

> How to set GMT? This is the App setting: [image] and this is the web UI setting: `CET-1CEST,M3.5.0,M10.5.0/3` Should be correct for Germany.

In the app, set GMT0 universal... (I don't remember exactly the name).
The web ui is correct.

> I do not know/own hexdump. Tried wit NPP: [image] Cursor is after QFUS at position 789.

hexdump is embedded in the hack, you can run it using ssh.

Rushmed, 2021-12-20:
Ok, I dont need the zoom feature. Save your time.

With:
`hexdump -C /tmp/mmap.info`
I found QFUS @ `00000310` so 784

roleoroleo, 2021-12-20:
Ok.
I will publish an official release asap.

Rushmed, 2021-12-20:
Thanks a lot Sir, I am looking forward to test it!

croessi, 2022-01-05:
The r35gb Beta works like a charm on Yi Dome Guard YRS. Super Thanks @roleoroleo! Only the Serial Number is not displayed correctly, as mentioned. 

Unfortunaltely Presets in PTZ are not working - seems like an issue of the camera itself: https://forum.yitechnology.com/t/yi-dome-guard-preset-bookmarks-unavailable/6523/20 Any suggestions?

Rushmed, 2022-01-05:
I can confrirm the presets point too.

rt400, 2022-01-07:
i have the same camera, after hack using r35gb the camera works fine until that i start to config the camera then i got stuck and the move up and stay there.

croessi, 2022-01-07:
@rt400 maybe you have to enable swap, ths helped with stability issues on my side.

roleoroleo, 2022-01-07:
@croessi 
@Rushmed 
Is the PTZ working with the original app?

croessi, 2022-01-08:
@roleoroleo The PTZ is working fine in the App and in the Hack. But Presets/Bookmarks are missing in the App (as described here: https://forum.yitechnology.com/t/yi-dome-guard-preset-bookmarks-unavailable/6523/20). I tried other ONVIF Apps with the same result. I have a Dome U as well, where Presets are working fine. Strangely I do not understand why it is disabled in the Guard, since it is abel to move back in the exact same position after reboot.

roleoroleo, 2022-01-08:
Probably it's a bug.
But I'm not able to enable presets if they don't work in the app.

Rushmed, 2022-01-08:
> Is the PTZ working with the original app?

Nope

rt400, 2022-01-09:
does r35gb still in beta ?

roleoroleo, 2022-01-09:
> does r35gb still in beta ?

Yes.

@Rushmed 
I need more details about the differences between two mmap.info
Could you please send me a copy of both versions?
If you don't want to publish them here please send me a mail to the address of my paypal.

Rushmed, 2022-01-09:
@roleoroleo Where to get these files?

rt400, 2022-01-09:
i want to help to... what do you need ?

roleoroleo, 2022-01-09:
> 
> @Rushmed I need more details about the differences between two mmap.info Could you please send me a copy of both versions? If you don't want to publish them here please send me a mail to the address of my paypal.

Sorry, forget this post.

roleoroleo, 2022-01-09:
Serial is already fixed, for ptz I think we need to wait for an upgrade from yi.

rt400, 2022-01-09:
i still get "000000" on serial

roleoroleo, 2022-01-09:
Yes, fixed but not pushed.

rt400, 2022-01-11:
@rt400 rtsp low quality is missing...

roleoroleo, 2022-01-11:
Does low res stream work?
If I remember correctly @Rushmed tested both streams.

roleoroleo, 2022-01-11:
Is it enabled in your config?

rt400, 2022-01-11:
> Is it enabled in your config?

Sorry i see it now that is only on high, change it to both and now i have low also .

rt400, 2022-01-11:
@roleoroleo How can I get package with the your fixes ?

roleoroleo, 2022-01-11:
At the moment I didn't build a package.
If you want, I could send you a beta.

Try this rtsp server, audio should work, but no auth is supported:
[rtsp_server.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7846459/rtsp_server.gz)
Remember to kill wd_rtsp.sh and rRTSPServer before to start it.

rt400, 2022-01-11:
thanks, i make a bulid witk your wiki. there is a way to upload it to the camera? or i have to put it on SDCARD ?

roleoroleo, 2022-01-11:
You have to update the sd card.

pannal, 2022-02-04:
> Hi, bought it on [Ali](https://de.aliexpress.com/item/1005001320188881.html?gatewayAdapt=glo2deu&spm=a2g0s.12269583.0.0.7ab524f8O8IZxi) but the PTZ profiles are also not working. Must be a bug on YIs side cause even with unhacked cam they arent working.

Thank you! Do you have an r30gb to compare against? As far as I understand, the r35gb is nearly two years newer - does that show?

Rushmed, 2022-02-04:
No I dont have a r30gb. I have a h201c which has working PTZ profiles with another @roleoroleo  mod.

closed 2023-01-07 by stale[bot]