roleoroleo/yi-hack-Allwinner-v2 #328: no mqtt motion messages? solved ! no app working after hack? solved!

Maxym77, 2022-01-02:
Hi, after some hours of test i have found this two solutions for this problems:

1) only online-offline messages in mqtt (no motions messages)
2) orignal app on phone won't work

In past, i have created the "Factory" directory (in /tmp/sd) for my personalizated wpa_supplicant.conf, and the problems had started, so just removing the directory and work perfectly.

roleoroleo, 2022-01-06:
> There is an solution for put forced the “mac” SSID of my repeater without Factory directory?

Could you explain me better?

Maxym77, 2022-01-06:
hi roleoroleo!
if you have a network with several repeaters, it may happen that the cam connects to the wrong repeater (for example the farthest one). I found a solution to force the correct repeater:

i have created the file /tmp/sd/yi-hack/startup.sh 
with this simple script:
#!/bin/sh
cp /tmp/sd/yi-hack/wpa_supplicant.conf /tmp/wpa_supplicant.conf

the /tmp/sd/yi-hack/wpa_supplicant.conf is:

ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
bssid=XX:XX:XX:XX:XX:XX:XX <-----  (the real mac of repeater i need to connect)
ssid="MAX"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="Massimo200177!"
}

at each reboot the cam will connect to the favorite repeater ... maybe at some point it switches to the other but 95% of the time it stays connected to the preferred one.

For the rest the cam is perfect with your hack, the only problems I encounter are:
- too many unnecessary motion (human detection does not work).. too many motion start / end messages arrived in mqtt even though nothing is moving in the room
- RTSP signal in tcp is fine but ffmpeg freeze after few seconds, while in UDP ffmpeg never freeze but there are too many artifacts.
If you could solve these points it would be a perfect cam with your hack!

roleoroleo, 2022-01-09:
Maybe startup.sh arrives too late and wpa_supplicant already started.
You could try to add a command that kills wpa_supplicant and copies the file.
The yi watchdog will restart wpa_supplicant again.

About motion detection you could try to descrease sensitivity.
If I leave my phone in the room where I have the cam, the led is enough to start the motion.
About human detection, I already tried to fix the problem but no success.

RTSP is a know issue but I can't find the problem, the grabber is ok and the server is live555 (so a "big" software).
I don't know how to improve the reliability.
Try to read this: https://github.com/roleoroleo/yi-hack-MStar/issues/36

Maxym77, 2022-01-09:
very tanks, i have added in startup.sh:
killall wpa_supplicant
seems works fine

my sensitivity is at minimum... human detection with inverted mqtt works but sometime there is event triggered without human in the rooms...  i think the problem also show up with the original software and no-hack.

I see the original app and the desktop app (https://www2.yitechnology.com/yi-home-pc-app) works very flawless.
I think we need to investigate here.

roleoroleo, 2022-01-10:
The original stream is encapsulated inside a custom protocol and I'm not able to decode it.

closed 2023-01-08 by stale[bot]