roleoroleo/yi-hack-Allwinner-v2 #329: y29ga base version 9.0.09.06_202103291039 support

kiddyfurby, 2022-01-03:
Hi roleoroleo, I was so excited when you started supporting the Y4, I brought 2 extra.
My original camera that you [developed support on](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/268) had firmware 0.1.9, base version 9.0.09.**03_202006090939**
The 2 new cameras came with firmware 0.1.9, but base version 9.0.09.**06_202103291039**

Is there any chance the difference in base version results in a different stream format?
There are 3 notable issues comparing to my -03 camera:
1. in the status page, serial number becomes 00000000000000000000
2. the RTSP stream is less compatible:
- unplayable on Pixel 6 VLC with regardless of hardware acceleration,
- after audio started playing, took a few seconds to show first frame on Mac OS VLC
3. camera would sometimes reboot itself upon starting a new RTSP session / changing settings from the web page (might be my USB multi-charger) 

I attach a dump from the new camera below just in case it helps:
[cam_2_dump_backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7800568/cam_2_dump_backup.zip)

Thank you!!

kiddyfurby, 2022-01-06:
I looked into the new fshare_frame_buf above, it is inline with https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/d44a3fe29c17aa51e37614e3161b146907c18101

1-Al, 2022-01-07:
Uses different SPS.

FWIW appears to use sp2305 sensor which is 1920x1080, not native 1296p!

kiddyfurby, 2022-01-07:
Hm...  I checked lsmod, and they ran different drivers
9.0.09.03_202006090939  sp2305_mipi
9.0.09.06_202103291039 sp2305_mipi_y211_y291

but both are sp2305 and the RTSP stream from camera with 9.0.09.03_202006090939 does play properly on all devices.

roleoroleo, 2022-01-07:
I will check it.

kiddyfurby, 2022-01-09:
In my case, I already have the 9.0.09.03 working, so if 9.0.09.06 is supported in upcoming yi-hack releases it means everything is working.

Maybe we can come back to work on support for multiple firmware base versions if incompatible base firmware becomes the normal with yi cams in the future.

roleoroleo, 2022-01-09:
The backup you posted is made after hacking.
Do you have a factory backup?

roleoroleo, 2022-01-09:
Try this:
[new.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7835638/new.tar.gz)

roleoroleo, 2022-01-09:
@kiddyfurby 
I need more details about the differences between two mmap.info to create an automatic detection.
Could you please send me a copy of both versions?
If you don't want to publish them here please send me a mail to my paypal address.

kiddyfurby, 2022-01-10:
I have no factory backup for -06 but I could buy a new cam, that could take a few days.

mmap's are here: I've replaced the ssid and wpa password in plain text so perhaps some checksum could break but I guess the files are good enough for you?
[mmap.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7836373/mmap.zip)

kiddyfurby, 2022-01-10:
> Try this: [new.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7835638/new.tar.gz)

The stream is playable on Pixel 6 VLC and Home Assistant WebRTC now!! THANK YOU!
Let me know if you would still like to receive a clean firmware dump?

The RTSP stream has always been considerably less smooth comparing to the recorded mp4, regardless of firmware base version - I will make some recordings to demonstrate.

roleoroleo, 2022-01-10:
> Let me know if you would still like to receive a clean firmware dump?

I doesn't matter.

kiddyfurby, 2022-01-14:
Had some random reboot when using the web UI and sometimes just randomly.
I turned off the low quality rtsp stream and its been running well ever since.

[image]
stats from 9.0.09.03
[image]
stats from 9.0.09.06

Perfectly happy about the higher load as long as they stream consistently and don't reboot randomly.

roleoroleo, 2022-01-14:
Is swap space enabled?

kiddyfurby, 2022-01-15:
No it wasn’t, I will try.

roleoroleo, 2022-01-15:
Random reboots are normally the effect of a low memory problem.
When the cam detects a motion and tries to take a snapshot, the memory is not enough and the kernel reboots the cam.

hengfeb, 2022-03-22:
@kiddyfurby  
Could you help me to fix "On Paring" -- "This camera can only use in china", plz!
Henry

0xBURP, 2022-12-09:
@hengfeb My camera did say that to me, it's not a problem.
Follow these steps:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Install-the-hack-without-the-Yi-app

closed 2023-04-08 by stale[bot]