roleoroleo/yi-hack-Allwinner-v2 #330: Black screen in stream both on mobile phone and home assistant (frigade)

YoykoKoszko, 2022-01-03:
Hi all

I have problem where my camera stream is black page when I try to use it on mobile phone (vlc android) and home assistant (frigade integration) it was OK on laptop. What do I have to do?

roleoroleo, 2022-01-06:
> Hi all
> 
> I have problem where my camera stream is black page when I try to use it on mobile phone (vlc android) and home assistant (frigade integration) it was OK on laptop. What do I have to do?

Which app are you using on your laptop?

YoykoKoszko, 2022-01-06:
I am using same app on both pc and phone which is VLC player.

roleoroleo, 2022-01-07:
I don't use frigade but I can confirm that this cam works with:
- ha + webrtc
- vlc on pc
- vlc on andrtoid

With vlc use the rtsp address rtsp://192.168.1.200/ch0_0.h264
With your IP.

closed 2023-01-08 by stale[bot]