roleoroleo/yi-hack-Allwinner-v2 #335: QG2101B  firmware   9.0.41.99_202109231032 use b091qp cannot connect to wifi and rtsp

pengkaiwei, 2022-01-14:
Hi, thank you very much for your contribution.
My cam chip is QG2101B and the firmware version is 9.0.41.99\_202109231032.
I tried [#249 (reply in thread)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/249#discussioncomment-1378470),
But wifi can't find the device, plug in the network cable directly, you can see the device ip, but rtsp still has no image.

Below are some pictures of my device, I hope it can help you. This is my factory firmware backup.

[image]

[image]
[image]
[image]
[image]
[image]

This is my factory firmware backup.

[cam backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7871565/cam.backup.zip)

roleoroleo, 2022-01-14:
The cam is bq091qp.
But there are different versions of b091qp.
I have to check.

pengkaiwei, 2022-01-15:
thank you

> The cam is bq091qp. But there are different versions of b091qp. I have to check.

Thank you and I look forward to your good news.
If you need help with anything, just let me know.

closed 2023-01-08 by stale[bot]

forgani, 2023-01-13:
Hi, I have the same camera. 
@pengkaiwei did you make it and whether the last release supports this version. Thanks

forgani, 2023-01-13:
**awesome** . Good work I confirm that everything is fine. works perfect. many thanks